ghdl/ghdl #2: ghdl crashes in VHDL-2008 mode, but not in VHDL-93 mode

Paebbels, 2015-11-24:
I wrote a simple sorting network (odd even sort), that compiles in the following simulators:
- GHDL 0.33 (--std 93)
- QuestaSim 10.4c (--93 and --2008)

But when I compile the same code with GHDL and switch to `--std=08`, it crashes with the following error message:

    ******************** GHDL Bug occured ****************************
    Please report this bug on http://gna.org/projects/ghdl
    GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
    Compiled with GNAT Version: GPL 2015 (20150428-49)
    In directory: D:\git\PoC\temp\ghdl\
    Command line:
    C:\Tools\GHDL\0.33\bin\ghdl.exe -r --syn-binding -P. -Paltera -Pxilinx -Posvvm -Pvuint --std=08 --work=test sortnet_OddEvenSort_tb
    Exception CONSTRAINT_ERROR raised
    Exception information:
    Exception name: CONSTRAINT_ERROR
    Message: ortho_code-decls.adb:122 index check failed
    ******************************************************************

The shrinked example design is attached as *.txt files (VHDL files can't be uploaded):
 - [sortnet_OddEvenSort.vhdl.txt](https://github.com/tgingold/ghdl/files/42934/sortnet_OddEvenSort.vhdl.txt)
 - [sortnet_OddEvenSort_tb.vhdl.txt](https://github.com/tgingold/ghdl/files/42935/sortnet_OddEvenSort_tb.vhdl.txt)

(All files are compiled into a library called `test`.)

tgingold, 2015-11-24:
Does it crash without the -P flags (-P. ...) ?

I cannot reproduce the crash on unix, I will investigate on Windows.

Paebbels, 2015-11-25:
I removed the `-P` parameters.
VHDL-93 compiles
VHDL-2008 crashes

    ghdl.exe -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding --std=08 --work=test sortnet_OddEvenSort.vhdl
    ghdl.exe -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding --std=08 --work=test sortnet_OddEvenSort_tb.vhdl
    ghdl.exe -r --syn-binding --std=08 --work=test sortnet_OddEvenSort_tb

tgingold, 2015-11-26:
I cannot reproduce on windows too, using either master or the 0.33 release.
However, I have filed the test.

Can you reproduce the crash on linux ?
Can you reproduce the crash on windows using the 'official' 0.33 binary ?
How did you compile your binary ?

Puzzling...
Tristan.

Paebbels, 2015-11-26:
I tried to compile GHDL from sources to reproduce this issue with the newest version.
The linker reports an unknown reference in win32.c. The error is reported as issue #3.

Paebbels, 2015-11-26:
Issue #3 is solved. I can neither reproduce the error from above in VHDL-93 nor in VHDL-2008.

But after recompilation of GHDL and a new analyses of all libs I have the following problem:
Simulation compiles in VHDL-93 but in VHDL-2008 mode, it does not find the IEEE library...

Paebbels, 2015-11-26:
I think I got it ...
Are VHDL-2008 and a non standard IEEE library mutual exclusive?
Should GHDL ignore/complain about `--ieee=(synopsys|mentor)` if `--std=08`?

tgingold, 2015-11-27:
A warning is now emitted for this case.  Thank you for the suggestion.
Tristan.

Paebbels, 2015-12-01:
So, but I think I have to reopen this issue :(.

I created the next sorting network `sortnet_BitonicSort` and hit the error again.
GHDL is compiled fresh from the sources: 0.34dev (20151126).
As last time, iSim, xSim and vSim have no complains, so I expect the VHDL source is good.

The error is now reproducible in GHDL's 93c and 08 modes. For example:
```
******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
In directory: D:\git\PoC\temp\ghdl\
Command line:
c:\Tools\GHDL\0.34dev\bin\ghdl.exe -r --syn-binding --std=93c --work=test sortnet_BitonicSort_tb
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: trans-chap1.adb:850 access check failed
******************************************************************
```

The ghdl flow:
```PowerShell
running analysis for every vhdl file...
  ghdl.exe -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -P. --std=08 --work=test D:\git\PoC\src\sort\sortnet\sortnet_BitonicSort.vhdl
  ghdl.exe -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -P. --std=08 --work=test D:\git\PoC\tb\sort\sortnet\sortnet_BitonicSort_tb.vhdl
running simulation...
  ghdl.exe -r --syn-binding -P. --std=08 --work=test sortnet_BitonicSort_tb
ERROR while executing ghdl command: c:\Tools\GHDL\0.34dev\bin\ghdl.exe -r --syn-binding -P. --std=08 --work=test sortnet_BitonicSort_tb
```

Here are the new files for bug reproduction:
 - [sortnet_BitonicSort.vhdl.txt](https://github.com/tgingold/ghdl/files/47959/sortnet_BitonicSort.vhdl.txt)
 - [sortnet_BitonicSort_tb.vhdl.txt](https://github.com/tgingold/ghdl/files/47960/sortnet_BitonicSort_tb.vhdl.txt)

tgingold, 2015-12-01:
On 01/12/15 01:01, Patrick Lehmann wrote:
> So, but I think I have to reopen this issue :(.
>
> I created the next sorting network |sortnet_BitonicSort| and hit the
> error again.
> GHDL is compiled fresh from the sources: 0.34dev (20151126).
> As [...]

I can reproduce this, but I suppose this is a very different issue.

Investigating.

tgingold, 2015-12-01:
On 01/12/15 01:01, Patrick Lehmann wrote:
> So, but I think I have to reopen this issue :(.
>
> I created the next sorting network |sortnet_BitonicSort| and hit the
> error again.
> GHDL is compiled fresh from the sources: 0.34dev (20151126).
> As [...]

This is a case of recursive entity instantiation, not yet supported by
ghdl...
(your error is fixed but then ghdl crashes a little bit later).
As a work-around, use component instantiation in 
sortnet_BitonicSort_Merge and sortnet_BitonicSort_Sort

Interesting design...

Tristan.

Paebbels, 2015-12-01:
> Interesting design...An easy ("software"-like) description uses recursion to describe the bitonic algorithm. I already have a non recursive version requiring 4 nested generate loops and some math extensions like calculation of the triangular number. That version can also be pipelined.It's hard to decide where to put a pipeline register in a recursive description, because there is no easy way to calculate (STAGE_INDEX mod ADD_REGISTER_AFTER = 0)especially in the bitonic algorithm because it's build from the center recursive to the inputs and recursive to the outputs :(.The goal is to test osvvm for verification.Ok, if the bug is known, I think you can close the issue again :).Regards    Patrick -----------------------------------Wissenschaftliche HilfskraftTechnische Universität DresdenFakultät InformatikInstitut für Technische InformatikLehrstuhl VLSI-Entwurfssysteme, Diagnostik und Architektur         01062 Dresden, GERMANYTel.:  +49 351 463-38451                                       Fax:  +49 351 463-38324E-Mail: Patrick.Lehmann@tu-dresden.de   WWW:    http://vlsi-eda.inf.tu-dresden.de-------- Ursprüngliche Nachricht --------Von: tgingold  Datum:01.12.2015  04:37  (GMT+01:00) An: tgingold/ghdl  Cc: "Lehmann, Patrick"  Betreff: Re: [ghdl] ghdl crashes in VHDL-2008 mode, but not in VHDL-93 mode (#2) 

This is a case of recursive entity instantiation, not yet supported by
ghdl...
(your error is fixed but then ghdl crashes a little bit later).
As a work-around, use component instantiation in 
sortnet_BitonicSort_Merge and sortnet_BitonicSort_Sort

Interesting design...

Tristan.

—
view it on GitHub.

tgingold, 2015-12-01:
On 01/12/15 08:12, Patrick Lehmann wrote:
> recursion to describe the bitonic algorithm. I already have a non
> recursive version requiring 4 nested generate loops and some math
> extensions like calculation of the triangular number. That version [...]

I will keep this issue open and close it when the bug is fixed.

Tristan.

diogratia, 2016-03-26:
I found a simple recursion error in ghdl-0.33 (llvm, OS X), default --std (93c):
```vhdl
entity visible is
    port (
        foo: in boolean
    );
end entity;

architecture foo of visible is
    signal foo_int: boolean;
begin
FUMBLE: 
    entity work.visible
        port map (
            foo => foo_int
        );
end architecture;
```
With or without the port declaration and port map.  (And for whatever reason people just love to cite recursion on Stackoverflow).

closed 2016-07-03 by tgingold