ghdl/ghdl #3: gnatlink error on Windows: win32.c:99: undefined reference to `_imp__StackWalk64@36'

Paebbels, 2015-11-26:
I tried to compile GHDL on Windows to reproduce issue #2 with the latest version built from git sources. The compilation executes until gnatbind and exits with the following error message:

```Bash
gnatbind -aID:\git\GitHub\ghdl\dist\mcode\windows -aID:\git\GitHub\ghdl\src  aID:\git\GitHub\ghdl\src\ghdldrv -aID:\git\GitHub\ghdl\src\psl -aID:\git\GitHub\ghdl\src\grt -aID:\git\GitHub\ghdl\src\ortho -aID:\git\GitHub\ghdl\src\ortho\mcode -aID:\git\GitHub\ghdl\src\vhdl -aID:\git\GitHub\ghdl\src\vhdl\translate -x ghdl_jit.ali
gnatlink ghdl_jit.ali -O -g -o ghdl.exe grt-cbinding.o clock.o grt-cvpi.o memsegs_c.o win32.o -Wl,--stack,8404992
win32.o: In function `get_bt_from_context':
D:/git/GitHub/ghdl/src/grt/config/win32.c:99: undefined reference to `_imp__StackWalk64@36'
collect2.exe: error: ld returned 1 exit status
gnatlink: error when calling C:\Tools\GNAT\2015\bin\gcc.exe
```

The compilation was called with this line:
```PowerShell
.\winbuild.ps1 -Clean -Compile
```

Built environment:
```
OS:       Windows 7 Enterprise - German - x86-64
Shell:    PowerShell 4.0
Compiler: GNAT GPL 2015 (20150428-49)`
```

tgingold, 2015-11-26:
Can you try with branch issue3 ?

Paebbels, 2015-11-26:
The changes from branch 'issue3' solved the linker error.

closed 2015-11-26 by Paebbels