ghdl/ghdl #4: GHDL doesn't set a non-zero return exit value on errors in Linux (gcc backend)

Paebbels, 2015-11-30:
I was up to finish the vendor specific compile scripts for GHDL on Linux, when I noticed that GHDL (gcc backend) does not set a non-zero exit code if an error occurred. The Windows version (mcode backend) with returns "1".

```PowerShell
Analyzing primitive 'C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11CLK_MGT.vhd'
return=0
Analyzing primitive 'C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11_CUSTOM.vhd'
ERROR: C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11_CUSTOM.vhd:328:13: no declaration for "gt11"
ERROR: C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11_CUSTOM.vhd:328:13: component name expected
return=1
```

The return line is reported by:
`Write-Host "return=$LastExisCode"`

Here is the equivalent Linux output:
```Bash
Analyzing primitive '/opt/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/secureip/GT11_CUSTOM.vhd'
/opt/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/secureip/GT11_CUSTOM.vhd:328:13: no declaration for "gt11"
/opt/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/secureip/GT11_CUSTOM.vhd:328:13: component name expected
return 0
```

The return line is reported by:
`echo "return $?"`

Is it intended to be always zero? How can I test if an error occurred, except for parsing the complete output line by line :).

The [source code](https://github.com/Paebbels/ghdl/blob/paebbels/lib-compile-linux/libraries/vendors/compile-xilinx-ise.sh) for the compile script is available on my GHDL fork on branch `paebbels/lib-compile-linux`.

tgingold, 2015-11-30:
On 30/11/15 18:09, Patrick Lehmann wrote:
> I was up to finish the vendor specific compile scripts for GHDL on
> Linux, when I noticed that GHDL (gcc backend) does not set a non-zero
> exit code if an error occurred. The Windows version (mcode backend) with
> returns "1".
>
> Analyzing primitive'C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11CLK_MGT.vhd'
> return=0
> Analyzing primitive'C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11_CUSTOM.vhd'
> ERROR: C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11_CUSTOM.vhd:328:13: no declarationfor  "gt11"
> ERROR: C:\Xilinx\14.7\ISE_DS\ISE\vhdl\src\unisims\secureip\GT11_CUSTOM.vhd:328:13: component name expected
> return=1
>
> The return line is reported by:
> |Write-Host "return=$LastExisCode"|
>
> Here is the equivalent Linux output:
>
> Analyzing primitive'/opt/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/secureip/GT11_CUSTOM.vhd'
> /opt/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/secureip/GT11_CUSTOM.vhd:328:13: no declarationfor  "gt11"
> /opt/Xilinx/14.7/ISE_DS/ISE/vhdl/src/unisims/secureip/GT11_CUSTOM.vhd:328:13:
> component name expected
> return 0
>
> The return line is reported by:
> |echo "return $?"|
>
> Is it intended to be always zero? How can I test if an error occurred,
> except for parsing the complete output line by line :).

That's because you're using a pipe:

       The return status of a pipeline is the exit status of the last 
command,
        unless  the  pipefail  option  is enabled.  If pipefail is 
enabled, the
        pipeline's return status is the value of the last  (rightmost) 
command
        to  exit  with a non-zero status, or zero if all commands exit 
success-
        fully.

Tristan.

Paebbels, 2015-11-30:
Damnit!

I totally missed that...
PowerShell distinguishes the exit codes between the last pipe command and the last native/shell command :).

I fixed it with `set -o pipefail`.

closed 2015-11-30 by Paebbels

tgingold, 2015-11-30:
> Damnit!
>
> I totally missed that...
> PowerShell distinguishes the exit codes between the last pipe command
> and the last native/shell command :).

I agree that this is not obvious.
But I knew this was not a ghdl issue!