ghdl/ghdl #8: Problems with parallel invocation of GHDL - GCC backend

kraigher, 2015-12-18:
A VUnit+GHDL user has reported a [problem](https://github.com/LarsAsplund/vunit) with running parallel simulations of the same test bench using GHDL with the GCC backend. 
It seems similar to the problem I reported in [ticket_87](https://sourceforge.net/p/ghdl-updates/tickets/87/) which were fixed for the LLVM backend.

The problem manifests itself in the following way when running --elab-run on multiple tests in parallel on the same top level:
```
Error: can't open e~tb_test_lib-tb.s for reading: No such file or directory
```

tgingold, 2015-12-18:
Do you have a more verbose log ?  It looks like (but I am not sure) that vunit is spawning multiple instances of ghdl with the same executable name.

I think the fix for ticket 87 affects both llvm and gcc.  So the issue should also happen with the llvm backend.

kraigher, 2015-12-18:
The fix for 87 which works for LLVM backend was that you added a `-o` flag to specify a folder where all output artifacts were writte during `--elab-run`. VUnit gives an unique output path for each test run to ensure conflict free parallel operation. Maybe the flag is not fully working as required for gcc backend?

kraigher, 2015-12-18:
I think maybe the flag does not work for the .s files. There was a similar problem with .o files with LLVM. See http://sourceforge.net/p/ghdl-updates/tickets/87/#36a6

tgingold, 2015-12-18:
Yes, I will double check for .s files.  That's a difference as there is not .s files for llvm.

tgingold, 2015-12-20:
No.  The .s file is created with the right name:

$ ghdl_gcc --elab-run -v -o ../dir3/hello1 hello
ghdl1-gcc [...] --elab hello -l ../dir3/e~hello1.lst -quiet -o ../dir3/e~hello1.s ../dir3/e~hello1
/usr/bin/as -o ../dir3/e~hello1.o ../dir3/e~hello1.s
/usr/bin/gcc -o ../dir3/hello1 ../dir3/e~hello1.o [...]
./../dir3/hello1

So I still need a verbose log to understand this issue.
Tristan.

kraigher, 2015-12-20:
@tgingold I found the problem on the VUnit side. It turns out we did not provide the `-o` flag for the GCC backend.
I am really sorry to have wasted your time!

closed 2015-12-20 by kraigher

tgingold, 2015-12-20:
No problems.

BTW, as ghdl 0.33 was released and supports VUNIT, the README.md of VUNIT can be updated.

Thanks,
Tristan.

kraigher, 2015-12-20:
Thanks for the reminder. I have now updated it.