ghdl/ghdl #9: GHDL Bug occured on invalid assignment.

vicencb, 2015-12-29:
The following selected signal assignment is invalid because of a duplicate entry:
```vhdl
entity tmp is end entity;
architecture arch of tmp is
	signal b : bit;
begin
	with true select b <=
	 '0' when false | true,
	 '1' when true;
end architecture;
```

Ghdl instead of reporting the error fails with:
```sh
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 5.3.0
In directory: /tmp/
Command line:
ghdl -m tmp
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: iirs.adb:2475
Call stack traceback locations:
0x4f307f 0x419e14 0x43c4ee 0x4b50bf 0x4b9f98 0x49b7ab 0x49e224 0x49e87e 0x4c5d0b 0x46b84c 0x45c988 0x46aac2 0x475a46 0x49109a 0x480eb3 0x405706 0x4031bf 0x7f5b60f1660e 0x402747 0xfffffffffffffffe
******************************************************************
```

It has been tested wirh d3d07baf3546cd5aa52d3632762f18c4f4982141 and 2ac84541d494e06b7782cd11515264fd43f7532f, both failed.

Thanks!

tgingold, 2015-12-29:
This is a crash during the error report.
I am testing the fix.

Thank you for reporting this issue,
Tristan.

tgingold, 2015-12-30:
I have just pushed the fix.  Thanks!

closed 2015-12-30 by tgingold