ghdl/ghdl #10: Why is an unconstrained constant vector in GHDL not considered to be static?

Paebbels, 2016-01-02:
I have the following code:

```VHDL
architecture rtl of test id
  type T_TUPLE is record
    A  : NATURAL;
    B  : NATURAL;
  end record;
  type T_VECTOR is array (NATURAL range <>) of T_TUPLE;

  constant LIST : T_VECTOR := ((8, 32), (8, 20), (8, 36));
begin
  genTests : for i in LIST'range generate
    constant LOCAL_A : NATURAL := LIST(i).A;
    constant LOCAL_B : NATURAL := LIST(i).B;
  begin
    -- my tests
  end generate;
end architecture;
```

I get the following error message for my generate statement:

> ... range must be a static discrete range

When I change my constant to a constrained variant it works...

```VHDL
constant LIST : T_VECTOR(0 to 2) := ((8, 32), (8, 20), (8, 36));
```

Why is the 2. example more static?

I also tried this code:

```VHDL
constant initial_LIST : T_VECTOR := ((8, 32), (8, 20), (8, 36));
constant LIST : T_VECTOR(initial_LIST'range) := initial_LIST ;
```

The error is the same at the generate statement. But there is no problem at the second constant declaration.

My local GHDL version is 0.33 (20150921). This PC had no build environment to check it in 0.34dev.

diogratia, 2016-01-03:
Note by adding an entity declaration 

```vhdl
entity test_id is
end entity;
```

your code will analyze, elaborate and simulate (while not actually doing anything) in ghdl-0.33 (llvm OS X).  

It won't analyze with the standard set to -2008:

> ghdl -a --std=08 test_id.vhdl
> test_id.vhdl:14:3: range must be a static discrete range
>ghdl: compilation error

Paebbels, 2016-01-03:
Thanks for the VHDL-93 test.

I updated my code example.
I'm using VHDL-2008.

tgingold, 2016-01-03:
On 02/01/16 19:55, Patrick Lehmann wrote:
> I have the following code:
>
> architecture  rtlof  test id
>    type  T_TUPLEis  record
>      A  :NATURAL;
>      B  :NATURAL;
>    end  record;
>    type  T_VECTORis  array  (NATURAL  range  <>)of  T_TUPLE;
>
>    constant  LIST : T_VECTOR:=  ((8,32), (8,20), (8,36));
> begin
>    genTests :for  iin  LIST'range  generate
>      constant  LOCAL_A :NATURAL  :=  LIST(i)A;
>      constant  LOCAL_B :NATURAL  :=  LIST(i)B;
>    begin
>      -- my tests
>    end  generate;
> end  architecture;
>
> I get the following error message for my generate statement:
>
>     range must be a static discrete range
>
> When I change my constant to a constrained variant it works
>
> constant  LIST : T_VECTOR(0  to  2):=  ((8,32), (8,20), (8,36));
>
> Why is the 2 example more static?

I think GHDL follows LRM 7.4.2 Globally static primaries:

A globally static range is either a range of the second form (see 3.1) 
whose bounds are globally static expressions [not our case], or a range 
of the first form whose prefix denotes either a globally static subtype 
[not our case] or an object that is of a globally static subtype [our case].

A globally static index constraint is an index constraint for which each 
index subtype of the corresponding array type is globally static and in 
which each discrete range is globally static. A globally static array 
subtype is a constrained array subtype formed by imposing on an 
unconstrained array type a globally static index constraint. A globally 
static record subtype is a record type whose fields are all of globally 
static subtypes. A globally static access subtype is a subtype denoting 
an access type. A globally static file subtype is a subtype denoting a 
file type.

So an unconstrained type is not globally static.

Tristan.

Paebbels, 2016-01-03:
I see your point.

Should we move this discussion to the VHDL mailing list?

In my eyes the right hand side of the declaration is composed of integer literals, which are globally static, aren't they? So there is only one way to infere the bounds of the array type, right?

My second thought: How can GHDL construct a constant if it is not static? Or asked in another way why is it static enough to construct a constant, but not enough to act as a static range for a generate statement?

(I admit that I have not read all of the LRM - so you have much more expertise on the topic - but I can't see any way why a compiler can not solve the problem.)

If so, I think we should ask on the mailing list if the rules should be revised.

Would this problem be a canfidate for your relaxed-rules switch?

tgingold, 2016-01-04:
On 03/01/16 23:46, Patrick Lehmann wrote:
> I see your point.
>
> Should we move this discussion to the VHDL mailing list?

That's up to you!

> In my eyes the right hand side of the declaration is composed of integer
> literals, which are globally static, aren't they? So there is only one
> way to infere the bounds of the array type, right?

Yes, that's true.

> My second thought: How can GHDL construct a constant if it is not
> static? Or asked in another way why is it static enough to construct a
> constant, but not enough to act as a static range for a generate statement?

As you know, the language is defined by the LRM and GHDL tries to follow it.

> (I admit that I have not read all of the LRM - so you have much more
> expertise on the topic - but I can't see any way why a compiler can not
> solve the problem.)

It could solve it (and solve it with --std=93c).

> If so, I think we should ask on the mailing list if the rules should be
> revised.
>
> Would this problem be a canfidate for your relaxed-rules switch?

Yes.  The LRM rules are not very clear (at least to me) on that point.
Discussion is welcome.

Tristan.

JimLewis, 2016-01-06:
HI Tristan, 
From section 5.3.2.2 (VHDL-2008)
The index range for each index of an array object or array subelement of a composite object is determined as follows:
. . .
b) For a constant declared by an object declaration, if the subtype of the constant defines the index
range, the index range of the constant is that defined by the subtype; otherwise, the index range of
the constant is the corresponding index range of the initial value [our case].

Hence, for a constant, the subtype is constrained.

Jim

tgingold, 2016-01-12:
Now fixed.

closed 2016-01-12 by tgingold