ghdl/ghdl #11: Creating an array of protected types causes a GHDL crash.

Paebbels, 2016-01-03:
I have created an array type of an protected type. The file analyzes but causes a crash in elaboration. (GHDL 0.33 - 20150921 - windows)

According to [this SO question](http://stackoverflow.com/q/27553803/3719459) it's currently not allowed in VHDL-2008, but I think GHDL should exits with an error message instead of an "TYPES.INTERNAL_ERROR exception in trans.adb:551

Regards
    Patrick

tgingold, 2016-01-03:
On 03/01/16 14:52, Patrick Lehmann wrote:
> I have created an array type of an protected type The file analyzes but
> causes a crash in elaboration (GHDL 033 - 20150921 - windows)
>
> According to this SO question [...]

An error is now report during analysis, will push soon.

Thanks,
Tristan.

tgingold, 2016-01-04:
Now fixed.  Thank you for the report.

closed 2016-01-04 by tgingold