ghdl/ghdl #13: Error compiling

clearscene, 2016-01-05:
Hi,

I'm attempting to compile ghdl on 32bit Linux, but am getting an error.
All goes well until:

```
gnatmake -o ghdl_mcode -aI./src -aI./src/vhdl -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/grt -aI./src/ortho -aI./src/ortho/mcode -gnaty3befhkmr -gnatwae -aO. -gnatf -gnat05 -g -gnata ghdl_jit.adb -bargs -E -largs memsegs_c.o chkstk.o  jumps.o times.o grt-cbinding.o grt-cvpi.o fstapi.o lz4.o fastlz.o -ldl -lm -Wl,--version-script=./src/grt/grt.ver -Wl,--export-dynamic
gcc-4.6 -c -I./ -I./src -I./src/vhdl -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/grt -I./src/ortho -I./src/ortho/mcode -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -I- /home/wbooth/git/ghdl/src/grt/grt-lib.adb
grt-lib.ads:26:09: warning: no entities of "Rtis" are referenced
gnatmake: "/home/wbooth/git/ghdl/src/grt/grt-lib.adb" compilation error
make: *** [ghdl_mcode] Error 4
```

My system reports:
Linux netbook 3.13.0-39-generic #66-Ubuntu SMP Tue Oct 28 13:31:23 UTC 2014 i686 i686 i686 GNU/Linux

Any advice on this would be appreciated.

Cheers

tgingold, 2016-01-06:
On 05/01/16 21:47, Wayne Booth wrote:
> Hi,
>
> I'm attempting to compile ghdl on 32bit Linux, but am getting an error
> All goes well until:
>
> |gnatmake -o ghdl_mcode -aI/src -aI/src/vhdl -aI/src/psl
> -aI/src/vhdl/translate -aI/src/ghdldrv [...]

Hello,

simply remove the 'with grt.rtis; use grt.rtis;' line in grt-lib.ads
I will fix it in the repository and investigate why it was not caught.

Thank you for the report,
Tristan.

tgingold, 2016-01-07:
I think this is definitely fixed.  I got a full build ok on Ubuntu 12.04 (from travis-ci)

Tristan.

closed 2016-01-07 by tgingold