ghdl/ghdl #14: VHDL-2008 Matching Equality Relationals with erroneous warning message

JimLewis, 2016-01-05:
Matching Equality Relationals are producing the following error message:
std_logic_1164.vhdl:58:3:@0ms:(assertion error): STD_LOGIC_1164: '-' operand for matching ordering operator
std_logic_1164.vhdl:58:3:@0ms:(assertion error): STD_LOGIC_1164: '-' operand for matching ordering operator
Oddly it seems that 

each call produces two warnings

JimLewis, 2016-01-05:
[alert_if_equal_issue.txt](https://github.com/tgingold/ghdl/files/79183/alert_if_equal_issue.txt)
This file requires at least the OSVVM 2015.06 release, it may require my current working version for 2016.01.

JimLewis, 2016-01-05:
I should also note that my simulation still produces correct results, so other than this annoying message, it works as expected.

JimLewis, 2016-01-05:
Impact analysis:  impacts OSVVM AlertLogPkg procedures: AlertIfEqual and AlertIfNotEqual for std_logic, std_logic_vector, unsigned, and signed.

tgingold, 2016-01-06:
On 05/01/16 23:57, JimLewis wrote:
> Matching Equality Relationals are producing the following error message:
> std_logic_1164vhdl:58:3:@0ms <https://github.com/0ms>:(assertion error):
> STD_LOGIC_1164: '-' operand for matching ordering [...]

I think this is because ?= for std_ulogic_array calls ?= for std_ulogic,
and the warning is emitted by the latter.

I will fix that (not high priority however).

Thanks for the report,
Tristan.

tgingold, 2016-01-07:
This issue is now fixed.

closed 2016-01-07 by tgingold