ghdl/ghdl #15: Referencing an Alias within a use clause

JimLewis, 2016-01-06:
## Issue
Numeric_std_unsigned includes aliases to functions to provide a wider range of names and sometimes fix the evils of the past.  I needed to reference a couple of functions from numeric_std_unsigned without replacing "=" (so I can correctly check for 'U').  Hence, to get access to to_slv, I did:
```
use ieee.numeric_std_unsigned.to_slv ; 
````
## Error messages:
to_slv_issue.vhd:14:25: no overloaded function found matching 'to_slv'

## Test Case and Work Around
Test case:  [to_slv_issue.txt](https://github.com/tgingold/ghdl/files/79225/to_slv_issue.txt)
 
Work around:  [to_slv_issue_work_around.txt](https://github.com/tgingold/ghdl/files/79226/to_slv_issue_work_around.txt)

## Impact
This was done in a testbench model that was testing a memory model package (OSVVM 2015.06).  When the memory location is uninitialized, it returns "UUUU", however, it the memory location was written to with an 'X' it returns "XXXX".  I needed both an easy conversion from integer to std_logic_vector as well as a way to check for the values "UUUU" and "XXXX".  I don't expect these to happen often.

tgingold, 2016-01-06:
On 06/01/16 01:00, JimLewis wrote:
> Issue
>
> Numeric_std_unsigned includes aliases to functions to provide a wider
> range of names and sometimes fix the evils of the past I needed to
> reference a couple of functions from [...]

I can reproduce this issue and I think I understand the issue.

Tristan.

tgingold, 2016-01-07:
This issue is now fixed.

closed 2016-01-07 by tgingold