ghdl/ghdl #16: Reading a file from within a protected type (file identifier local to pt)

JimLewis, 2016-01-06:
## Issue
When trying to read from a file from within a protected type using a local file identifier, the following error message results:
c:\tools\ghdl-0.33\bin\ghdl.exe:error: overflow detected
c:\tools\ghdl-0.33\bin\ghdl.exe:error: simulation failed
## Test Case
VHDL Test File:  [pt_file_read_issue.txt](https://github.com/tgingold/ghdl/files/79264/pt_file_read_issue.txt)
Required file to be read:  [TbAlertLog_t8_log1.txt](https://github.com/tgingold/ghdl/files/79265/TbAlertLog_t8_log1.txt)
## Impact:  Important to OSVVM
Initialization of Memory Models created using MemoryPkg requires file reading.   A coverage database may be written and then read in (used to randomize a configuration at the start of a test).  The initialization for log enables within AlertLogPkg can be accomplished by reading a file.  This issue prevents these from working.

tgingold, 2016-01-06:
Development version of ghdl includes a backtrace which is useful:
error: overflow detected
  from: osvvm.textutilpkg.skipwhitespace at TextUtilPkg.vhd:150
  from: osvvm.textutilpkg.emptyorcommentline at TextUtilPkg.vhd:211
  from: osvvm.alertlogpkg.readlogenables at AlertLogPkg.vhd:2676
  from: osvvm.alertlogpkg.readlogenables at AlertLogPkg.vhd:2707
  from: process work.pt_file_read_issue(t8_file).testproc at pt_file_read_issue.vhdl:59

In osvvm.textutilpkg.skipwhitespace, you assume that the first character of L is at index 1, which is
not correct.  Use L (L'left).
I think the function was copied from vhdl 2008.

Thanks for the report,
Tristan.

tgingold, 2016-01-06:
Likewise in EmptyOrCommentLine.

JimLewis, 2016-01-07:
Thanks for the feedback, I will root that out of OSVVM packages.

closed 2016-01-07 by tgingold