ghdl/ghdl #18: Function overloading vs Indexing

JimLewis, 2016-01-06:
## Issue
overload_index_issue.txt:20:35: can't resolve overload for function call, slice or indexed name
overload_index_issue.txt:20:35: possible interpretations are:
c:/SynthWorks/Dev/_osvvm/src/CoveragePkg.vhd:414:21: function randcovpoint [real return integer]
overload_index_issue.txt:20:35: indexed name
## Visible functions
function randcovpoint [real return integer]
function randcovpoint [return integer_vector]
## Thoughts
There are a number of places where this happens and the function input is an integer.  Hence, perhaps it is possible for the LRM to be unclear about whether function overload or indexing is possible, however, in this case, type real clearly cannot be used as a type for indexing as it is not a discrete type and is not used to index any array.  

Going further, even in the cases where the input parameter is an integer, I have not encountered any issues among two popular commercially simulator vendors (and 3 different simulators).  
## Test Case
[overload_index_issue.txt](https://github.com/tgingold/ghdl/files/79482/overload_index_issue.txt)
## Impact
OSVVM has numerous functions that are overloaded to return either integer or integer_vector.  This is going to create a confusing issue for the user community - particularly since other tools do not have an issue with it.

diogratia, 2016-01-06:
Because I really didn't want to go poking through OSVVM 2015-03 and I wanted to be able to abstract the issue, searching in CoveragePkg.vhd I found:
```vhdl
    impure function RandCovPoint return integer ;
    impure function RandCovPoint ( PercentCov : real ) return integer ;
    impure function RandCovPoint return integer_vector ;
    impure function RandCovPoint ( PercentCov : real ) return integer_vector ;
    impure function RandCovPoint (AtLeast : integer ) return integer ;
    impure function RandCovPoint (AtLeast : integer ) return integer_vector ;
```
So I created a reproducer not dependent on OSVVM:
```vhdl
package fum is 
    type fie is protected 
        impure function foo return integer;
        impure function foo(input: real) return integer;
        impure function foo return integer_vector;
        impure function foo (input: real) return integer_vector;
        impure function foo (input: integer) return integer;
        impure function foo(input: integer) return integer_vector;
        
    end protected fie;
end package;

package body fum is
    type fie is protected body
        variable answer:    integer := 42;
    
        impure function foo return integer is
        begin
            return answer;
        end;
        
        impure function foo(input:real) return integer is
        begin
            return integer(input) + answer;
        end;
        
        impure function foo return integer_vector is 
            variable conv_vector: integer_vector (0 to 1);
        begin
            conv_vector(0) := answer;
            conv_vector(1) := 0;
            return conv_vector;
        end; 
        
        impure function foo (input: real) return integer_vector is
            variable conv_vector: integer_vector (0 to 1);
        begin
            conv_vector(0) := integer(input) + answer;
            conv_vector(1) := 0;
        end;
        
        impure function foo (input: integer) return integer is
        begin
            return answer + input;
        end;
        
        impure function foo(input: integer) return integer_vector is 
            variable conv_vector: integer_vector (0 to 1);
        begin
            conv_vector(0) := input + answer;
            conv_vector(1) := 0;
            return conv_vector;
        end; 
    end protected body fie;
end package body;

--------

use work.fum.all;

entity overload_index_issue is
end entity overload_index_issue ;
architecture dumbit_down of overload_index_issue is
    shared variable fee: fie;
begin

    process
        variable Data : integer ; 
    begin
        Data := fee.foo(120.0); 
        wait;    
    end process; 
end architecture dumbit_down; 
```
And that graciously reproduced the error:

>ghdl -a --std=08 integer_ambig.vhdl
>integer_ambig.vhdl:70:24: can't resolve overload for function call, slice or indexed name
>integer_ambig.vhdl:70:24: possible interpretations are:
>integer_ambig.vhdl:4:25: function foo [real return integer]
>integer_ambig.vhdl:70:24: indexed name
>ghdl: compilation error

Now for the fun part. I commented out the functions foo returning an integer_vector and it analyzed successfully. (And this demonstrates the value of a small reproducer).

U-commented the `impure function foo return integer_vector;` and got the failure again.

Commented that one out and um-commented the other two and it analyzes. 

The problem has to do with the lack of arguments and the return type.  (And as Jim noted a real literal shouldn't be eligible for use as a discrete index).

tgingold, 2016-01-07:
On 07/01/16 00:29, Dio Gratia wrote:

> Now for the fun part. I commented out the functions foo returning an
> integer_vector and it analyzed successfully. (And this demonstrates the
> value of a small reproducer).
>
> U-commented the |impure [...]

Thanks, I will investigate from that.

Tristan.

tgingold, 2016-01-07:
I am testing a fix for this issue.

Tristan.

JimLewis, 2016-01-07:
From a point of practicality, for some reason other simulators do not have issues with integer parameters in this situation.  Not sure why.  Have you looked at the VHDL-87 interpretations?  While it may not be detailed in the LRM, the interpretation may be in there.  Unfortunately I do not have a copy of that document.

tgingold, 2016-01-07:
This issue is now fixed.

closed 2016-01-07 by tgingold