ghdl/ghdl #20: Type Real:  Write, Read, Write produces a different number

JimLewis, 2016-01-07:
##  Issue
While writing this, I found something relate that may be the root cause, but I am going to post it separately.  See integer to real conversions

While we expect floating point to be handled differently on different machines and different tools.  OTOH, when working with a single tool, one would expect the number to be relatively stable.  I was checking out some of the file writing capability of OSVVM.  I would write out a coverage database file, read it back in, and write it back out.  Then have a separate function that compares the two files, expecting them to be equal.  

Unfortunately something in GHDL is causing it to produce a different number.  

## Test Case
[fp_write_read_issue.txt](https://github.com/tgingold/ghdl/files/81419/fp_write_read_issue.txt)

## Observations
Other simulators look like they may be rounding while printing real numbers.  

## Impact
Low.  I am looking into changes in my testbenches so I can avoid this issue.

JimLewis, 2016-01-07:
On further inspection, I think the issue may be with printing, which is what this addresses and not internal number representation.  So rather than submit a separate issue, I will add a test case to this one.

This test case examines values 30 to 35, converts them to real, prints them, then truncates them and converts them back to integer and prints them.  On my build, 33 and 34 do not print nicely as a real, however when truncated and printed as an integer, they print as 33 and 34.
[integer_2_real.txt](https://github.com/tgingold/ghdl/files/81491/integer_2_real.txt)

tgingold, 2016-01-10:
I can reproduce the issue with the second test case.  Will fix.

Thanks!

vicencb, 2016-07-28:
Hello,
related to floating point rounding there is this error:
```vhdl
entity fp_err is begin
	assert 0.56  hr  = 2016 sec;
	assert 0.565 hr  = 2034 sec;
	assert 0.56  hr /= 2016000000000000256 fs;
	assert 0.565 hr /= 2033999999999999744 fs;
end entity;
architecture none of fp_err is begin end architecture;
```
The four assertions fail and they should not.
If this is unrelated to this bug, then I can open a new one if you want.

Regards,
  Vicenç.

Paebbels, 2016-07-29:
0.56 hours are not 2016 seconds.

Every physical value is defined as a multiple of the base unit (fs). 0.56 is a floating point physical unit literal, which cannot be expressed as a fixed-point number with a fixed number of fractional digits. So when 0.56 is converted to femto-seconds, the following conversion is performed:

```
a = (double) 0.56
# a = 0x3fe1eb851eb851ec
b = a * 60**2 * 1000**5
# b = 0x43bbfa45519b3001
# b = 2.0160000000000003e+18
c = (int) b
#           sec  ms  us  ns   ps    fs
# c = 2 016 000 000 000 000 256
```

So due to the limited precision of floating point data types, 0.56 hours, which we expect to be 2016 seconds, equals 2016 sec plus 256 fs. This is a common problem to all floating point numbers and can be addressed by introducing an [epsilon boundary](https://en.wikipedia.org/wiki/Machine_epsilon).

commit 391c91e9 (2017-04-21): Add testcase for #20

Close #20
closed 2017-04-21 by tgingold