ghdl/ghdl #21: libbacktrace.a not installed with make install

kraigher, 2016-01-08:
When building GHDL with libbacktrace.a to enable back traces using the LLVM backend the file is not copied to the install directory. Thus when simulating the following error occurs:
```console
gcc: error: /opt/ghdl/lib/ghdl/libbacktrace.a: No such file or directory
```

kraigher, 2016-01-08:
Also even when manually copying the file to the install location I get no backtraces.

tgingold, 2016-01-08:
On 08/01/16 20:09, kraigher wrote:
> Also even when manually copying the file to the install location I get
> no backtraces.

Do you use -g when analyzing ?

kraigher, 2016-01-08:
No I did not at first. Adding the -g flag when analyzing makes no difference though.

tgingold, 2016-01-08:
On 08/01/16 20:36, kraigher wrote:
> No I did not at first. Adding the -g flag when analyzing makes no
> difference though.

What is your testcase ?  The backtrace is displayed only in case of
fatal assertion or error.

Tristan.

kraigher, 2016-01-08:
**bug.vhd**
```vhdl
package pkg is
  procedure fail;
end package;

package body pkg is
  procedure fail_inner is
  begin
    assert false;
  end procedure;

  procedure fail is
  begin
    fail_inner;
  end procedure;
end package body;

use work.pkg.all;

entity ent is
end entity;

architecture a of ent is
begin
  main : process
  begin
    fail;
    report "after fail";
  end process;
end architecture;
```

**test.sh**
```shell
set -ex
ghdl -a -g bug.vhd
ghdl --elab-run ent --assert-level=error
```

**console output**
```console
sh test.sh 
+ ghdl -a -g bug.vhd
+ ghdl --elab-run ent --assert-level=error
bug.vhd:8:5:@0ms:(assertion error): Assertion violation
./ent:error: assertion failed
./ent:error: simulation failed
```

kraigher, 2016-01-08:
Built using the following configure:
```
./configure --with-llvm-config=/usr/bin/llvm-config-3.5 --prefix=/opt/ghdl --with-backtrace-lib=/usr/lib/gcc/x86_64-linux-gnu/4.8/libbacktrace.a
```

tgingold, 2016-01-08:
On 08/01/16 21:03, kraigher wrote:
> Built using the following configure:
>
> |./configure --with-llvm-config=/usr/bin/llvm-config-3.5
> --prefix=/opt/ghdl
> --with-backtrace-lib=/usr/lib/gcc/x86_64-linux-gnu/4.8/libbacktrace.a |

That should do it.

What is the output of:
ghdl --elab-run -v ent --assert-level=error

What is the content in the build tree of grt-backtraces-impl.ads ?

Tristan

kraigher, 2016-01-08:
**test.sh with -v**
```console
sh test.sh 
+ ghdl -a -g bug.vhd
+ ghdl --elab-run -v ent --assert-level=error
/opt/ghdl/bin/ghdl1-llvm -P/opt/ghdl/lib/ghdl/v93/std/ -P/opt/ghdl/lib/ghdl/v93/ieee/ --elab ent -l e~ent.lst -c -o e~ent.o e~ent
/usr/bin/gcc -o ent e~ent.o /opt/ghdl/lib/ghdl/v93/std/std_standard.o bug.o /opt/ghdl/lib/ghdl/libgrt.a -ldl -lm -Wl,--version-script=/opt/ghdl/lib/ghdl/grt.ver -Wl,--export-dynamic /opt/ghdl/lib/ghdl/libbacktrace.a -L./ -lz
./ent --assert-level=error
bug.vhd:8:5:@0ms:(assertion error): Assertion violation
./ent:error: assertion failed
./ent:error: simulation failed
```

**grt-backtraces-impl.ads**
```ada
with Grt.Backtraces.Jit;
package Grt.Backtraces.Impl renames Grt.Backtraces.Jit;
```

tgingold, 2016-01-08:
On 08/01/16 21:10, kraigher wrote:
> *test.sh with -v*
>
> sh test.sh
> + ghdl -a -g bug.vhd
> + ghdl --elab-run -v ent --assert-level=error
> /opt/ghdl/bin/ghdl1-llvm -P/opt/ghdl/lib/ghdl/v93/std/
> -P/opt/ghdl/lib/ghdl/v93/ieee/ --elab ent -l e~ent.lst -c -o e~ent.o e~ent
> /usr/bin/gcc -o ent e~ent.o /opt/ghdl/lib/ghdl/v93/std/std_standard.o
> bug.o /opt/ghdl/lib/ghdl/libgrt.a -ldl -lm
> -Wl,--version-script=/opt/ghdl/lib/ghdl/grt.ver -Wl,--export-dynamic
> /opt/ghdl/lib/ghdl/libbacktrace.a -L./ -lz
> ./ent --assert-level=error
> bug.vhd:8:5:@0ms:(assertion error): Assertion violation
> ./ent:error: assertion failed
> ./ent:error: simulation failed
>
> *grt-backtraces-impl.ads*
>
> with  Grt.Backtraces.Jit;
> package  Grt.Backtraces.Impl  renames  Grt.Backtraces.Jit;

That's not correct: this should be Grt.Backtraces.Gcc

Try to remove grt-backtraces-impl.ads and to rebuild.  Check the content 
of this file.

I suppose you have first configured without --with-backtrace-lib and 
then with this switch.  And apparently the file was not regenerated...

Tristan.

kraigher, 2016-01-08:
> I suppose you have first configured without --with-backtrace-lib and 
then with this switch.  And apparently the file was not regenerated...

Yes I did run configure without `--with-backtrace-lib` at some time before using the flag. So the file was apparently not generated when adding the flag. I did `make clean` before building again though.

I removed the file and ran configure again. Now the file contains:
```ada
with Grt.Backtraces.Gcc;
package Grt.Backtraces.Impl renames Grt.Backtraces.Gcc;
```

I still get no backtraces however:
```console
sh test.sh 
+ ghdl -a -g bug.vhd
+ ghdl --elab-run -v ent --assert-level=error
/opt/ghdl/bin/ghdl1-llvm -P/opt/ghdl/lib/ghdl/v93/std/ -P/opt/ghdl/lib/ghdl/v93/ieee/ --elab ent -l e~ent.lst -c -o e~ent.o e~ent
/usr/bin/gcc -o ent e~ent.o /opt/ghdl/lib/ghdl/v93/std/std_standard.o bug.o /opt/ghdl/lib/ghdl/libgrt.a -ldl -lm -Wl,--version-script=/opt/ghdl/lib/ghdl/grt.ver -Wl,--export-dynamic /opt/ghdl/lib/ghdl/libbacktrace.a -L./ -lz
./ent --assert-level=error
bug.vhd:8:5:@0ms:(assertion error): Assertion violation
./ent:error: assertion failed
./ent:error: simulation failed
```

tgingold, 2016-01-09:
Can you debug ?

If so, set a breakpoint on backtrace_create_state and backtrace_pcinfo.
Check that they are called and their return value.

I am puzzled at that point.

Tristan.

kraigher, 2016-01-09:
There are no functions named \*backtrace\* in the symbol table when debugging with gdb:
```
(gdb) info functions .*backtrace.*
All functions matching regular expression ".*backtrace.*":
```

kraigher, 2016-01-09:
Ok I am probably debugging the wrong thing... wait a moment.

kraigher, 2016-01-09:
Now I am doing the right thing. `backtrace_create_state` is called once first then `backtrace_pcinfo` is called four times. 
```console
sh test.sh 
+ ghdl -a -g bug.vhd
+ ghdl -e ent
+ gdb --args ./ent --assert-level=error
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./ent...done.
(gdb) info functions .*backtrace.*
All functions matching regular expression ".*backtrace.*":

File /home/kraigher/repo/ghdl/src/grt/grt-backtraces-gcc.adb:
function grt.backtraces.gcc.cb (a1: system.address; a2: system.address; a3: system.address; a4: integer; a5: system.address) return integer;
procedure grt.backtraces.gcc.error_cb (a1: system.address; a2: system.address; a3: integer);
function grt.backtraces.gcc.symbolizer (a1: system.address; a2: system.address; a3: natural; a4: system.address) return RETURN;

File /home/kraigher/repo/ghdl/src/grt/grt-backtraces.adb:
function grt.backtraces.address_image (a1: grt.types.integer_address) return grt.backtraces.address_image_string;
function grt.backtraces.backtrace_option (a1: string) return boolean;
procedure grt.backtraces.demangle_err (a1: grt.types.ghdl_c_string);
procedure grt.backtraces.demangle_op_err (a1: character; a2: character);
function grt.backtraces.file_basename (a1: grt.types.ghdl_c_string) return grt.types.ghdl_c_string;
function grt.backtraces.is_eq (a1: grt.types.ghdl_c_string; a2: string) return boolean;
procedure grt.backtraces.put_err_backtrace (a1: <ref> grt.errors.backtrace_addrs);
procedure grt.backtraces.register;

File src/grt/config/jumps.c:
procedure grt_save_backtrace (a1: access backtrace_addrs; a2: int);

Non-debugging symbols:
0x0000000000401e00  backtrace@plt
0x0000000000486010  backtrace_pcinfo
0x0000000000486070  backtrace_syminfo
0x00000000004860d0  backtrace_open
0x0000000000486180  backtrace_close
0x00000000004861d0  backtrace_create_state
0x0000000000486fd0  backtrace_initialize
0x0000000000487120  backtrace_get_view
0x00000000004871d0  backtrace_release_view
0x0000000000487220  backtrace_free
0x0000000000487270  backtrace_alloc
0x0000000000487400  backtrace_vector_grow
0x0000000000487510  backtrace_vector_finish
0x0000000000487530  backtrace_vector_release
0x000000000048a590  backtrace_dwarf_add
(gdb) break backtrace_pcinfo
Breakpoint 1 at 0x486010
(gdb) break backtrace_create_state
Breakpoint 2 at 0x4861d0
(gdb) run
Starting program: /home/kraigher/repo/ghdl_bugs/backtrace/ent --assert-level=error
bug.vhd:8:5:@0ms:(assertion error): Assertion violation
/home/kraigher/repo/ghdl_bugs/backtrace/ent:error: assertion failed

Breakpoint 2, 0x00000000004861d0 in backtrace_create_state ()
(gdb) info locals
No symbol table info available.
[... 95 lines cut ...]
#0  0x0000000000486010 in backtrace_pcinfo ()
#1  0x000000000047fc27 in grt.backtraces.gcc.symbolizer (pc=(system.address) 0x40ba95, filename=(system.address) 0x0, lineno=0, subprg=(system.address) 0x0)
    at /home/kraigher/repo/ghdl/src/grt/grt-backtraces-gcc.adb:123
#2  0x0000000000422574 in grt.backtraces.put_err_backtrace (bt=...) at /home/kraigher/repo/ghdl/src/grt/grt-backtraces.adb:238
#3  0x0000000000404f5f in grt.errors.error_e_call_stack (bt=...) at /home/kraigher/repo/ghdl/src/grt/grt-errors.adb:267
#4  0x00000000004039fb in grt.lib.do_report (msg=..., str=0x0, default_str=..., severity=2, loc=0x48cbc0 <_UI00000001>) at /home/kraigher/repo/ghdl/src/grt/grt-lib.adb:84
#5  0x0000000000403a80 in <__ghdl_assert_failed> (str=0x0, severity=2, loc=0x48cbc0 <_UI00000001>) at /home/kraigher/repo/ghdl/src/grt/grt-lib.adb:92
#6  0x0000000000402c7f in work__pkg__fail_inner (PARAMS=<optimized out>) at /home/kraigher/repo/ghdl_bugs/backtrace/bug.vhd:8
#7  0x0000000000402bbb in work__pkg__fail (PARAMS=<optimized out>) at /home/kraigher/repo/ghdl_bugs/backtrace/bug.vhd:13
#8  0x0000000000402d35 in work__ent__ARCH__a__main__PROC (INSTANCE=<optimized out>) at /home/kraigher/repo/ghdl_bugs/backtrace/bug.vhd:26
#9  0x000000000040ba95 in grt.processes.run_processes (postponed=false) at /home/kraigher/repo/ghdl/src/grt/grt-processes.adb:771
#10 0x000000000040bd02 in grt.processes.initialization_phase () at /home/kraigher/repo/ghdl/src/grt/grt-processes.adb:835
#11 0x000000000040c0ec in grt.processes.simulation_init () at /home/kraigher/repo/ghdl/src/grt/grt-processes.adb:1048
#12 0x000000000040c31c in grt.processes.simulation () at /home/kraigher/repo/ghdl/src/grt/grt-processes.adb:1132
#13 0x000000000041f422 in __ghdl_run_through_longjump (func=0x40c30f <grt.processes.simulation>) at src/grt/config/jumps.c:232
#14 0x000000000040568b in grt.main.run () at /home/kraigher/repo/ghdl/src/grt/grt-main.adb:174
#15 0x000000000042a404 in ghdl_main (argc=2, argv=(system.address) 0x7fffffffdf58) at /home/kraigher/repo/ghdl/src/grt/ghdl_main.adb:59
#16 0x000000000041f6b4 in main (argc=2, argv=(system.address) 0x7fffffffdf58) at src/grt/main.adb:31
(gdb) continue
Continuing.
/home/kraigher/repo/ghdl_bugs/backtrace/ent:error: simulation failed
[Inferior 1 (process 796) exited with code 01]
(gdb) 

```

kraigher, 2016-01-09:
When debugging further it seems that Backtraces.Impl.Symbolizer returns null values causing Put_Err_Backtrace to print nothing:

gdb locals in Put_Err_Backtrace after Backtraces.Impl.Symbolizer call
```
filename = (system.address) 0x0
lineno = 0
subprg = (system.address) 0x0
unknown = true
```

tgingold, 2016-01-10:
On 09/01/16 10:08, kraigher wrote:
> When debugging further it seems that Backtraces.Impl.Symbolizer returns
> null values causing Put_Err_Backtrace to print nothing:
>
> gdb locals in Put_Err_Backtrace after Backtraces.Impl.Symbolizer call
>
> |filename = (system.address) 0x0 lineno = 0 subprg = (system.address) 0x0
> unknown = true |

 From which version of gcc are you using libbacktrace ? I have tested 
with gcc-4.9.2

You need to use -g both at analysis and elaboration.

The last solution is to debug backtrace_pcinfo.
(You can first set a breakpoint on Error_Cb which is called on errors).

Tristan.

tgingold, 2016-07-03:
I was not able to reproduce the issue.  Works for me.
Please reopen if you have new infos.

closed 2016-07-03 by tgingold