ghdl/ghdl #22: How to build ghdl minus libraries?

imd1, 2016-01-17:
Hi Tristan

I'm attempting to cross native build ghdl with gcc backend (Cygwin build, Mingw host/target), and make errors when trying to build the std/ieee libraries. In order to work around this temporarily, I'd like to omit everything library related so make/make install completes to give me an executable, and then try compiling the libraries in Windows using a batch file similar to the mcode winbuild.bat, to at least check the executable works.  I'm a configure/make newbie, so I was wondering if you have any advice.

BTW, I've successfully built from source the following configurations:
GHDL Mcode under windows
Cygwin native build of GHDL with GCC backend.
Cross native build of GCC without GHDL included

Many thanks
Ian

Paebbels, 2016-01-17:
Hello Ian,

makefiles are made of targets and rules to build a target. A common shell like bash have tab-completion for makefiles, to show all possible targets.

The default target is `all`, which builds ghdl and pre-compiles the IEEE libraries. I think you are looking for the target `ghdl_gcc`:
`~/> make ghdl_gcc`

You wrote that you build ghdl with batch files on Windows. I provided the PowerShell translation for these old batch scripts. Do you mind testing them? PowerShell is included in every Windows since XP :).

```PowerShell
PS> .\winbuild.ps1 -Compile
PS> # for a detailed help
PS> .\winbuild.ps1 -Help
```

Regards
    Patrick

tgingold, 2016-01-18:
On 17/01/16 22:57, Patrick Lehmann wrote:
> Hello Ian,
>
> makefiles are made of targets and rules to build a target. A common
> shell like bash have tab-completion for makefiles, to show all possible
> targets.
>
> The default target is |all|, which builds ghdl and pre-compiles the IEEE
> libraries. I think you are looking for the target |ghdl_gcc|:
> |~/> make ghdl_gcc|
>
> You wrote that you build ghdl with batch files on Windows. I provided
> the PowerShell translation for these old batch scripts. Do you mind
> testing them? PowerShell is included in every Windows since XP :).
>
> PS >  .\winbuild.ps1  -Compile
> PS >  # for a detailed help
> PS >  .\winbuild.ps1  -Help

Yes, Patrick is correct: there are Windows script to build the libraries.
So all you need for a cross build is to build the compiler.

As the libraries are built after the compiler, a standard
gcc+ghdl cross native build should almost do it.
However, it was never tested.  There are certainly some nits
to fix in the Makefiles.

So I don't have any particular advices, except try it!

Regards,
Tristan.

imd1, 2016-01-18:
Hi Patrick

Here's a snippet from the library failure:

../../../../ghdl -a --GHDL1=./ghdl1 --ieee=none -O2 -g --std=87 --bootstrap --work=std ../../src/std/textio.v87
C:\Users\user\Desktop\ghdl\build\gcc\gcc\ghdl.exe: installation problem: ./ghdl1 not found
Makefile:301: recipe for target 'std.v87' failed
make[3]: *** [std.v87] Error 1
make[3]: Leaving directory '/cygdrive/c/Users/user/Desktop/ghdl/build/gcc/gcc/vhdl'
../../../source/gcc-4.9.3/gcc/vhdl/Make-lang.in:129: recipe for target 's-ghdllib' failed
make[2]: *** [s-ghdllib] Error 2
make[2]: Leaving directory '/cygdrive/c/Users/user/Desktop/ghdl/build/gcc/gcc'
Makefile:3990: recipe for target 'all-gcc' failed
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory '/cygdrive/c/Users/user/Desktop/ghdl/build/gcc'
Makefile:874: recipe for target 'all' failed
make: *** [all] Error 2

The ghdl1.exe is present in the same folder as ghdl.exe.

imd1, 2016-01-18:
Hi

I've found the targets in gcc/vhdl/Makefile and hacked out the lines that perform the library analysis. I've yet to find a better way of achieving this by specifying one or more make targets instead of "all"

Regards
Ian

imd1, 2016-01-18:
Hi

A similar hack to gcc/vhdl/Makefile  allows make install to complete.

Regards
Ian

tgingold, 2016-01-19:
On 18/01/16 13:45, imd1 wrote:
> Hi
>
> A similar hack to gcc/vhdl/Makefile allows make install to complete.

So, you were able to build ghdl+gcc with cygwin ?

Could you post your patches, as some people could be interested.

Thanks,
Tristan.

imd1, 2016-01-19:
Too much of a hack at this stage - not happy with it for various reasons.

To clarify, no hacks are needed for a native Cygwin build, hacks are only required when cross native compiling to mingw

To summarise, I still have no way of compiling GHDL minus the libraries

tgingold, 2016-02-05:
With gcc, you can do 'make ghdl1' to just build the compiler.  Look at the Make-lang.in in vhdl/

imd1, 2016-02-13:
Which directory should I be in when I invoke make ghdl1?

tgingold, 2016-02-14:
On 13/02/16 09:42, imd1 wrote:
> Which directory should I be in when I invoke make ghdl1?

In gcc/ (of the object dir) you can do: make ghdl1.exe

Tristan.

imd1, 2016-02-16:
OK, I cd'ed to a folder where make ghdl1.exe found a target, and I get the following error:

user@user-THINK-1 /cygdrive/c/Users/user/Desktop/ghdl/build/gcc/gcc
$ make ghdl1.exe
x86_64-w64-mingw32-g++ -c -DIN_GCC     -DGENERATOR_FILE -I. -Ibuild -I../../../source/gcc-4.9.3/gcc -I../../../source/gcc-4.9.3/gcc/build -I../../../source/gcc-4.9.3/gcc/../include -I./../intl -I../../../source/gcc-4.9.3/gcc/../libcpp/include  \
        -o build/genmddeps.o ../../../source/gcc-4.9.3/gcc/genmddeps.c
In file included from ../../../source/gcc-4.9.3/gcc/genmddeps.c:19:0:
../../../source/gcc-4.9.3/gcc/system.h:358:22: fatal error: sys/wait.h: No such file or directory
 #include <sys/wait.h>
                      ^
compilation terminated.
Makefile:2366: recipe for target 'build/genmddeps.o' failed
make: *** [build/genmddeps.o] Error 1

tgingold, 2016-02-17:
On 16/02/16 11:15, imd1 wrote:
> OK, I cd'ed to a folder where make ghdl1.exe found a target, and I get
> the following error:
>
> user@user-THINK-1 /cygdrive/c/Users/user/Desktop/ghdl/build/gcc/gcc
> $ make ghdl1.exe
> x86_64-w64-mingw32-g++ -c -DIN_GCC -DGENERATOR_FILE -I. -Ibuild
> -I../../../source/gcc-4.9.3/gcc -I../../../source/gcc-4.9.3/gcc/build
> -I../../../source/gcc-4.9.3/gcc/../include -I./../intl
> -I../../../source/gcc-4.9.3/gcc/../libcpp/include \
> -o build/genmddeps.o ../../../source/gcc-4.9.3/gcc/genmddeps.c
> In file included from ../../../source/gcc-4.9.3/gcc/genmddeps.c:19:0:
> ../../../source/gcc-4.9.3/gcc/system.h:358:22: fatal error: sys/wait.h:
> No such file or directory
> #include
> ^
> compilation terminated.
> Makefile:2366: recipe for target 'build/genmddeps.o' failed
> make: *** [build/genmddeps.o] Error 1

genmddeps.o is a build tool and therefore should be built using
a native compiler (and not a cross one).

How have you configured gcc ?

imd1, 2016-02-25:
The previous post happened when I ran make ghdl1.exe on a previously successful make all with hacks included.  When I run make ghdl1.exe from a clean state, make cannot find the target.  I suspect one or more make targets must be run first before running make ghdl1.exe? If so, I need to know the list.

tgingold, 2016-02-26:
On 25/02/16 15:22, imd1 wrote:
> The previous post happened when I ran make ghdl1.exe on a previously
> successful make all with hacks included. When I run make ghdl1.exe from
> a clean state, make cannot find the target. I suspect one [...]

It's hard to know in which state you are.

When you configure gcc, it simply creates a few files.  The top-level
make configure several subdirectories (one of them is gcc) and sub
Makefiles are created.

After that, if you go in the gcc/ directory, 'make ghdl1.exe' should
work.  Unless it is 'make ghdl'.  You have to look at the Makefile
(I know, it's big) for more info, as I have never configured gcc
in cross mode.

Regards,
Tristan.

imd1, 2016-02-27:
Hi Tristan
I will go back and try out a few ideas, and will also return to using a "native mode" so you can reproduce my results more easily. Leave it with me.
Regards
Ian

closed 2016-07-03 by tgingold