ghdl/ghdl #23: Jumps.c errors when cross native compiling to Mingw

imd1, 2016-01-19:
I'm attempting to cross native build ghdl with gcc backend (Cygwin build, Mingw host/target), and make errors when trying to compile jumps.c. The first error involves siginfo_t not being defined.  Googling reveals that mingw may not support siginfo_t.  In order to work around this temporarily, I've deleted the contents of jumps.c, and the build proceeds, but I suspect I may hit a run time issue at some point.

Can anyone help with a) a way of keeping the content of jumps.c and b) whether omitting its content will work going forwards

tgingold, 2016-01-19:
On windows, src/grt/config/win32.c should be used instead of src/grt/config/jumps.c

Tristan.

closed 2016-01-19 by tgingold

imd1, 2016-01-20:
Presumably this means that there is a bug in a make file?

tgingold, 2016-01-20:
On 20/01/16 10:32, imd1 wrote:
> Presumably this means that there is a bug in a make file?

Yes.  Cross-compilation was never tried and may require
some adjustments (in particular in src/grt/Makefile.inc).

Tristan.