ghdl/ghdl #24: GHDL compiler error

igmar, 2016-01-19:
I get this : 

******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
In directory: /Users/igmar/SpiderOak Hive/msx3/vhdl/
Command line:
/usr/local/bin/ghdl1-llvm --workdir=/Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/z80 --work=z80 --std=08 -P/Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/vunit_lib -P/Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/osvvm -P/Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/z80 -P/Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/vunit_lib/ -P/Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/osvvm/ -P/Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/z80/ -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/ieee/ -c -o /Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/ghdl/libraries/z80/tb_z80_noop.o /Users/igmar/SpiderOak Hive/msx3/vhdl/vunit_out/preprocessed/z80/tb_z80_noop.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: no field Named_Entity
Load address: 0x1038fb000
Call stack traceback locations:
0x1043bd791 0x103929ab5 0x103aaca8e 0x103acb458 0x103acc872 0x103a96756 0x103ace0a1 0x103ad62ac 0x103a2e88e 0x103a86fec 0x10393bc0e 0x1038fccda
******************************************************************
ghdl: compilation error

tgingold, 2016-01-19:
Could you send us your design or at least a reproducer.
Thanks in advance.

bwimm, 2016-05-31:
Hi,
I encountered the same error message.
```
******************** GHDL Bug occured `****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 4.6
In directory: /home/foo/Desktop/ghdl_crash/
Command line:
ghdl -a --std=08 crash.vhd -r test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: no field Named_Entity
Call stack traceback locations:
0x7f46b70da44a 0x4aa1d3 0x625e0a 0x64396e 0x6448e6 0x6102ba 0x610fd4 0x61131f 0x6115f2 0x64623a 0x64dab8 0x548825 0x6001b6 0x5fe2bd 0x4bce5d 0x4082fd 0x7f46b68fff43 0x407907 0xfffffffffffffffe
******************************************************************
```

The reduced file is attached.
[crash.vhd](https://github.com/tgingold/ghdl/files/290937/crash.vhd.txt)

tmeissner, 2016-05-31:
@bwimm 
I'm pretty sure that hierarchical names are not supported by GHDL at the moment. Anyway, their use shouldn't cause a GHDL assertion failure.

tgingold, 2016-05-31:
On 31/05/16 14:44, T. Meissner wrote:
> @bwimm <https://github.com/bwimm>
> I'm pretty sure that hierarchical names are not supported by GHDL at the
> moment. Anyway, their use shouldn't cause a GHDL assertion failure.

Yes, they aren't yet supported, but there is work in progress.

tmeissner, 2016-05-31:
> Yes, they aren't yet supported, but there is work in progress.
Oh, cool, nice to hear :)

herbertp, 2017-02-14:
Here is a way shorter example which causes the same GHDL Bug.
[top.vhd.txt](https://github.com/tgingold/ghdl/files/773239/top.vhd.txt)

Best,
Herbert

lainy, 2017-04-20:
Looking forward to this to simplify some testbenches.

tgingold, 2017-04-20:
On 20/04/17 07:13, Lain wrote:
> Looking forward to this to simplify some testbenches.

I don't forget it.  But this will require a redesign of the elaboration.
I have some ideas for that.

Tristan.

Xiretza, 2021-01-17:
Would this be a candidate for [v1.0](https://github.com/ghdl/ghdl/milestone/9)?

eine, 2021-01-17:
@Xiretza, I normally use the v1.0 milestone post-mortem. You will find that most issues are labeled but not milestoned until Tristan actually pushes some fix and closes it. That is, the milestone represents what was done this last year, not necessarily what is to be done. Apart from that, I did milestone some VHDL 2008 which are very demanded. Therefore, I will milestone this now, but be aware that doesn't imply any real change in priority.

Xiretza, 2021-01-17:
Understood, thanks!

nathanaelhuffman, 2021-03-26:
This feature (external/hierarchical names) is the only thing I'm aware of that is preventing me from using GHDL regularly. I have a testbench/testharness split design pattern where the testharness entity has no signals and grabbing a couple of testharness signals up in the testbench is pretty convenient, so count me as a +1.

JimLewis, 2021-05-15:
> Here is a way shorter example which causes the same GHDL Bug.
> [top.vhd](https://github.com/tgingold/ghdl/files/773239/top.vhd.txt)
> 
> Best,
> Herbert

@herbertp   Your usage there of an external name is incorrect.   The object you reference in the external name must be elaborated before the external name.   In your example, you would have to change it to:

```vhdl
library IEEE;
use IEEE.std_logic_1164.all;

entity top is
    port (
	data : out std_logic
    );
end entity top;

architecture RTL of top is
 --   alias int is << signal sub.intern : std_logic >>;
begin
    SUB_inst : entity work.sub;

    data <= << signal sub.intern : std_logic >> ;
end RTL;

```

goekce, 2021-12-22:
@JimLewis you are almost right, but you have to use the name of the instance instead of the entity AFAIK:

```
SUB_inst : entity work.sub;

data <= << signal SUB_inst.intern : std_logic >> ; 
```

I also hope that hierarchial names are integrated soon!

commit 07ccb535 (2023-09-18): testsuite/gna: add a test for #24

Close #24
closed 2023-09-18 by tgingold