ghdl/ghdl #25: TYPES.INTERNAL_ERROR is raised if overlapping physical units are declared.

Paebbels, 2016-01-23:
I made an error by defineing overlapping physical units. Unfortunately, GHDL handled this error with an exception.

This code should reproduce it:

    type ANGLE is range INTEGER'range units
      sec;
      min = 60 sec;
      deg = 60 min;
    end unit;

My code collides with **seco**nds and **min**utes from type TIME.

GHDL 0.34dev - build date 2016.01.20

*angles are measured in degree'minutes"seconds*

Paebbels, 2016-01-23:
The exception seams to be raised by `INTEGER'range`. If I use literals like 10000, then there is no exception.

tgingold, 2016-01-24:
On 23/01/16 19:49, Patrick Lehmann wrote:
> The exception seams to be raised by |INTEGER'range|. If I use literals
> like 10000, then there is no exception.

Yes.  Integer'Range is not valid.
I have fixed this issue, will push as soon as regtesting is ok.

Thanks!
Tristan.

Paebbels, 2016-01-24:
What about the second issue part?

When there are double used secondary units, GHDL reports non are found, if I rename one unit, GHDL finds exactly one unit (as expected).

tgingold, 2016-01-24:
On 24/01/16 14:11, Patrick Lehmann wrote:
> What about the second issue part?
>
> When there are double used secondary units, GHDL reports non are found,
> if I rename one unit, GHDL finds exactly one unit (as expected).

Can you provide a testcase.  I am not sure I understand correctly the
issue.

Tristan.

Paebbels, 2016-01-24:
I'm trying to find an small example, while doing so I found these 2 sub issues :).

The first invokes an exception while analysis. The second reports a strange message, because I can't see any error.

[1_SecondaryUnit.vhdl.txt](https://github.com/tgingold/ghdl/files/102516/1_SecondaryUnit.vhdl.txt)
[2_SecondaryUnit.vhdl.txt](https://github.com/tgingold/ghdl/files/102517/2_SecondaryUnit.vhdl.txt)

tgingold, 2016-01-25:
On 24/01/16 20:57, Patrick Lehmann wrote:
> I'm trying to find an small example, while doing so I found these 2 sub
> issues :).

Great!

> The first invokes an exception while analysis. The second reports a
> strange message, because I can't see any error.
>
> 1_SecondaryUnit.vhdl.txt [...]

Thanks.  I will push the fix soon.

Tristan.

tgingold, 2016-01-25:
Pushed.

closed 2016-01-25 by tgingold