ghdl/ghdl #26: Missing source code line number.

Paebbels, 2016-01-24:
I managed to calculate negative delay times, which causes GHDL to stop with an error.

> ghdl.exe:error: negative timeout clause
> ghdl.exe:error: simulation failed

Is it possible to report the VHDL source code line? I suspect it's in one of my procedures....

tgingold, 2016-01-25:
That should be doable...

Brian-Drummond, 2017-10-04:
I've had a brief look at this, and I can report the name of the erroneous process, but I can't offhand see how to get closer than that ( I don't know how source code locations are handled at runtime).

Here's a brief reproducer

```vhdl
entity Time_travel is
end Time_travel;

architecture Behavioral of Time_travel is

constant start_time : time := 10 ms; 
constant stop_time : time := 5 ms; 

procedure delay(t : time) is
begin
	wait for t;
end delay;

begin

McFly : process is
begin
        wait for start_time;
        report " Started " severity NOTE;        
	delay(stop_time - start_time);
--        wait for stop_time - start_time;
        report " Stopped " severity NOTE;   
        wait;
end process;
end Behavioral;
```
And two patches 27ea531 and 1f4105e at
https://github.com/Brian-Drummond/ghdl/commits/master
as a tentative fix.

tgingold, 2017-10-05:
The location of the wait statement can be passed to ghdl_process_wait_timeout and ghdl_process_wait_set_timeout.
(See how it is done in Gen_Program_Error: filename and line number are passed to the subprogram).

Brian-Drummond, 2017-10-05:
Thanks for the hints, I'll have another look.

closed 2017-10-06 by tgingold