ghdl/ghdl #27: Mingw run time error during VUnit example elaboration

imd1, 2016-01-25:
I'm attempting to cross native build ghdl with gcc backend (Cygwin build, Mingw host/target). Having replaced jumps.o with win32.o as Make targets, the build succeeds but when I run the VUnit user guide example, I get a run time exception (0xC0000005: Access violation reading location 0x00000000fffdf000.) during the elaboration step.

Difficult to debug, but I was wondering if there are other run time targets that should be include i.e. I currently have GRT_TARGET_OBJS=win32.o math.o clock.o.  Also, do I need to be using pthreads?

Any help would be appreciated.

tgingold, 2016-01-25:
Two advices:
- you should try with a very simple vhdl design
- compile your design with -g, and use gdb to get a backtrace.

imd1, 2016-01-25:
Thanks for the suggestions

Do you have a simple example I could use? This ensures it should work and gives me the command line as I have only used GHDL via VUnit.

tgingold, 2016-01-25:
On 25/01/16 21:01, imd1 wrote:
> Thanks for the suggestions
>
> Do you have a simple example I could use? This ensures it should work
> and gives me the command line as I have only used GHDL via VUnit.

You can try the hello world or full adder design from the doc:

http://ghdl.readthedocs.org/en/latest/Starting_with_GHDL.html#the-hello-world-program

Tristan.

imd1, 2016-01-25:
I compiled the hello world example and get the same run time error

The commands I used were:

ghdl -a --std=08 hello.vhd
ghld -e --std=08 hello_world
.\hello_world.exe

Note that I'm using 2008 as this is required by VUnit and I have only compiled the libraries for this standard so far.

I've uploaded my files (not including the ghdl exe):

[hello_world.zip](https://github.com/tgingold/ghdl/files/104014/hello_world.zip)

imd1, 2016-01-25:
Regarding the use of gdb.

I do not have gdb, so I could download the source code and cross native compile?

imd1, 2016-01-25:
I downloaded a gdb binary for windows and had some success using gdb backtrace

(gdb) run
Starting program: C:\Users\user\Desktop\ghdl\test\hello_world\hello_world.exe
[New Thread 8828.0x2604]

Program received signal SIGSEGV, Segmentation fault.
__ghdl_run_through_longjump (func=func@entry=0x40afd0 <grt.main.ghdl_elaborate_wrapper>)
    at ../../../../source/gcc-4.9.3/gcc/vhdl/grt/config/win32.c:114
114     ../../../../source/gcc-4.9.3/gcc/vhdl/grt/config/win32.c: No such file or directory.
(gdb) backtrace
#0  __ghdl_run_through_longjump (func=func@entry=0x40afd0 <grt.main.ghdl_elaborate_wrapper>)
    at ../../../../source/gcc-4.9.3/gcc/vhdl/grt/config/win32.c:114
#1  0x000000000040b0ac in grt.main.run ()
    at /cygdrive/c/Users/user/Desktop/ghdl/source/gcc-4.9.3/gcc/vhdl/grt/grt-main.adb:146
#2  0x00000000004179f2 in ghdl_main (argc=<optimized out>, argv=<optimized out>)
    at /cygdrive/c/Users/user/Desktop/ghdl/source/gcc-4.9.3/gcc/vhdl/grt/ghdl_main.adb:59
#3  0x0000000000454da0 in main (argc=<optimized out>, argv=<optimized out>)
    at ../../../../source/gcc-4.9.3/gcc/vhdl/grt/main.adb:31
#4  0x00000000004013e8 in __tmainCRTStartup ()
    at ../../../source/mingw-w64-v4.0.4/mingw-w64-crt/crt/crtexe.c:332
#5  0x000000000040151b in mainCRTStartup ()
    at ../../../source/mingw-w64-v4.0.4/mingw-w64-crt/crt/crtexe.c:212
(gdb)

I'm assuming that the seg fault occurred at line 114 in win32.c (?).Here's lines 113-114:

  /* Get current handler.  */
  asm ("mov %%fs:(0),%0" : "=r" (prev));

Some form of inline assembler?

imd1, 2016-01-25:
The target for all my builds is x86_64-w64-mingw32.  Is this target supported I wonder?

tgingold, 2016-01-26:
On 25/01/16 23:59, imd1 wrote:
> The target for all my builds is x86_64-w64-mingw32. Is this target
> supported I wonder?

No, however it shouldn't be hard to try this configuration.
Edit win32.c and comment out the x86 specific code:

int
__ghdl_run_through_longjump (int (*func)(void))
{
   int res;
#if 0
   struct exception_registration er;
   struct exception_registration *prev;

   /* Get current handler.  */
   asm ("mov %%fs:(0),%0" : "=r" (prev));

   /* Build regisration.  */
   er.prev = prev;
   er.handler = ghdl_SEH_handler;

   /* Register.  */
   asm ("mov %0,%%fs:(0)" : : "r" (&er));
#endif

   run_env_en = 1;
   res = setjmp (run_env);
   if (res == 0)
     res = (*func)();
   run_env_en = 0;

#if 0
   /* Restore.  */
   asm ("mov %0,%%fs:(0)" : : "r" (prev));
#endif

   return res;
}

Tristan.

imd1, 2016-01-26:
Tristan

Your suggestion worked for the hello world example....success :-)
I will try the VUnit example next.

Question: should the removed code be replaced by something else, or can I leave it just removed?

Ian

imd1, 2016-01-26:
I can confirm that the VUnit examples all work now.

tgingold, 2016-01-27:
On 26/01/16 23:28, imd1 wrote:
> Tristan
>
> Your suggestion worked for the hello world example....success :-)
> I will try the VUnit example next.
>
> Question: should the removed code be replaced by something else, or can
> I leave it just removed?

Ideally, it should be replaced by the corresponding win64 code.  But 
that code cannot be written in C :-(

I suppose I will add it to my TODO list.

Tristan.

tgingold, 2016-01-27:
On 26/01/16 23:47, imd1 wrote:
> I can confirm that the VUnit examples all work now.

Great, and congratulation!

tgingold, 2016-07-03:
Implemented by #108

closed 2016-07-03 by tgingold