ghdl/ghdl #28: Deep recursion detection

Paebbels, 2016-01-25:
**Suggested feature**

I implemented a method like this in a protected type:
```
type JSON is protected
  impure function HasError return BOOLEAN;
end protected;

type JSON is protected body
  variable pHasError : BOOLEAN := FALSE;
  impure function HasError return BOOLEAN is
  begin
    return HasError;
  end function;
end protected body;
```

The typo is at the return statement where the method is called recursively instead of `pHasError`.

I assume GHDL has a control and data flow analysis, so is it possible to detect (deep) recursions and emit a warning?

What happens in case of such a recursive call? GHDL.exe was killed by Windows :).

tgingold, 2016-01-26:
That's possible, but recursion is allowed.  So to produce meaningful messages, we need to be careful.

Paebbels, 2016-01-28:
I thought of a warning or note if a depth of e.g. 1000 is reached.

- Small recursions should not hit this limit. So there is no change in GHDLs behavior.
- Intended recursion are known by the user. He can:
    - ignore the warning or
    - disable it? or
    - set another limit?
- Not intended recursion can be detected, if the warning is written faster than the simulation crashes :).

Proposed message:

```
test.vhdl:123:45: warning: deep recursion detected
  test.vhdl:385: function myfunc1
  test.vhdl:236: function myfunc3
```

So maybe GHDL prints just the first proposed line and the call stack / the call cycle in a verbose mode.

This warning could also be applied the generate statements if a similar structure like the control flow graph exists.

tgingold, 2016-02-02:
The problem is that detecting recursions has a runtime impact.

On windows, it would be possible to catch the stack overflow and to display a better message.

Paebbels, 2016-02-04:
I don't know the internals of GHDL but I would like to list my thoughts on the problem. Maybe one could be a solution :).

Form my theoretical computer science course (I hope I remember correctly):
Edge classification (normal, forward, backward, sideward) and cycle detection in a directed graph is performed by graph coloring and topological sorting in the depth-first-search (DFS) tree.

Calling procedures or functions increases a depth counter for the current active call stack. I'm not sure how VHDL handles calls, but I think a new delta cycle is already created for each call, so the maximum depth would be bound by `MaxDeltaCycles`. Is this the same behavior for elab-time and runtime? I assume there are no delta cycles in elab-time, so this boundary does not exist.

How about tagging each procedure/function with a `CallCount` field, which is incremented by each call to the callable. So if the function is called twice, (uncalled function have `CallCount = 0`), it is increased and checked against the warning limit.

Has this increment plus check such an high performance impact?

I assume GHDL optimizes the code :), so how many function are implemented as real function, which need to be called in every simulation step? I could imaging that a decoder implemented as a function is transformed into a look-up-table/ROM. Do you have some kind of statistics how many functions are evaluated per cycle?

tgingold, 2016-02-08:
No, there is no delta cycle created by a function call.

At elaboration time, it could be easy to detect cycle/recursion and to report them.  However, it doesn't mean this will happen during simulation.

I am not ready to add one counter per function, that's too heavy.

And no, GHDL doesn't transform functions.  One routine is generated for each function.

closed 2016-02-11 by Paebbels