ghdl/ghdl #30: GHDL internal error

JohnRTC, 2016-01-28:
I have encountered a situation in which GHDL crashes out with an internal error.  Details are contained within the attached zip archive, which was created with the current zip utility on Ubuntu 14.04 LTS.  The archive contains source code, build files, and all messages GHDL emits before it crashes.

 I raised this issue yesterday on the mailing list.  The files expand to about 512k.  Unfortunately, I am unable to reproduce the error with smaller files. I did not include object files as these total about 4Mbytes,

Cheers
John
[files.zip](https://github.com/tgingold/ghdl/files/108825/files.zip)

tgingold, 2016-01-29:
Hello,

I am unable to reproduce the same crash using the development version of ghdl.
However, ghdl reports a bound check failure at tb-alu.vhdl:630.  The corresponding lines are:
	u33: magnitudeN
			generic map(
				N => 8
			)
			port map(
				a => counter_out(21 downto 0),
				b => x"3fffff",
				equal => done,
				lt => open,
				gt => open
			);
According to the declaration, length of both A and B should be 8.

Regards,
Tristan.

tgingold, 2016-01-29:
Hello,

I can reproduce the crash with ghdl 0.33, so the reason for the crash has been fixed.

I still have an error with the development version of ghdl, I am investigating it.

Thanks,
Tristan.

tgingold, 2016-01-29:
The second issue in the component instantiation is the B value.
The length of x"3fffff" is 6*4=24, while the length of B is 22.
You should write: 	b => (21 downto 0 => '1'),
or:                            b => (others => '1'),

Regards,
Tristan.

closed 2016-01-29 by tgingold