ghdl/ghdl #31: Another crash report

JohnRTC, 2016-01-31:
The attached code causes an internal error crash.  While fixing the compile-time errors stops the crashing, I am inclined to think compilers should not crash regardless of input, correct or otherwise.  The installed software is GHDL 0.33 (Dunoon edition) running on a current version of Ubuntu 14.04 LTS.  The VHDL file may or may not cause a development version to crash.

I'm not pushing, but when is the next scheduled release of GHDL?  While I have other things I can do, the other GHDL issue I reported (other than the coding error) is a bit of a show-stopper for me.

The files were deflated with zip.

Cheers
John

[files.zip](https://github.com/tgingold/ghdl/files/111069/files.zip)

tgingold, 2016-02-02:
Hello,

> The attached code causes an internal error crash While fixing the
> compile-time errors stops the crashing, I am inclined to think
> compilers should not crash regardless of input, correct or otherwise [...]

Yes, this crash was already fixed in the development version.

> I'm not pushing, but when is the next scheduled release of GHDL?
> While I have other things I can do, the other GHDL issue I reported
> (other than the coding error) is a bit of a show-stopper for me

I understand.
Is it possible for you to rebuild GHDL ?  On Ubuntu 14, this is
mainly a matter of configure;make;make install.

Currently, I have no way to automatically create and publish
binaries from travis-ci.  Ideas are welcome.

Tristan.

JohnRTC, 2016-02-02:
Thank you, Tristan.  I am willing to have a go at rebuilding GHDL.  Are there scripts that work on relative path names to do the work?  Where can I find the source code?

Perhaps you would consider posting development binaries at a reasonable interval. say once every two to four weeks.

tgingold, 2016-02-03:
On 02/02/16 21:55, JohnRTC wrote:
> Thank you, Tristan. I am willing to have a go at rebuilding GHDL. Are
> there scripts that work on relative path names to do the work? Where can
> I find the source code?

Clone the repository from github, and read README.md
Once the ada compiler is installed (apt-get install gnat), that should
be very easy.

> Perhaps you would consider posting development binaries at a reasonable
> interval. say once every two to four weeks.

I am considering that, but this should be automatic.  I don't want to
spend too much time on that.

Regards,
Tristan.

JohnRTC, 2016-02-03:
I obtained the zip file under the Download Zip button on the code page, compiled it and installed it.  Other than not signing myself in as root, it went without any issues.

Unfortunately, 0.34 did not produce any binary outputs (including .o files).  I did not see any messages indicating code was being analysed with the same (working) build files I used with 0.33.

ghdl -a filename

returns immediately with nothing other than a file of name work-obj93.cf produced.

I installed the new version of ghdl on top of 0.33.  Is it worth removing all ghdl files and starting over?

tgingold, 2016-02-04:
On 04/02/16 00:47, JohnRTC wrote:
> I obtained the zip file under the Download Zip button on the code page,
> compiled it and installed it. Other than not signing myself in as root,
> it went without any issues.
>
> Unfortunately, 0.34 did [...]

No.  When configured by default, ghdl uses its own code generator and
doesn't write object files.

Be sure to use 'ghdl --elab-run' (or 'ghdl -r') to simulate your
design.

Regards,
Tristan.

JohnRTC, 2016-02-06:
I am collecting the source code for a gcc build. Is it necessary to use the versions of gcc, mpfr, mpc, and gmp specified in BUILD.txt?  Also, multiple search engines are unable to find a version of gcc written in Ada at least within the top 30 results.  Will the C version do?

Paebbels, 2016-02-07:
The easiest way to build GHDL is to use the LLVM backend. You just need gnat (the Ada compiler), the GCC and LLVM-3.5. Configure GHDL -with-llvm-config=... and launch make.

The pure GCC flow is more complicated.

tgingold, 2016-02-07:
On 07/02/16 00:10, JohnRTC wrote:
> I am collecting the source code for a gcc build. Is it necessary to use
> the versions of gcc, mpfr, mpc, and gmp specified in BUILD.txt? Also,
> multiple search engines are unable to find a version of gcc written in
> Ada at least within the top 30 results. Will the C version do?

Yes, there is only one gcc!  Be sure to use gcc 4.9.x

JohnRTC, 2016-02-08:
Probably the last two questions on building GHDL with gcc:

1) What is the significance of the prefix directory on line 61 of BUILD.txt?  It seems that a relative path is not acceptable; using /usr/local/ requires signing in as root on Ubuntu to build GHDL.

2) Are newer versions for mpfr, mpc and gmp than those specified in BUILD.txt acceptable?  It appears that the instructions will backdate the floating point software supplied with a lot of systems.  On mine, this will move mpfr from a 2013 version to a 2009 version. MPC is also backdated to 2009.  GMP appears to be hidden on the software list.

tgingold, 2016-02-08:
On 08/02/16 04:14, JohnRTC wrote:
> Probably the last two questions on building GHDL with gcc:
>
> 1) What is the significance of the prefix directory on line 61 of
> BUILD.txt? It seems that a relative path is not acceptable; using
> /usr/local/ requires signing in as root on Ubuntu to build GHDL.

That's the directory in which GHDL will be installed.
You can use something like --prefix=$HOME/local
if needed.

> 2) Are newer versions for mpfr, mpc and gmp than those specified in
> BUILD.txt acceptable? It appears that the instructions will backdate the
> floating point software supplied with a lot of systems. [...]

Yes, newer version should work.

Regards,
Tristan.

JohnRTC, 2016-02-12:
After adding g++ and m4 to build gcc and the numeric libraries, I have decided that removal of the debris from failed builds is the excuse I needed to rebuild my computer with a much larger linux partition and smaller windoze partition.  The other is removal of micro$oft spyware.  While this is probably overkill for the linux partition, it is guaranteed to work without fear of screwing things up by deleting the wrong stuff.

Please add g++ and m4 to the utilities list required for rebuilding GHDL with a gcc back end.  g++ is required to compile gcc and m4 is required to compile at least one of the numeric packages (latest versions, don't remember which package requires m4).  Neither are installed by default on Ubuntu 14.04; it stands to reason that this may be the case for other distributions.

JohnRTC, 2016-02-19:
I've found a way to compile GHDL without using root.  However, configuring gcc doesn't work with the command found in BUILD.txt  It claims none of ISL versions 0.10, 0.11, 0.12, and 0.14 can be found.

I found libisl10-0.12.2-2 (64-bit) has been installed by looking with synaptic.  I have rebuilt my machine with Debian 8.2.

Is there a workaround for this problem?

Thank you in advance

John

tgingold, 2016-02-19:
On 19/02/16 19:52, JohnRTC wrote:
> I've found a way to compile GHDL without using root. However,
> configuring gcc doesn't work with the command found in BUILD.txt It
> claims none of ISL versions 0.10, 0.11, 0.12, and 0.14 can be [...]

You don't need isl to build gcc.  That's an optionnal library.

Tristan.

JohnRTC, 2016-03-01:
isl isn't needed for c, c++, fortran, or the rest of the standard languages, but with VHDL, the gcc configuration script fails out with an error if it can't find isl as shown in the attached script output.  The same thing happens with CLooG, and that is as far as I've got.

I've decided to use this as an excuse to learn how to build Debian packages and update packages as required so there are no conflicts between what comes in the Debian installation package and the versions I download, which are a version or two ahead of those supplied by Debian.
[no-isl.zip](https://github.com/tgingold/ghdl/files/153432/no-isl.zip)

JohnRTC, 2016-03-19:
The crashing on isl and CLooG absence was due to missing <package)-dev packages.  Due to bandwidth cost issues (Can$9/1Gbyte), I used the minimal install of Debian 8.2 which includes few if any -dev packages.  As an aside, I am using the same version of the numerics packages as are included in Debian 8.2: gmp 6.0.0, mpfr 3.1.2, and mpc 1.0.2

Now that the above issues have been settled, the gcc configure script is complaining that the compiler driver doesn't understand Ada.  I suspect this is a configuration problem, but don't know for sure.  I searched the internet for a solution, but came up empty other than old bug reports and hits not involving the use of Ada from what I could see.

I have installed all of the gnat compiler packages that I have any reason to believe are relevant:
gnat/stable,now 4.9 all [installed]
gnat-4.9/stable,now 4.9.2-1 amd64 [installed,automatic]
gnat-4.9-base/stable,now 4.9.2-1 amd64 [installed,automatic]
gnat-4.9-sjlj/stable,now 4.9.2-1 amd64 [installed]
gnat-gps/stable,now 5.3dfsg-1 amd64 [installed,automatic]
gnat-gps-common/stable,now 5.3dfsg-1 all [installed,automatic]
gnat-gps-dbg/stable,now 5.3dfsg-1 amd64 [installed]
gnat-gps-doc/stable,now 5.3dfsg-1 all [installed,automatic]
libgnat-4.9/stable,now 4.9.2-1 amd64 [installed,automatic]
libgnatcoll-doc/stable,now 1.6gpl2014-6 all [installed,automatic]
libgnatcoll-gtk1.6/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll-gtk1.6-dev/stable,now 1.6gpl2014-6 amd64 [installed]
libgnatcoll-iconv1.6/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll-iconv1.6-dev/stable,now 1.6gpl2014-6 amd64 [installed]
libgnatcoll-python1.6/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll-python1.6-dev/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll-readline1.6/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll-sqlite-bin/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll-sqlite1.6/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll-sqlite1.6-dev/stable,now 1.6gpl2014-6 amd64 [installed]
libgnatcoll1.6/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatcoll1.6-dev/stable,now 1.6gpl2014-6 amd64 [installed,automatic]
libgnatprj4.9/stable,now 4.9.2-1 amd64 [installed,automatic]
libgnatprj4.9-dev/stable,now 4.9.2-1 amd64 [installed,automatic]
libgnatvsn4.9/stable,now 4.9.2-1 amd64 [installed,automatic]
libgnatvsn4.9-dev/stable,now 4.9.2-1 amd64 [installed,automatic]

Running the makefile eventually fails with a claim that no Ada compiler is installed.  Removing the Debian packaged Ada compiler (upstream source Adacore 2014) and replacing it with the 2015 version directly from Adacore did not make a discernible difference.

The ghdl package was downloaded from github about 2230 GMT March 18, about an hour after that day's commit.

Relevant excerpt of gcc configure script output:
checking for gnatbind... gnatbind
checking for gnatmake... gnatmake
checking whether compiler driver understands Ada... no
checking how to compare bootstrapped objects... cmp --ignore-initial=16 $$f1 $$f2
checking for objdir... .libs
checking for the correct version of gmp.h... yes
checking for the correct version of mpfr.h... yes
checking for the correct version of mpc.h... yes
checking for the correct version of the gmp/mpfr/mpc libraries... yes
checking for version 0.10 of ISL... no
checking for version 0.11 of ISL... no
checking for version 0.12 of ISL... yes
checking for version 0.17.0 of CLooG... no
checking for version 0.18.0 of CLooG... yes
*** This configuration is not supported in the following subdirectories:
     gnattools target-libada target-libstdc++-v3 target-libgfortran target-libgo target-libffi target-libbacktrace target-zlib target-libjava target-libobjc target-boehm-gc target-libcilkrts target-libitm target-libsanitizer target-libvtv
    (Any other directories should still work fine.)
checking for default BUILD_CONFIG... 

gcc configure, config.log, configure console output, and gcc make output are found in files.zip.

Cheers
John
[files.zip](https://github.com/tgingold/ghdl/files/181033/files.zip)

tgingold, 2016-03-21:
On 19/03/16 22:31, JohnRTC wrote:
> The crashing on isl and CLooG absence was due to missing <package)-dev
> packages. Due to bandwidth cost issues (Can$9/1Gbyte), I used the
> minimal install of Debian 8.2 which includes few if any -dev [...]

Try:
which gcc
gcc -v

and post the output.
I am pretty sure that the gcc for Ada is not the first in the path.

Tristan.

JohnRTC, 2016-03-21:
Hello Tristan
The output does not contain any reference to gnat.
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9.3/lto-wrapper
Target: x86_64-unknown-linux-gnu
Configured with: ../gcc-4.9.3/configure --prefix=/usr/local --enable-languages=c,vhdl --disable-bootstrap --with-gmp=/usr/local --disable-lto --disable-multilib
Thread model: posix
gcc version 4.9.3 (GCC) 
jennifer-minerva@laptop:~$

tgingold, 2016-03-21:
On 21/03/16 17:03, JohnRTC wrote:
> Hello Tristan
> The output does not contain any reference to gnat.
> $ gcc -v
> Using built-in specs.
> COLLECT_GCC=gcc
> COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9.3/lto-wrapper [...]

So you are using the wrong gcc.
Be sure the gcc for the Ada compiler is the one in front of you path.

Tristan.

JohnRTC, 2016-03-21:
Hello Tristan

Debian has installed gnat in the same directory as gcc: /usr/bin.  Is this a problem?

I found this link for rebuilding gcc:  

http://www.linuxfromscratch.org/blfs/view/svn/general/gcc-ada.html

Is there another way to get the correct version of gcc that understands Ada?

If I use the info in the link above, I plan to substitute gcc 4.9.3 for the later version.  The link warns of kernel issues if I use 5.3.  Will this work?

Cheers
John

tmeissner, 2016-03-22:
I can only advice you to build GHDL with LLVM backend, that's far more easier then building with GCC.

Yesterday I have built GHDL in 10 minutes, the only I had to do was :
* install GNAT (apt-get install gnat)
* install LLVM (apt-get install llvm-3.5)
* install libedit (apt-get install libedit)

```
configure --with-llvm-config=<path to llvm-config>
make 
make install
```

voilá

tgingold, 2016-03-22:
On 21/03/16 23:33, JohnRTC wrote:
> Hello Tristan
>
> Debian has installed gnat in the same directory as gcc: /usr/bin. Is
> this a problem?

No, this isn't a problem, as long as the 'gcc' program can handle
and compile Ada sources.

> I found this link for rebuilding gcc:
>
> http://www.linuxfromscratch.org/blfs/view/svn/general/gcc-ada.html
>
> Is there another way to get the correct version of gcc that understands Ada?
>
> If I use the [...]

If the gcc driver that can handle Ada has another name (like gnatgcc),
configure gcc with: CC=gnatgcc xxxx/configure ...

Tristan.

closed 2016-07-03 by tgingold