ghdl/ghdl #32: Receive error on generic package

Xaerdna, 2016-02-02:
Hi,

The following error is produced when compiling:

/root/workspace/uvvm/uvvm_vvc_framework/src/ti_generic_queue_pkg.vhd:26:10: 'signal', 'constant', 'variable', 'file' or identifier expected
/root/workspace/uvvm/uvvm_vvc_framework/src/ti_generic_queue_pkg.vhd:26:32: ',' or ':' expected after identifier
/root/workspace/uvvm/uvvm_vvc_framework/src/ti_generic_queue_pkg.vhd:26:32: (found: ;)

Line 26 - 28 from the source code:

generic (type t_generic_element;
         GC_QUEUE_COUNT_MAX       : natural := 1000;
         GC_QUEUE_COUNT_THRESHOLD : natural := 950);

The documentation states that only parts of vhdl 2008 is supported, however I can't find any reference to generic packages. Does ghdl support this yet? Or do you have any other ideas about what goes wrong here?

Thanks

tgingold, 2016-02-02:
Hello,

type interfaces are currently not supported by ghdl.  Maybe in the future...

Regards,
Tristan.

Paebbels, 2016-09-07:
There is progress on type interfaces since 4e6853861f3809d45424b6eb372ccd91ed76f0cb.
Stay tuned!

tgingold, 2017-10-22:
Has been implemented.

closed 2017-10-22 by tgingold