ghdl/ghdl #33: 'others' choice not allowed for an aggregate in this context

RandomReaper, 2016-02-03:
Dear all,

It seems that unconstrained vectors are not supported by ghdl, is my code buggy or is it a missing feature?

example : 
```
library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;

entity test is
port
(
	data				: out std_ulogic_vector
);
end test;

architecture rtl of test is
begin
	data <= (others => '0');
end architecture rtl;
```

ghdl gives me:`test.vhd:14:16: 'others' choice not allowed for an aggregate in this context`.

Thank you in advance

Best regards

Paebbels, 2016-02-03:
It `test` your top-level / test? If so that's not allowed, because `others` can not guess how many  elements need to be generated.

RandomReaper, 2016-02-03:
Dear @Paebbels,

Sure no, `test` is not my top level.
The command I used is `ghdl -a test.vhd`, and I think the analysis should accept this file.

I even tested that, in only one file:
```
library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;
	
entity test is
port
(
	data				: out std_ulogic_vector
);
end test;

architecture rtl of test is
begin
	data <= (others => '0');
end architecture rtl;

library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;

entity test_top is
end test_top;

architecture rtl of test_top is
	signal data : std_ulogic_vector(1 downto 0);
begin
i_test: entity work.test
port map
(
	data => data
);
end architecture rtl;
```

Same error (`'others' choice not allowed for an aggregate in this context.`)

diogratia, 2016-02-03:
IEEE Std 1076-2008 9.3.3.3 Array aggregates paragraph 7 (in part):

> The index range of an array aggregate that has an others choice shall be determinable from the context. That is, an array aggregate with an others choice shall appear only in one of the following contexts:
>> a) As an actual associated with a formal parameter, formal generic, or formal port (or member thereof), where either the formal (or the member) is declared to be of a fully constrained array subtype, or the formal designator is a slice name
>> b) As the default expression defining the default initial value of a port declared to be of a fully constrained array subtype
>> c) As the default expression for a generic constant declared to be of a fully constrained array subtype
>> d) As the result expression of a function, where the corresponding function result type is a fully constrained array subtype
>> e) As a value expression in an assignment statement, where the target is a declared object (or member thereof), **and either the subtype of the target is a fully constrained array subtype or the target is a slice name**

This constraint can also be found in -1993 7.3.2.2 Array aggregates paragraph 4 d).

tgingold, 2016-02-04:
Yes.  Although I don't understand the rational of these VHDL rules, GHDL conform to them.

Tristan.

closed 2016-02-04 by tgingold

RandomReaper, 2016-02-04:
Dear @tgingold,

Thank you for your quick response, and for the nice tip (`data(data'range) <= (others => '0');`)!

I thought it was valid since I used this kind of construct with some commercial VHDL tools.

Best regards

Marc