ghdl/ghdl #37: ghdl: unknown option '-Wl,' for command '-e'

ambikeshwar1991, 2016-02-05:
I am getting this error, while linking C object file to vhdl binary. 
`ghdl: unknown option '-Wl,' for command '-e'`

tgingold, 2016-02-05:
Which version of ghdl are you using: what is the output of ghdl --version ?
There are different backend for ghdl: only ghdl with gcc or llvm backend supports -Wl.

ambikeshwar1991, 2016-02-05:
I have compiled the same source code from your git, output of git --version is as follows
`GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.6
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`

tgingold, 2016-02-05:
With the 'mcode code generator', there is no use of the linker.  So -Wl is not supported.
You need to rebuild ghdl with llvm-3.5 or gcc-49.  See README and BUILD file.

tgingold, 2016-02-07:
Can you explain why you need to link with a C file ?  That could be made available on all ghdl backends.

ambikeshwar1991, 2016-02-07:
I am working on a project for my organization named as FOSSEE at IIT Bombay India. Our motive is to create a mixed mode circuit simulator using GHDL and NgSpice. Where we are creating a digital model in NgSpice using VHDL code which will be simulated by using GHDL and the values will between GHDL and NgSpice will be communicate using socket programming. We have created a Test bench which act as a server and take the values from NgSpice , simulate it and give the values back to Ngspice.  This way we are doing VHDL co simulation.

ambikeshwar1991, 2016-02-08:
I tried installing it with the instruction for gcc. I am getting following error:
`gcc: error trying to exec 'gnat1': execvp: No such file or directory`

Can you help me in this.

tgingold, 2016-02-08:
On 08/02/16 07:53, Ambikeshwar Srivastava wrote:
> I tried installing it with the instruction for gcc. I am getting
> following error:
> |gcc: error trying to exec 'gnat1': execvp: No such file or directory|
>
> Can you help me in this.

You need an Ada compiler (gnat).
On Ubuntu, simply install the gnat package.

Tristan.

ambikeshwar1991, 2016-02-08:
I have installed gnat: Still I am getting following error:
`"GHDL_ADAFLAGS=-Wall -gnata" "GRT_FLAGS=-O2 -g" "GRT_ADAFLAGS=-gnatpn" "LIB_CFLAGS=-O2 -g" "GNATMAKE=gnatmake" "GNATBIND=gnatbind" "CFLAGS=-g -O2" "VHDL_LIB_DIR=/usr/local/lib/gcc/x86_64-unknown-linux-gnu/4.9.2/vhdl" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "libexecsubdir=/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9.2" ADAC="gcc" CC="gcc" libgrt.a grt.lst
make[3]: Entering directory `/home/fossee/mygit/clone/gcc-4.9.2/gcc-objs/gcc/vhdl'
make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
gnatmake -c -aI../../../gcc/vhdl/grt -gnatec../../../gcc/vhdl/grt/grt.adc -gnat05 \
	  ghdl_main -gnatpn -cargs -O2 -g
gnatmake: objects up to date.
gnatbind -Lgrt_ -o run-bind.adb -n ghdl_main.ali
gcc -c -O2 -g -gnatec../../../gcc/vhdl/grt/grt.adc -gnat05 -o run-bind.o run-bind.adb`

tgingold, 2016-02-08:
On 08/02/16 09:36, Ambikeshwar Srivastava wrote:
> I have installed gnat: Still I am getting following error:
> |"GHDL_ADAFLAGS=-Wall -gnata" "GRT_FLAGS=-O2 -g" "GRT_ADAFLAGS=-gnatpn"
> "LIB_CFLAGS=-O2 -g" "GNATMAKE=gnatmake" "GNATBIND=gnatbind" [...]

Quote issue ?  Where is the error message ?

ambikeshwar1991, 2016-02-09:
I was making the source code by using 
`make -j 8` command

ambikeshwar1991, 2016-02-12:
Here is the error
` make[3]: warning: jobserver unavailable: using -j1. Add +' to parent make rule. gnatmake -c -aI../../../gcc/vhdl/grt -gnatec../../../gcc/vhdl/grt/grt.adc -gnat05 \ ghdl_main -gnatpn -cargs -O2 -g gnatmake: objects up to date. gnatbind -Lgrt_ -o run-bind.adb -n ghdl_main.ali gcc -c -O2 -g -gnatec../../../gcc/vhdl/grt/grt.adc -gnat05 -o run-bind.o run-bind.adb gcc: error: run-bind.adb: Ada compiler not installed on this system `

tgingold, 2016-02-12:
On 12/02/16 08:09, Ambikeshwar Srivastava wrote:
> Here is the error
> |make[3]: warning: jobserver unavailable: using -j1. Add +' to parent
> make rule. gnatmake -c -aI../../../gcc/vhdl/grt
> -gnatec../../../gcc/vhdl/grt/grt.adc -gnat05 \ ghdl_main [...]

Did you try to use 'make ADAC=gnatgcc' to build ghdl ?

ambikeshwar1991, 2016-02-13:
yes I tried this and still getting error:
`gnatbind -Lgrt_ -o run-bind.adb -n ghdl_main.ali
gcc -c -O2 -g -gnatec../../../gcc/vhdl/grt/grt.adc -gnat05 -o run-bind.o run-bind.adb
gcc: error: run-bind.adb: Ada compiler not installed on this system
make[3]: *** [run-bind.o] Error 1
make[3]: Leaving directory `/home/fossee/mygit/clone/gcc-4.9.2/gcc-objs/gcc/vhdl'
make[2]: *** [vhdl/libgrt.a] Error 2
make[2]: Leaving directory `/home/fossee/mygit/clone/gcc-4.9.2/gcc-objs/gcc'
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory `/home/fossee/mygit/clone/gcc-4.9.2/gcc-objs'
make: *** [all] Error 2
`

tgingold, 2016-02-14:
On 13/02/16 10:20, Ambikeshwar Srivastava wrote:
> yes I tried this and still getting error:
> |gnatbind -Lgrt_ -o run-bind.adb -n ghdl_main.ali
> gcc -c -O2 -g -gnatec../../../gcc/vhdl/grt/grt.adc -gnat05 -o run-bind.o
> run-bind.adb
> gcc: error: [...]

ok, can you also try with: 'make CC=gnatgcc' ?

Tristan.

closed 2016-06-21 by tgingold