ghdl/ghdl #38: Using an alias to a protected type's method causes a GHDL exception.

Paebbels, 2016-02-08:
I have a shared variable in VHDL-2008, which is a protected type. To offer a common interface for all language versions, I'm using aliases to hide the shared variable name in the call to the method.

Declaring the alias is no problem, but the usage causes a crash.

See my attached reproducer and the embedded issue report for more details.

[bugreport_aliasprotected.vhdl.txt](https://github.com/tgingold/ghdl/files/121437/bugreport_aliasprotected.vhdl.txt)

tgingold, 2016-02-10:
Apparently, this is valid.  I will investigate.

Xiretza, 2019-11-24:
This now works as expected:

```
$ ghdl -a --std=08 bugreport_aliasprotected.vhdl
$ ghdl -e --std=08 test
$ ghdl -r --std=08 test
bugreport_aliasprotected.vhdl:163:16:@0ms:(report note): wrapGet:      7  expected: 7
bugreport_aliasprotected.vhdl:165:16:@0ms:(report note): wrapGet:      5  expected: 5
bugreport_aliasprotected.vhdl:166:16:@0ms:(report note): wrapExcahnge: 5  expected: 5
bugreport_aliasprotected.vhdl:167:16:@0ms:(report note): wrapGet:      3  expected: 3
bugreport_aliasprotected.vhdl:169:16:@0ms:(report note): wrapGet:      0  expected: 0
```

closed 2021-01-17 by eine