ghdl/ghdl #39: Compiling file with --std=08 and --ieee=synopsys

go2sh, 2016-02-11:
Hello there,

I try to compile the altera mega function simulation files (altera_mf). If I try to use --std= with --ieee the compilation fails with: "cannot find resource library "ieee""
If I remove the --std option everything is fine. 

There are some other things with these options, that seem wrong. With --std=93 and --ieee=synopsys I get an error, with --ieee=synopsys only I get the same message but as warning.

Do I miss something with those two options?

Best regards,
Chris

Paebbels, 2016-02-11:
Hello Chris,

The Altera simulation library can only be compiled in VHDL-93 mode with enabled Synopsys flavour. 
So please add `--std=93c` and `--ieee=synopsys` to your GHDL commands. These packages have more problems, why I recommend the shipped compile scripts for vendor libraries. These scripts are not yet part of the installation process (`make install`), but the can be downloaded from GitHub or found in your local repository, if you cloned GHDL onto your local computer.

GHDL provides vendor library compile scripts for Linux (Bash script) and Windows (PowerShell script). See the [libraries/vendors](https://github.com/tgingold/ghdl/tree/master/libraries/vendors) folder for the scripts [`compile-altera.sh` ](https://github.com/tgingold/ghdl/blob/master/libraries/vendors/compile-altera.sh) or [`compile-altera.ps1`](https://github.com/tgingold/ghdl/blob/master/libraries/vendors/compile-altera.ps1) and my provided [README.md](https://github.com/tgingold/ghdl/blob/master/libraries/vendors/README.md) for information on how to use these scripts.

**In short:**
 1. Configure the path to your Quartus installation in the `config.sh/ps1` file.
 2. Go to your destination folder.
 3. Run the script from this folder.
 4. A subfolder named `altera` is created, including all pre-compiled packages and primitives.
 5. Use GHDL with `-Paltera` to reference the created folder.

tgingold, 2016-02-11:
On 11/02/16 13:51, Christoph Seitz wrote:
> Hello there,
>
> I try to compile the altera mega function simulation files (altera_mf).
> If I try to use --std= with --ieee the compilation fails with: "cannot
> find resource library "ieee""
> If I remove [...]

In addition to Patrick's answer I would add that there is no synopsys
version of IEEE in vhdl 2008.  So you should try without --ieee.
Maybe GHDL have to emit a better error message.

Tristan.

go2sh, 2016-02-12:
One additional question: Is it possible to mix VHDL-93c and VHDL-2008 together, or should every thing  be the same standard? The documentation only mentions, that VHDL-87 and VHDL-93 are not compatible. I'm just asking, because it seems that modelsim is capable of mixing 93 and 2008.

The second part of my first post still persists:
`
C:\Users\go2sh>ghdl -a --std=93 --ieee=synopsys C:\Python27\Lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd
> C:\Python27\Lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd:595:27: universal integer bound must be numeric literal or attribute
`
and
`
C:\Projekte\veg102\Impl\Simulation\frame_sync>ghdl -a --std=93c --ieee=synopsys C:\Python27\Lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd
C:\Python27\Lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd:595:27:warning: universal integer bound must be numeric literal or attribute
`
The upper command doesn't compile and the lower command does.

Many thanks for your help.
Chris

tgingold, 2016-02-12:
On 12/02/16 10:54, Christoph Seitz wrote:
> One additional question: Is it possible to mix VHDL-93c and VHDL-2008
> together, or should every thing be the same standard? The documentation
> only mentions, that VHDL-87 and VHDL-93 are not compatible. I'm just
> asking, because it seems that modelsim is capable of mixing 93 and 2008.

No, it is not possible to mix 93 and 2008.

> The second part of my first post still persists:
> `
> C:\Users\go2sh>ghdl -a --std=93 --ieee=synopsys
> C:\Python27\Lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd [...]

With the strict vhdl 93 definition, your string_ops.vhd file is not 
valid and therefore rejected by ghdl.

--std93c slightly relaxes the rules.

Regards,
Tristan.

go2sh, 2016-02-16:
Okay, thank you very much. I close it.

closed 2016-02-16 by go2sh

Paebbels, 2016-07-01:
Update to the old question from @go2sh:

The updated `compile-altera.sh` script now supports VHDL-2008 compilation with GHDL. The other scripts (Lattice, Xilinx, ...) have also a `--vhdl2008` option. We introduced a `synopsys` flavor for VHDL-2008 like in VHDL-93.

go2sh, 2016-07-01:
Awesome. Now we can use ghdl exclusivly for unit testing. :+1: