ghdl/ghdl #40: GHDL Bug occured: trans-chap4.adb:306

RandomReaper, 2016-02-12:
Dear all,

ghdl crashed with the following message:

```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 5.2.1
In directory: /home/pim/git/bug_report/
Command line:
/usr/local/bin/ghdl1-llvm -P/usr/local/lib/ghdl/v93/std/ -P/usr/local/lib/ghdl/v93/ieee/ -c -o test.o test.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: trans-chap4.adb:306
Call stack traceback locations:
0x7fe4d3db5431 0x586799 0x593b0a 0x594373 0x5355b7 0x537350 0x5373fc 0x56a1ca 0x56695a 0x56707d 0x5b365f 0x5544f4 0x5b8816 0x46ce48 0x42fe13 0x7fe4d3233a3e 0x42f277 0xfffffffffffffffe
******************************************************************
```

How to reproduce : `ghdl -a test.vhd`, with the following `test.vhd`:
```vhdl
library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;

entity a is
port
(
	foo : in std_ulogic_vector
);
end a;

architecture rtl of a is
begin
end rtl;

library ieee;
	use ieee.std_logic_1164.all;
	use ieee.numeric_std.all;

entity b is
port
(
	bar		: inout	std_logic_vector(7 downto 0)
);
end b;

architecture rtl of b is
begin

i_a: entity work.a
port map
(
	foo => std_ulogic_vector(bar)
);
end rtl;
```

The problematic line is `foo => std_ulogic_vector(bar)`. The problem can be easily worked around by doing that:
```vhdl
architecture rtl of b is
	signal baz : std_ulogic_vector(bar'range);
begin

baz <= std_ulogic_vector(bar);

i_a: entity work.a
port map
(
	foo => baz
);
end rtl;
```

Best regards

Marc

tgingold, 2016-02-12:
On 12/02/16 11:19, Marc Pignat wrote:
> Dear all,
>
> ghdl crashed with the following message:

I can reproduce the crash.  I will investigate it.

Thank you for the report,
Tristan.

tgingold, 2016-02-14:
I think your reproducer is invalid:  the association
   foo => std_ulogic_vector(bar)
is not allowed by vhdl rules as both 'foo' and 'std_ulogic_vector' is unconstrained.
That should be detected by ghdl.

Tristan.

tgingold, 2016-02-17:
This error is now reported by ghdl.

Thank you for the bug report.
Tristan.

closed 2016-02-17 by tgingold

RandomReaper, 2016-02-18:
Good job, thank you!

Best regards

Marc