ghdl/ghdl #41: GHDL loops forever if a file is opened via file_open (...).

Paebbels, 2016-02-13:
GHDL loops forever if a file is opened via file_open (...). I get a message from Windows that ghdl.exe got killed.

A reproducer is attached.
Tested with GHDL 0.34dev on Windows  (mcode) in VHDL-93 mode.

[fileio.v93.vhdl](https://github.com/tgingold/ghdl/files/129421/fileio.v93.vhdl.txt)

diogratia, 2016-02-14:
```vhdl
	procedure LogFile_Open(FileName : STRING; OpenKind : T_LOGFILE_OPEN_KIND := WRITE_MODE) is
		variable OpenStatus		: FILE_OPEN_STATUS;
	begin
		LogFile_Open(FileName, OpenKind);
	end procedure;
```
What's going to happen if you call this routine recursively? There's going to be an equivalent process for the concurrent procedure call (currently commented out in architecture tb of bugreport) containing a sequential procedure call and the sequential procedure call is going to call itself.   

What's the effect of a sequential procedure calling itself?  You'd expect a procedure call to be tagged with the calling location (under certain conditions, involving wait statements to know what process to suspend. Without a wait statement you should eventually run out of calling stack, and that's exactly what happened on ghdl-0.33-llvm on my OS X 10.10.5:

> 
> Process:               bugreport [40215]
> Path:                  /Users/USER/Desktop/bugreport
> Identifier:            bugreport
> Version:               0
> Code Type:             X86-64 (Native)
> Parent Process:        ghdl [40214]
> Responsible:           iTerm [40177]
> User ID:               501
> 
> Date/Time:             2016-02-14 21:18:17.380 +1300
> OS Version:            Mac OS X 10.10.5 (14F27)
> Report Version:        11
> Anonymous UUID:        E3A03FA0-82EC-FA12-239C-94800EE699E8
> 
> Sleep/Wake UUID:       5F429778-6150-45C6-9059-8B06D098DBAA
> 
> Time Awake Since Boot: 1000000 seconds
> Time Since Wake:       1200 seconds
> 
> Crashed Thread:        0  Dispatch queue: com.apple.main-thread
> 
> Exception Type:        EXC_BAD_ACCESS (SIGILL)
> Exception Codes:       KERN_PROTECTION_FAILURE at 0x00007fff5cc5bff8
> 
> VM Regions Near 0x7fff5cc5bff8:
>     MALLOC_TINY            00007ff482800000-00007ff483600000 [ 14.0M] rw-/rwx SM=PRV  
> --> STACK GUARD            00007fff5945c000-00007fff5cc5c000 [ 56.0M] ---/rwx SM=NUL  stack guard for thread 0
>     Stack                  00007fff5cc5c000-00007fff5d45c000 [ 8192K] rw-/rwx SM=COW  thread 0
> 
> Thread 0 Crashed:: Dispatch queue: com.apple.main-thread

(And it took most of ten minutes, I must have some big virtual memory).

If there's a ghdl flaw, it's that there is no recursion limit for procedures calling themselves, otherwise this is an operator error, calling a procedure recursively with no limit.

Paebbels, 2016-02-14:
Damn! I made it again ... a deep recusion.

The `Status` argument is missing at position 1, so the procedure calls itself...

tgingold, 2016-02-15:
On 14/02/16 10:56, Patrick Lehmann wrote:
> Damn! I made it again ... a deep recusion.
>
> The |Status| argument is missing at position 1, so the procedure calls
> itself...

It could be very easy to emit a message for direct recursion like this.
Would that be useful ?

It could also be possible to emit a warning for direct recursion that
appear outside an if/for/case statement.  Would that be useful ?

Tristan.

Paebbels, 2016-02-15:
Both ideas could help a lot to see the fault in the design.