ghdl/ghdl #42: Exception raised by faulty VHDL code.

Paebbels, 2016-02-16:
I did some experiments on custom defined attributes and uncovered an exception. See my attached reproducer for details.

I'm not sure if my code contains correct VHDL, so I just want to report an abnormal program exit.

[bugreport_attribute.vhdl](https://github.com/tgingold/ghdl/files/131656/bugreport_attribute.vhdl.txt)

diogratia, 2016-02-16:
There are two additional errors besided the unreported attribute parsing error.

```vhdl
architecture tb of test is
	function to_string(slv : STD_LOGIC_VECTOR) return STRING is
		-- variable Result		: STRING(slv'length - 1 downto 0); -- ERROR
        variable Result         : STRING (1 to slv'length);  -- FIXED
	begin
		for i in slv'range loop
			-- Result(i + 1)	:= STD_LOGIC'image(slv(i));  -- ERROR
            Result(i + 1)	:= character'value(STD_LOGIC'image(slv(i))); -- FIXED
		end loop;
		return Result;
	end function;
	
	attribute serialize		: string;
	
	signal mySignal		: STD_LOGIC_VECTOR(7 downto 0);
	-- attribute serialize of mySignal		: signal is to_string[STD_LOGIC_VECTOR return STRING];  -- ERROR
    attribute serialize of mySignal : signal is to_string(mySignal); -- FIXED
	
begin
```
Then:

>ghdl -a bugreport_attribute.vhdl
>ghdl -e test
>ghdl -r test
>bugreport_attribute.vhdl:82:16:@0ms:(report note): mySignal=UUUUUUUU

tgingold, 2016-02-17:
This is now fixed.

closed 2016-02-18 by tgingold