ghdl/ghdl #43: Generated process statement crashes with

go2sh, 2016-02-16:
Hi there,
I got the following message from ghdl while simulating a process created by a generate statement:
> ******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
In directory: C:\Projekte\test\
Command line:
C:\Projekte\ghdl-dev\bin\ghdl.exe --elab-run --std=08 --work=lib --workdir=C:\Pr
ojekte\test\vunit_out\ghdl\libraries\lib -PC:\Projekte\test\vunit_out\ghdl\libra
ries\vunit_lib -PC:\Projekte\test\vunit_out\ghdl\libraries\lib test_tb beh -grun
ner_cfg=enabled_test_cases : test1,output path : C::/Projekte/test/vunit_out/tes
ts/lib.test_tb.test1/,active python runner : true --assert-level=error
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
Exception name: ORTHO_CODE.SYNTAX_ERROR
Message: ortho_code-exprs.adb:461
******************************************************************

I attached the source code for the vunit run. If you change the string in line 37 to a value without i every thing is fine. I also stumbled on some other issues arround this code with Program_Error raised but I cannot reproduce them yet.

[test_tb.vhd.txt](https://github.com/tgingold/ghdl/files/132681/test_tb.vhd.txt)
[run.py.txt](https://github.com/tgingold/ghdl/files/132682/run.py.txt)

tgingold, 2016-02-16:
On 16/02/16 15:38, Christoph Seitz wrote:
> Hi there,
> I got the following message from ghdl while simulating a process created
> by a generate statement:

I cannot (yet) reproduce the issue.

Do you know how to display executed commands by vunit ?

Tristan.

tgingold, 2016-02-16:
Never mind, I can reproduce the crash.

go2sh, 2016-02-16:
How do you debug those issues? I tried it with gnat tools under windows, but I was unsuccessful..

tgingold, 2016-02-17:
On 16/02/16 23:09, Christoph Seitz wrote:
> How do you debug those issues? I tried it with gnat tools under windows,
> but I was unsuccessful..

First you need to rebuild ghdl from the sources in order to get debug 
symbols.

Then run ghdl within gdb.  There is a .gdbinit in the sources that
set a breakpoint on exceptions.  So ghdl would be stopped at the
place of the crash.  Then investigation could start...

Tristan.

tgingold, 2016-02-17:
I have just pushed a fix.  Unfortunately, there is no obvious workaround.

Tristan.

closed 2016-02-17 by tgingold