ghdl/ghdl #44: Unhandled exception for assignment of signed(std_logic_vector(unsigned)) to a part of signed vector

prabhpreet, 2016-03-04:
Output:
```
eval_pos: cannot handle IIR_KIND_SUBSTRACTION_OPERATOR (loopfilter/loopfilter.vhd:22:37)

******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: 5.3.0
In directory: /home/prabhpreet/Documents/College/Notes/SEM8/DSP Project/Work/vhdl/
Command line:
/usr/lib/gcc/x86_64-unknown-linux-gnu/4.9.3/ghdl1 --std=93c --workdir=workdir -P/usr/lib/gcc/x86_64-unknown-linux-gnu/4.9.3/vhdl//v93/std/ -P/usr/lib/gcc/x86_64-unknown-linux-gnu/4.9.3/vhdl//v93/ieee/ -quiet -o workdir/loopfilter.s loopfilter/loopfilter.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:63
Call stack traceback locations:
0x64be13 0x65e1d7 0x660340 0x6dc748 0x6913b7 0x693454 0x698218 0x6a46c8 0x6a58a3 0x6bc076 0x6e40d9 0x6b2d13 0x6e5d55 0x632d6d 0x6100ce 0x9f97af 0x9fb44e 0x632dc9 0x60d1c7 0x7fd9e046670e 0x60db47 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:63
Call stack traceback locations:
0x64be13 0x65e1d7 0x660340 0x6dc748 0x6913b7 0x693454 0x698218 0x6a46c8 0x6a58a3 0x6bc076 0x6e40d9 0x6b2d13 0x6e5d55 0x632d6d 0x6100ce 0x9f97af 0x9fb44e 0x632dc9 0x60d1c7 0x7fd9e046670e 0x60db47 0xfffffffffffffffe
ghdl: compilation error
```
Input file (shortened):
```
-- loopfilter.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity loopfilter is
  port(clk, reset: in std_logic;
      mult_error_op:in signed(38 downto 0);
      f_desired: in unsigned(5 downto 0);
      f_word_output: out unsigned(5 downto 0));
end loopfilter;

architecture loopfilter_arch of loopfilter is

  signal beta_e, alpha_e: signed(45 downto 0);
  constant alpha: signed(7 downto 0) := "01110011";
  constant beta: signed(7 downto 0) := "00011010";
  constant plus_one: signed(alpha_e'range) := ('0', others => '1');

...

begin

...

  ADDERS : process(clk, reset)
  variable freq_add_var : signed(beta_e'range);
  variable n_total_var: signed(alpha_e'range);
  variable temp: signed(n_total_var'range);
  begin
    if reset = '1' then
      freq_add_var := (others => '0');
      freq_add_var(42 downto 37) := signed(std_logic_vector(f_desired)); --This line is causing problems.
      n_total_var := (others => '0');
    elsif 
      ...
    end if;
   ...

  end process;

end loopfilter_arch;

```

I was also getting errors when I tried assigning `freq_add_var` as an aggregate

```
freq_add_var := (42 downto 37 => signed(std_logic_vector(f_desired)), others=> '0');
```

That also gave me an exception and weird errors.

Full file at: [https://ptpb.pw/eyf_](https://ptpb.pw/eyf_)

diogratia, 2016-03-06:
Creating an analyzable version of the shortened code and adding a testbench to reach the assignment statement:

```vhdl
-- loopfilter.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity loopfilter is
  port(clk, reset: in std_logic;
      mult_error_op:in signed(38 downto 0);
      f_desired: in unsigned(5 downto 0);
      f_word_output: out unsigned(5 downto 0));
end loopfilter;

architecture loopfilter_arch of loopfilter is

  signal beta_e, alpha_e: signed(45 downto 0);
  constant alpha: signed(7 downto 0) := "01110011";
  constant beta: signed(7 downto 0) := "00011010";
  constant plus_one: signed(alpha_e'range) := ('0', others => '1');

-- ...

begin

-- ...

  ADDERS : process(clk, reset)
  variable freq_add_var : signed(beta_e'range);
  variable n_total_var: signed(alpha_e'range);
  variable temp: signed(n_total_var'range);
  begin
    if reset = '1' then
      freq_add_var := (others => '0');
      freq_add_var(42 downto 37) := signed(std_logic_vector(f_desired)); --This line is causing problems.
      n_total_var := (others => '0');
    elsif FALSE then --  was just elsif
--      ...
    end if;
--   ...

  end process;

end loopfilter_arch;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity loopfilter_tb is
end entity;

architecture foo of loopfilter_tb is
    signal clk:             std_logic := '0';
    signal reset:           std_logic := '0';
    signal mult_error_op:   signed (38 downto 0) := (others => '0');
    signal f_desired:       unsigned (5 downto 0) := (others => '0');
    signal f_word_output:   unsigned (5 downto 0);
begin

DUT:
    entity work.loopfilter
        port map (
            clk => clk,
            reset => reset,
            mult_error_op => mult_error_op, 
            f_desired => f_desired,
            f_word_output =>  f_word_output
        );

STIMULIS:
process
    begin
        wait for 10 ns;
        reset <= '1';
        wait for 10 ns;
        wait;
    end process;        
end architecture;
```
Gives:

> %% ghdl -a loopfilter.vhdl
> %% ghdl -e loopfilter_tb
> %% ghdl -r loopfilter_tb
> %%

No error using the the llvm version of ghdl 0.33 (On OS X). Please check with your gcc ghdl. Notice there is no subtraction operator at the line you indicated in the shortened code.

>eval_pos: cannot handle IIR_KIND_SUBSTRACTION_OPERATOR (loopfilter/loopfilter.vhd:22:37)

Nor at line 22 of the original file [loopfilter.vhd](https://ptpb.pw/eyf_ "loopfilter.vhd").

Can you help reproduce the error?

Your complete file would take dummying up a q_one_dot_fp_multiplier and bit more work in a testbench.

Where did  the gcc version of ghdl you are using come from?

tgingold, 2016-03-07:
There is a link to the full file (last line).

I think this is a duplicate of a previous issue.  Indeed, ghdl 0.33 fails to elaborate this design, but that was fixed in the development version.

I will add the reproducer to the test suite.

prabhpreet, 2016-03-07:
Hi, gcc -v is

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-unknown-linux-gnu/5.3.0/lto-wrapper
Target: x86_64-unknown-linux-gnu
Configured with: /build/gcc-multilib/src/gcc-5-20160209/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release
Thread model: posix
gcc version 5.3.0 (GCC) 
```

The q_one_dot_fo_multiplier file: [link](https://ptpb.pw/im61).

I had modified the code, but I'm not facing an exception ([Link to file](https://ptpb.pw/UOkU), another component used [link](https://ptpb.pw/hFi_)). Instead, I'm facing an error which I'm not facing on another VHDL simulation tool (Libero IDE), on this particular line:

      F0: freq_add_reset <= (42 downto 37 => signed(std_logic_vector(f_desired)), others => '0');

Error:
```
loopfilter/loopfilter.vhd:42:48: can't match type conversion with type std_ulogic
loopfilter/loopfilter.vhd:42:48: (location of type conversion)
ghdl: compilation error
```
I'm using the 93c standard. I wonder if this is a bug, or a syntax error.

tgingold, 2016-03-26:
Hello,

sorry I missed your last message.
All the files are analysed by ghdl.

The line
`F0: freq_add_reset <= (42 downto 37 => signed(std_logic_vector(f_desired)), others => '0');`
is not part of any file and looks incorrect in vhdl93.

Regards,
Tristan.

closed 2016-03-26 by tgingold