ghdl/ghdl #45: Support to evaluate PSL endpoints in VHDL code

tmeissner, 2016-03-08:
GHDL already supports the PSL `endpoint` statement, which triggers at the last cycle of a sequence which was hit. However, you only can evaluate the `endpoint` in PSL code. In other simulators it is possible to read the value of the `endpoint` in VHDL code like this:

```vhdl
-- psl endpoint e_test is {req; not(grant); grant}@rising_edge(clk);

process is
begin
  wait until e_test;
  report "e_test hit";
  wait;
end process;
```

You can find an other example here: http://www.cvcblr.com/blog/?p=810

I don't know, if evaluating PSL endpoints in VHDL is defined in the PSL standard, but the tools that I now (Modelsim/Questa) support that. I will try on edaplayground, if Riviera also supports that, but I think it does. In the IEEE PSL standard `endpoint` was replaced by the `ended()` function, which is similar. But GHDL don't supports any PSL functions, so it would be a more easy way to enhance the already supported `endpoint` statement.

tmeissner, 2016-03-08:
Here is my test case:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library std;
use std.env.all;

entity psl_test_endpoint is
end entity psl_test_endpoint;

architecture test of psl_test_endpoint is

  signal s_rst_n : std_logic := '0';
  signal s_clk   : std_logic := '0';
  signal s_write : std_logic;
  signal s_read  : std_logic;

begin

  s_rst_n <= '1' after 100 ns;
  s_clk   <= not s_clk after 10 ns;

  TestP : process is
  begin
    report "RUNNING psl_test_endpoint test case";
    report "==========================================";
    s_write <= '0';
    s_read  <= '0';
    wait until s_rst_n = '1' and rising_edge(s_clk);
    s_write <= '1';
    wait until rising_edge(s_clk);  -- endpoint is be active
    s_read  <= '1';
    wait until rising_edge(s_clk);
    s_write <= '0';
    s_read  <= '0';
    wait until rising_edge(s_clk);
    stop(0);
  end process TestP;

  -- psl default clock is rising_edge(s_clk);
  -- sequence & endpoint definition
  -- psl endpoint E_TEST is {not(s_write); s_write};

  -- It's not possible to use endpoints in VHDL code
  EndpointP: process is
  begin
    wait until E_TEST;
    report "TEST";
    wait;
  end process EndpointP;
```

`$ ghdl -a --std=08 -fpsl psl_test_endpoint.vhd`produces following error: `psl_test_endpoint.vhd:55:16: PSL declaration "e_test" not allowed in an expression`

tgingold, 2016-03-10:
Being able to reference endpoints in VHDL is indeed a powerful feature.
I am currently reading 1850-2010 to understand the enhancements since PSL v1.1

Do you need support for PSL functions ?

tmeissner, 2016-03-10:
PSL functions would be a nice addition to GHDL, but they aren't a blocker at the moment. If you consider to implement PSL functions, the ended() function would be the standard way when evaluating PSL events in VHDL.

Regarding PSL 2010: the main addition I know is local variables in properties I think. The syntax is crude imho, but they make possible to write more readable properties where you needed property replication before.

Another question: is there a data structure in GHDL which accumulates how often PSL assertion and cover directives were triggered? I've looked into the code regarding PSL, but I don't really understand it works without knowledge of the internal structure of GHDL ;)

tgingold, 2016-03-11:
On 10/03/16 10:52, T. Meissner wrote:
> PSL functions would be a nice addition to GHDL, but they aren't a
> blocker at the moment. If you consider to implement PSL functions, the
> ended() function would be the standard way when evaluating PSL events in
> VHDL.

My understanding is that ended() is only available within PSL expressions.

> Regarding PSL 2010: the main addition I know is local variables in
> properties I think. The syntax is crude imho, but they make possible to
> write more readable properties where you needed property replication before.

Yes, but that's not easy to implement :-(

> Another question: is there a data structure in GHDL which accumulates
> how often PSL assertion and cover directives were triggered? I've looked
> into the code regarding PSL, but I don't really understand it works
> without knowledge of the internal structure of GHDL ;)

No, but that's much easier to implement.

Tristan.

tmeissner, 2016-03-11:
> My understanding is that ended() is only available within PSL expressions.
Maybe this isn't described in the PSL standard. It is also not mentioned in the book "Practical Introduction to PSL". They only claim, that ended() is a boolean which is true in the last cycle of a triggered sequence.

Modelsim/Questa supports evaluating ended() in HDL. The syntax is similar to evaluating endpoints in VHDL like this:

```vhdl
-- psl sequence s_test is {req; not(grant); grant}@rising_edge(clk);

process is
begin
  wait until ended(s_test);
  report "s_test hit";
  wait;
end process;
```

I've tried a similar example with Riviera 2015.01 on edaplayground (http://www.edaplayground.com/x/3sjM), but it don't work. So, is it a Mentor extension? I don't believe that, but I'm not sure. There are some articles on the web, that describe that behaviour, most with endpoints, but the behaviour should be the same.

Regarding the PSL data structure: It's content could be similar to other simulators. They store the assertion/cover name, counts of passes and failures, the type of assertion/cover (psl/vhdl) and in which design unit they are. Something like this (I use strings for fields which would printed as strings in a report):

```ada
type CoverageLog is record
    Name    : string;
    Passes  : natural;
    Fails   : natural;
    Type    : string;
    Unit    : string;
end record;
```

This could be used to export a coverage report in plain text and xml (which would be good for further processing by tools).

tgingold, 2016-03-18:
On 10/03/16 10:52, T. Meissner wrote:
> PSL functions would be a nice addition to GHDL, but they aren't a
> blocker at the moment. If you consider to implement PSL functions, the
> ended() function would be the standard way when evaluating [...]

This latter is now implemented.  I have posted this message to 
sourceforge ticket 24:

this enhancement is now implemented.  A new option, 
--psl-report=FILENAME write a JSON report to FILENAME.

Feedback is welcome.

Now I will investigate endpoints.

Tristan.

tmeissner, 2016-03-18:
Oh, cool. I'm at the linux days in Chemnitz this weekend, i think I will have time to test it between the lectures. Thanks :+1: 

Regarding endpoints/ended() in VHDL: I think now, they are valid in the modeling layer of PSL. So, in a PSL vunit, they should be valid in the selected HDL flavor. The evaluation of endpoints/ended() in normal VHDL architectures is a Mentor extension maybe.

tgingold, 2016-03-18:
On 18/03/16 20:12, T. Meissner wrote:
> Oh, cool. I'm at the linux days in Chemnitz this weekend, i think I will
> have time to test it between the lectures. Thanks :+1:

Have fun!

> Regarding endpoints/ended() in VHDL: I think now, they are valid in the
> modeling layer of PSL. So, in a PSL vunit, they should be valid in the
> selected HDL flavor. The evaluation of endpoints/ended() in normal VHDL
> architectures is a Mentor extension maybe.

Possibly.  But this is unambiguous and powerful.

tmeissner, 2016-03-18:
Mhm, I get no report file. What I'm doing wrong?

```
$ ghdl -a -fpsl --std=08 psl_test_endpoint.vhd 
$ ghdl -e -fpsl --std=08 psl_test_endpoint
$ ./psl_test_endpoint --psl-report=psl_report
psl_test_endpoint.vhd:31:5:@0ms:(report note): RUNNING psl_test_endpoint test case
psl_test_endpoint.vhd:32:5:@0ms:(report note): ==========================================
psl_test_endpoint.vhd:64:10:@30ns:(psl assertion error): ENDPOINT ASSERT
psl_test_endpoint.vhd:64:10:@50ns:(psl assertion error): ENDPOINT ASSERT
psl_test_endpoint.vhd:64:10:@70ns:(psl assertion error): ENDPOINT ASSERT
psl_test_endpoint.vhd:64:10:@90ns:(psl assertion error): ENDPOINT ASSERT
psl_test_endpoint.vhd:64:10:@110ns:(psl assertion error): ENDPOINT ASSERT
psl_test_endpoint.vhd:67:10:@130ns:(psl cover note): COVERED2
psl_test_endpoint.vhd:65:10:@130ns:(psl cover note): COVERED0
psl_test_endpoint.vhd:64:10:@130ns:(psl assertion error): ENDPOINT ASSERT
psl_test_endpoint.vhd:64:10:@150ns:(psl assertion error): ENDPOINT ASSERT
simulation stopped @170ns with status 0
```

tmeissner, 2016-03-18:
Mhm, testcase for ticket24 works as expected. But it uses 'elab-run' as option. And, it doesn't work anymore, if --std=08 is added in analysing. Then 'elab-run' gives an error:

```
$ ghdl -a --std=08 -fpsl psl.vhdl
$ ghdl --elab-run -fpsl psl --psl-report=psl.out
/usr/local/bin/ghdl1-llvm: cannot find entity or configuration psl
ghdl: compilation error
```

I always use --std=08 in my testbenches because of OSVVM. If I run the design as in my comment before, than it gicves no error, but also no psl-report output :/

tmeissner, 2016-03-18:
My testcase for the psl-report, which doesn't produce psl-export file:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library std;
use std.env.all;

entity psl_test_endpoint is
end entity psl_test_endpoint;

architecture test of psl_test_endpoint is

  signal s_rst_n : std_logic := '0';
  signal s_clk   : std_logic := '0';
  signal s_write : std_logic;
  signal s_read  : std_logic;

begin

  s_rst_n <= '1' after 100 ns;
  s_clk   <= not s_clk after 10 ns;

  TestP : process is
  begin
    report "RUNNING psl_test_endpoint test case";
    report "==========================================";
    s_write <= '0';  -- named assertion should hit
    s_read  <= '0';
    wait until s_rst_n = '1' and rising_edge(s_clk);
    s_write <= '1';
    wait until rising_edge(s_clk);
    s_read  <= '1';  -- assertion should hit
    wait until rising_edge(s_clk);
    s_write <= '0';
    s_read  <= '0';
    wait until rising_edge(s_clk);
    stop(0);
  end process TestP;

  -- psl default clock is rising_edge(s_clk);
  -- psl endpoint E_TEST is {not(s_write); s_write};
  -- psl sequence abc_seq is {not(s_write); s_write};

  -- psl ENDPOINT_ASSERT : assert always {E_TEST; s_read} report  "ENDPOINT ASSERT";
  -- psl COVERAGE0 : cover {not(s_write); s_write} report "COVERED0";
  -- psl COVERAGE1 : cover {abc_seq} report "COVERED1";
  -- psl COVERAGE2 : cover {E_TEST} report "COVERED2";

end architecture test;
```

tgingold, 2016-03-19:
On 18/03/16 21:31, T. Meissner wrote:
> Mhm, I get no report file. What I'm doing wrong?

Nothing, but you are stopping simulation with std.env.stop
I will fix.

Tristan.

tgingold, 2016-03-19:
On 18/03/16 23:28, T. Meissner wrote:
> My testcase for the psl-report, which doesn't produce psl-export file:

Thanks.  It now works on my setup, I will commit the fix soon.

Tristan.

tmeissner, 2016-03-19:
Yep, after I removed the ``stop()``statement, the report is generated. Looks good, great :+1: 

```json
{ "details" : [
 { "directive": "assertion",
   "name": ".psl_test_endpoint(test).endpoint_assert",
   "file": "psl_test_endpoint.vhd",
   "line": 52,
   "count": 1,
   "status": "failed"},
 { "directive": "cover",
   "name": ".psl_test_endpoint(test).coverage0",
   "file": "psl_test_endpoint.vhd",
   "line": 53,
   "count": 1,
   "status": "covered"},
 { "directive": "cover",
   "name": ".psl_test_endpoint(test).coverage1",
   "file": "psl_test_endpoint.vhd",
   "line": 54,
   "count": 1,
   "status": "covered"},
 { "directive": "cover",
   "name": ".psl_test_endpoint(test).coverage2",
   "file": "psl_test_endpoint.vhd",
   "line": 55,
   "count": 1,
   "status": "covered"}],
 "summary" : {
  "assert": 1,
  "assert-failure": 1,
  "assert-pass": 0,
  "cover": 3,
  "cover-failure": 0,
  "cover-pass": 3}
}
```
What I would like to add is pass counts not only for the summary, but also for each assert/cover detail.So, I can see if a assert was passed, when not, I have a coverage hole in my test case. It could look like this:

```json
{ "details" : [
 { "directive": "assertion",
   "name": ".psl_test_endpoint(test).endpoint_assert",
   "file": "psl_test_endpoint.vhd",
   "line": 52,
   "count-pass": 0,
   "count-failure": 1,
   "status": "failed"},
```

tgingold, 2016-03-19:
On 19/03/16 11:27, T. Meissner wrote:
> Yep, after I removed the |stop()|statement, the report is generated.
> Looks good, great :+1:

Thanks!

[...]

> What I would like to add is pass counts not only for the summary, but
> also for each assert/cover detail.So, I can see if a assert was passed,
> when not, I have a coverage hole in my test case. It could look like this:
>
> {"details"  : [
>   {"directive":"assertion",
>     "name":".psl_test_endpoint(test).endpoint_assert",
>     "file":"psl_test_endpoint.vhd",
>     "line":52,
>     "count-pass":0,
>     "count-failure":1,
>     "status":"failed"},

Unfortunately, the state machine only detects failure, not success.
I will investigate, but after endpoints.

tgingold, 2016-03-20:
On 18/03/16 23:28, T. Meissner wrote:
> My testcase for the psl-report, which doesn't produce psl-export file:

This is now fixed.

Thanks!
Tristan.

tmeissner, 2016-03-22:
It works. Thanks. BTW: I have a new repository where I collect all my test cases: https://github.com/tmeissner/vhdl_verification

tgingold, 2016-03-22:
On 22/03/16 14:34, T. Meissner wrote:
> It works. Thanks. BTW: I have a new repository where I collect all my
> test cases: https://github.com/tmeissner/vhdl_verification

Ok, I will bookmark it.

I have just pushed the enhancement for endpoints: it is now possible
to refer to an endpoint in vhdl code.

Tristan.

tmeissner, 2016-03-22:
Thanks :+1: 

For my test test extended by a VHDL process reading the endpoint:
https://github.com/tmeissner/vhdl_verification/blob/master/psl_endpoint_eval_in_vhdl/psl_endpoint_eval_in_vhdl.vhd)
I get following error:

```
$ make compile
Analyse & elaborate test ...
cd work; ghdl -a --std=08 -fpsl ../psl_endpoint_eval_in_vhdl.vhd
Assertion failed: ((i >= FTy->getNumParams() || FTy->getParamType(i) == Args[i]->getType()) && "Calling a function with a bad signature!"), function init, file Instructions.cpp, line 281.

Execution terminated by unhandled exception
Exception name: PROGRAM_ERROR
Message: unhandled signal
Load address: 0x10b306000
Call stack traceback locations:
0x10bc9ad27 0x7fff9653fea8
ghdl: compilation error
make: *** [compile] Error 1

```

tgingold, 2016-03-23:
On 22/03/16 22:03, T. Meissner wrote:
> Thanks :+1:
>
> For my test test extended by a VHDL process reading the endpoint:
> https://github.com/tmeissner/vhdl_verification/blob/master/psl_endpoint_eval_in_vhdl/psl_endpoint_eval_in_vhdl.vhd)
> I get following error:

I am fixing the crash, but your code should be: wait until E_TEST0;

Tristan.

tmeissner, 2016-03-23:
Yep, my fault. It works now, thanks :)

closed 2016-03-23 by tgingold

Masshat, 2016-09-26:
Hi Guys, with the last version of GHDL 0.34Dev, i have this error when compiling some code of @tmeissner : https://github.com/tmeissner/vhdl_verification/tree/master/osvvm_fsm_coverage

ghdl:warning: library std does not exists for v08
ghdl:warning: library ieee does not exists for v08
/usr/lib/gcc/x86_64-unknown-linux-gnu/4.9.3/ghdl1: cannot find "std" library
ghdl: compilation error

according to @tmeissner  there is no error in his side, here is also the PKGBUILD Used to build ghdl: https://aur.archlinux.org/packages/ghdl-gcc-git/

@tgingold  Do you have some idea tohow to debug this issue?
Thanks in advance.

tgingold, 2016-09-26:
On 26/09/16 20:17, Massine wrote:
> Hi Guys, with the last version of GHDL 0.34Dev, i have this error when
> compiling some code of @tmeissner <https://github.com/tmeissner> :
> https://github.com/tmeissner/vhdl_verification/tree/master/osvvm_fsm_coverage
>
> ghdl⚠️ library std does not exists for v08
> ghdl⚠️ library ieee does not exists for v08
> /usr/lib/gcc/x86_64-unknown-linux-gnu/4.9.3/ghdl1: cannot find "std" library
> ghdl: compilation error
>
> according to @tmeissner <https://github.com/tmeissner> there is no error
> in his side, here is also the PKGBUILD Used to build ghdl:
> https://aur.archlinux.org/packages/ghdl-gcc-git/
>
> @tgingold <https://github.com/tgingold> Do you have some idea tohow to
> debug this issue?

Looks like the archlinux package is not complete: vhdl 2008 libraries 
are missing, or the installation is not correct.

You can try to use the latest beta from github.

Tristan.

Masshat, 2016-09-26:
OK @tgingold I will try this and keep you informed.