ghdl/ghdl #47: universal integer bound must be numeric literal or attribute using 93 and 02

kraigher, 2016-03-18:
**bug.vhd:**
```vhdl
package pkg is
  procedure proc(s : string);
end package;

package body pkg is
  procedure proc(s : string) is
  begin
    for i in 0 to s'length - 1 loop
    end loop;
  end procedure;
end package body;
```

**console output:**
```console
> ghdl -a --std=93 bug.vhd 
bug.vhd:8:16: universal integer bound must be numeric literal or attribute
ghdl: compilation error
> ghdl -a --std=02 bug.vhd 
bug.vhd:8:16:warning: universal integer bound must be numeric literal or attribute
> ghdl -a --std=08 bug.vhd 
> 
```
The about code works well in other simulators using all VHDL standards. Does it have to be an error? There are lot of such code out there.

tgingold, 2016-03-18:
On 18/03/16 07:22, kraigher wrote:
> *bug.vhd:*
>
> package  pkg  is
>    procedure  proc(s :string);
> end  package;
>
> package  body  pkg  is
>    procedure  proc(s :string)is
>    begin
>      for  iin  0  to  s'length  -  1  loop
>      end  loop;
>    end  procedure;
> end  package  body;
>
> *console output:*
>
>>ghdl -a --std=93 bug.vhd
> bug.vhd:8:16: universal integer bound must be numeric literal or attribute
> ghdl: compilation error
>>ghdl -a --std=02 bug.vhd
> bug.vhd:8:16:warning: universal integer bound must be numeric literal or
> attribute
>>ghdl -a --std=08 bug.vhd
>>
>
> The about code works well in other simulators using all VHDL standards.
> Does it have to be an error? There are lot of such code out there.

It's clearly not allowed by the vhdl 93 standard.
Why are you using --std=93 instead of --std=93c which relaxes such weird 
rules ?

kraigher, 2016-03-18:
Ok I can use 93c wherever I need 93.

closed 2016-03-18 by kraigher