ghdl/ghdl #48: Stack overflow (or erroneous memory access)

Bamban, 2016-03-22:
Hi !

I got some issue when simulating VHDL produced by an HLS tool (AUGH). I can repeat the error from where I simulates but I was unable to make it happen again when concatenating file in one directory to create a tar archive. However, here is what ghdl tells me:
```
/home/bourgeal/local/opt/ghdl-mcode/bin/ghdl -c --workdir=work -v ../vhd/idct.d/vhdl/*.vhd ./src/*.vhd -r tb
Linking in memory

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 4.6
In directory: /home/bourgeal/repo/augh_testfiles/dft/
Command line:
/home/bourgeal/local/opt/ghdl-mcode/bin/ghdl -c --workdir=work -v ../vhd/idct.d/vhdl/add_163.vhd ../vhd/idct.d/vhdl/add_165.vhd ../vhd/idct.d/vhdl/add_166.vhd ../vhd/idct.d/vhdl/add_167.vhd ../vhd/idct.d/vhdl/add_168.vhd ../vhd/idct.d/vhdl/add_169.vhd ../vhd/idct.d/vhdl/add_170.vhd ../vhd/idct.d/vhdl/add_171.vhd ../vhd/idct.d/vhdl/add_172.vhd ../vhd/idct.d/vhdl/add_173.vhd ../vhd/idct.d/vhdl/add_174.vhd ../vhd/idct.d/vhdl/add_175.vhd ../vhd/idct.d/vhdl/add_176.vhd ../vhd/idct.d/vhdl/add_177.vhd ../vhd/idct.d/vhdl/add_178.vhd ../vhd/idct.d/vhdl/add_179.vhd ../vhd/idct.d/vhdl/add_180.vhd ../vhd/idct.d/vhdl/add_181.vhd ../vhd/idct.d/vhdl/add_182.vhd ../vhd/idct.d/vhdl/add_183.vhd ../vhd/idct.d/vhdl/add_184.vhd ../vhd/idct.d/vhdl/add_191.vhd ../vhd/idct.d/vhdl/add_194.vhd ../vhd/idct.d/vhdl/add_205.vhd ../vhd/idct.d/vhdl/add_206.vhd ../vhd/idct.d/vhdl/add_210.vhd ../vhd/idct.d/vhdl/add_211.vhd ../vhd/idct.d/vhdl/add_212.vhd ../vhd/idct.d/vhdl/add_226.vhd ../vhd/idct.d/vhdl/add_235.vhd ../vhd/idct.d/vhdl/add_236.vhd ../vhd/idct.d/vhdl/add_254.vhd ../vhd/idct.d/vhdl/add_255.vhd ../vhd/idct.d/vhdl/add_276.vhd ../vhd/idct.d/vhdl/add_277.vhd ../vhd/idct.d/vhdl/add_295.vhd ../vhd/idct.d/vhdl/add_296.vhd ../vhd/idct.d/vhdl/add_314.vhd ../vhd/idct.d/vhdl/add_315.vhd ../vhd/idct.d/vhdl/add_323.vhd ../vhd/idct.d/vhdl/add_338.vhd ../vhd/idct.d/vhdl/add_350.vhd ../vhd/idct.d/vhdl/add_382.vhd ../vhd/idct.d/vhdl/add_390.vhd ../vhd/idct.d/vhdl/add_393.vhd ../vhd/idct.d/vhdl/add_419.vhd ../vhd/idct.d/vhdl/add_420.vhd ../vhd/idct.d/vhdl/add_430.vhd ../vhd/idct.d/vhdl/add_466.vhd ../vhd/idct.d/vhdl/add_474.vhd ../vhd/idct.d/vhdl/add_483.vhd ../vhd/idct.d/vhdl/add_487.vhd ../vhd/idct.d/vhdl/add_508.vhd ../vhd/idct.d/vhdl/add_516.vhd ../vhd/idct.d/vhdl/add_537.vhd ../vhd/idct.d/vhdl/add_552.vhd ../vhd/idct.d/vhdl/add_564.vhd ../vhd/idct.d/vhdl/add_570.vhd ../vhd/idct.d/vhdl/add_600.vhd ../vhd/idct.d/vhdl/add_601.vhd ../vhd/idct.d/vhdl/add_602.vhd ../vhd/idct.d/vhdl/cmp_164.vhd ../vhd/idct.d/vhdl/cmp_202.vhd ../vhd/idct.d/vhdl/cmp_203.vhd ../vhd/idct.d/vhdl/cmp_204.vhd ../vhd/idct.d/vhdl/cmp_324.vhd ../vhd/idct.d/vhdl/cmp_325.vhd ../vhd/idct.d/vhdl/cmp_392.vhd ../vhd/idct.d/vhdl/cmp_396.vhd ../vhd/idct.d/vhdl/cmp_398.vhd ../vhd/idct.d/vhdl/cmp_400.vhd ../vhd/idct.d/vhdl/cmp_402.vhd ../vhd/idct.d/vhdl/cmp_404.vhd ../vhd/idct.d/vhdl/cmp_406.vhd ../vhd/idct.d/vhdl/cmp_408.vhd ../vhd/idct.d/vhdl/cmp_410.vhd ../vhd/idct.d/vhdl/cmp_411.vhd ../vhd/idct.d/vhdl/cmp_412.vhd ../vhd/idct.d/vhdl/cmp_413.vhd ../vhd/idct.d/vhdl/fsm_23.vhd ../vhd/idct.d/vhdl/input_split0.vhd ../vhd/idct.d/vhdl/input_split1.vhd ../vhd/idct.d/vhdl/mul_189.vhd ../vhd/idct.d/vhdl/mul_190.vhd ../vhd/idct.d/vhdl/mul_192.vhd ../vhd/idct.d/vhdl/mul_193.vhd ../vhd/idct.d/vhdl/mul_195.vhd ../vhd/idct.d/vhdl/mul_196.vhd ../vhd/idct.d/vhdl/mul_198.vhd ../vhd/idct.d/vhdl/mul_199.vhd ../vhd/idct.d/vhdl/mul_215.vhd ../vhd/idct.d/vhdl/mul_216.vhd ../vhd/idct.d/vhdl/mul_218.vhd ../vhd/idct.d/vhdl/mul_219.vhd ../vhd/idct.d/vhdl/mul_223.vhd ../vhd/idct.d/vhdl/mul_230.vhd ../vhd/idct.d/vhdl/mul_328.vhd ../vhd/idct.d/vhdl/mul_331.vhd ../vhd/idct.d/vhdl/mul_341.vhd ../vhd/idct.d/vhdl/mul_344.vhd ../vhd/idct.d/vhdl/mul_353.vhd ../vhd/idct.d/vhdl/mul_357.vhd ../vhd/idct.d/vhdl/mul_365.vhd ../vhd/idct.d/vhdl/mul_368.vhd ../vhd/idct.d/vhdl/mul_373.vhd ../vhd/idct.d/vhdl/mul_376.vhd ../vhd/idct.d/vhdl/mul_383.vhd ../vhd/idct.d/vhdl/mul_416.vhd ../vhd/idct.d/vhdl/mul_442.vhd ../vhd/idct.d/vhdl/mul_445.vhd ../vhd/idct.d/vhdl/mul_447.vhd ../vhd/idct.d/vhdl/mul_448.vhd ../vhd/idct.d/vhdl/mul_456.vhd ../vhd/idct.d/vhdl/mul_457.vhd ../vhd/idct.d/vhdl/mul_460.vhd ../vhd/idct.d/vhdl/mul_469.vhd ../vhd/idct.d/vhdl/mul_477.vhd ../vhd/idct.d/vhdl/mul_492.vhd ../vhd/idct.d/vhdl/mul_495.vhd ../vhd/idct.d/vhdl/mul_499.vhd ../vhd/idct.d/vhdl/mul_502.vhd ../vhd/idct.d/vhdl/mul_511.vhd ../vhd/idct.d/vhdl/mul_520.vhd ../vhd/idct.d/vhdl/mul_524.vhd ../vhd/idct.d/vhdl/mul_527.vhd ../vhd/idct.d/vhdl/mul_531.vhd ../vhd/idct.d/vhdl/mul_534.vhd ../vhd/idct.d/vhdl/mul_540.vhd ../vhd/idct.d/vhdl/mul_543.vhd ../vhd/idct.d/vhdl/mul_547.vhd ../vhd/idct.d/vhdl/mul_556.vhd ../vhd/idct.d/vhdl/mul_559.vhd ../vhd/idct.d/vhdl/mul_560.vhd ../vhd/idct.d/vhdl/mul_561.vhd ../vhd/idct.d/vhdl/mul_565.vhd ../vhd/idct.d/vhdl/mul_566.vhd ../vhd/idct.d/vhdl/mul_573.vhd ../vhd/idct.d/vhdl/mul_577.vhd ../vhd/idct.d/vhdl/mul_578.vhd ../vhd/idct.d/vhdl/mul_579.vhd ../vhd/idct.d/vhdl/mul_589.vhd ../vhd/idct.d/vhdl/mul_592.vhd ../vhd/idct.d/vhdl/mul_594.vhd ../vhd/idct.d/vhdl/mul_595.vhd ../vhd/idct.d/vhdl/mul_605.vhd ../vhd/idct.d/vhdl/muxb_162.vhd ../vhd/idct.d/vhdl/muxb_201.vhd ../vhd/idct.d/vhdl/muxb_263.vhd ../vhd/idct.d/vhdl/muxb_265.vhd ../vhd/idct.d/vhdl/muxb_322.vhd ../vhd/idct.d/vhdl/output_split0.vhd ../vhd/idct.d/vhdl/output_split1.vhd ../vhd/idct.d/vhdl/output_split2.vhd ../vhd/idct.d/vhdl/output_split3.vhd ../vhd/idct.d/vhdl/output_split4.vhd ../vhd/idct.d/vhdl/output_split5.vhd ../vhd/idct.d/vhdl/output_split6.vhd ../vhd/idct.d/vhdl/output_split7.vhd ../vhd/idct.d/vhdl/sub_157.vhd ../vhd/idct.d/vhdl/sub_159.vhd ../vhd/idct.d/vhdl/sub_160.vhd ../vhd/idct.d/vhdl/sub_185.vhd ../vhd/idct.d/vhdl/sub_186.vhd ../vhd/idct.d/vhdl/sub_187.vhd ../vhd/idct.d/vhdl/sub_188.vhd ../vhd/idct.d/vhdl/sub_197.vhd ../vhd/idct.d/vhdl/sub_200.vhd ../vhd/idct.d/vhdl/sub_207.vhd ../vhd/idct.d/vhdl/sub_208.vhd ../vhd/idct.d/vhdl/sub_209.vhd ../vhd/idct.d/vhdl/sub_213.vhd ../vhd/idct.d/vhdl/sub_214.vhd ../vhd/idct.d/vhdl/sub_217.vhd ../vhd/idct.d/vhdl/sub_220.vhd ../vhd/idct.d/vhdl/sub_227.vhd ../vhd/idct.d/vhdl/sub_243.vhd ../vhd/idct.d/vhdl/sub_262.vhd ../vhd/idct.d/vhdl/sub_284.vhd ../vhd/idct.d/vhdl/sub_303.vhd ../vhd/idct.d/vhdl/sub_332.vhd ../vhd/idct.d/vhdl/sub_337.vhd ../vhd/idct.d/vhdl/sub_345.vhd ../vhd/idct.d/vhdl/sub_354.vhd ../vhd/idct.d/vhdl/sub_362.vhd ../vhd/idct.d/vhdl/sub_369.vhd ../vhd/idct.d/vhdl/sub_370.vhd ../vhd/idct.d/vhdl/sub_377.vhd ../vhd/idct.d/vhdl/sub_391.vhd ../vhd/idct.d/vhdl/sub_429.vhd ../vhd/idct.d/vhdl/sub_437.vhd ../vhd/idct.d/vhdl/sub_446.vhd ../vhd/idct.d/vhdl/sub_449.vhd ../vhd/idct.d/vhdl/sub_461.vhd ../vhd/idct.d/vhdl/sub_478.vhd ../vhd/idct.d/vhdl/sub_484.vhd ../vhd/idct.d/vhdl/sub_488.vhd ../vhd/idct.d/vhdl/sub_489.vhd ../vhd/idct.d/vhdl/sub_496.vhd ../vhd/idct.d/vhdl/sub_503.vhd ../vhd/idct.d/vhdl/sub_517.vhd ../vhd/idct.d/vhdl/sub_521.vhd ../vhd/idct.d/vhdl/sub_528.vhd ../vhd/idct.d/vhdl/sub_544.vhd ../vhd/idct.d/vhdl/sub_553.vhd ../vhd/idct.d/vhdl/sub_562.vhd ../vhd/idct.d/vhdl/sub_563.vhd ../vhd/idct.d/vhdl/sub_567.vhd ../vhd/idct.d/vhdl/sub_574.vhd ../vhd/idct.d/vhdl/sub_580.vhd ../vhd/idct.d/vhdl/sub_585.vhd ../vhd/idct.d/vhdl/sub_586.vhd ../vhd/idct.d/vhdl/sub_593.vhd ../vhd/idct.d/vhdl/sub_596.vhd ../vhd/idct.d/vhdl/sub_599.vhd ../vhd/idct.d/vhdl/top.vhd ./src/assert_uut.vhd ./src/cp3_test.vhd ./src/fsm.vhd ./src/pkg_tb.vhd ./src/prog.vhd ./src/sync_ram.vhd ./src/tb.vhd -r tb
Exception STORAGE_ERROR raised
Exception information:
Exception name: STORAGE_ERROR
Message: stack overflow (or erroneous memory access)
Call stack traceback locations:
0x2aeec4f9519e 0x53bb51 0x559d19 0x562ee6 0x565b41 0x4baf79 0x55bc77 0x5f8663 0x5fd18a 0x5fb905 0x4bb2c1 0x408b4f 0x2aeec4f81eab 0x408137
******************************************************************
```

I tried to copy every source file into one single directory and use ghdl gain and then > no error :/

```
$ uname -a
Linux sequoia 3.2.0-4-amd64 #1 SMP Debian 3.2.73-2+deb7u3 x86_64 GNU/Linux
```

Tell me if you need mode info !

tgingold, 2016-03-22:
On 22/03/16 09:06, Albbrg wrote:
> Hi !
>
> I got some issue when simulating VHDL produced by an HLS tool (AUGH). I
> can repeat the error from where I simulates but I was unable to make it
> happen again when concatenating file in one directory to create a tar
> archive. However, here is what ghdl tells me:

[...]

Can you send me the sources, so that I can investigate ?

I understand that this should be quite large, so do not
hesitate to send me a tarball privately.

Thanks,
Tristan.

Bamban, 2016-04-01:
Sorry for the late reply,
I am unable to recreate the bug ... see (this new issue)[https://github.com/tgingold/ghdl/issues/50] :/

tgingold, 2016-04-01:
On 01/04/16 09:59, Albbrg wrote:
> Sorry for the late reply,
> I am unable to recreate the bug ... see (this new issue)[#50
> <https://github.com/tgingold/ghdl/issues/50>] :/

Ok, so I close this one.

closed 2016-04-01 by tgingold

Bamban, 2016-04-04:
Following the resolution of  [this issue](https://github.com/tgingold/ghdl/issues/50) that wasn't really one, the bug appears once again.
I will try to provide some more info about it.

dtp555-1212, 2018-11-18:
[vhdljunk.zip](https://github.com/ghdl/ghdl/files/2593007/vhdljunk.zip)

I got the same stack overflow (not a swap) with the attached file on V 0.35.