ghdl/ghdl #49: Really slow analyzing of memories with initialisations

gardners, 2016-04-01:
I am finding that GHDL is VERY slow analyzing some VHDL memory files, that I populate with values via the VHDL file.  It takes minutes on a 2.7GHz Core i7 to analyze each 64K x 9-bit memory.

To reproduce, type make cpu_test using the following:

https://github.com/gardners/gs4502b/commit/f3fd55365aeed8a8b2d52796962207d417526d1b

Paul.

gardners, 2016-04-01:
I should also say, this is the version of GHDL I am using (on a Mac):

ghdl --version
GHDL 0.31 (20140108) [Dunoon edition]
 Compiled with GNAT Version: GPL 2013 (20130314)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2014 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I see there is a version 0.34 now, which I will also try to build and test this problem with.

tgingold, 2016-04-01:
On 01/04/16 06:51, gardners wrote:
> I am finding that GHDL is VERY slow analyzing some VHDL memory files,
> that I populate with values via the VHDL file. It takes minutes on a
> 2.7GHz Core i7 to analyze each 64K x 9-bit memory.

Indeed, I can reproduce that.

Using the mcode backend, it takes only a couple of seconds.

I will investigate.

Tristan.

gardners, 2016-04-01:
Thanks for the speedy response. 

Meanwhile, I'm building 0.34 with the mcode back-end, as I don't care as much about the simulation speed, so I have an interim solution.

Paul.

tgingold, 2016-04-01:
I will mark this ticket as an enhancement.

I don't plan to work now on it, but I hope it will be fixed as a side effect of another change.

You could also read the ram content from a file during initialisation.
There is something that is not common in your ram: they are pre-initialised and their size is not static.

Thank you for the report,
Tristan.

gardners, 2016-04-02:
Yes, you are right that I should have stripped out the generics, since I am initialising the memories.  I'll strip that out.  It might be that that just fixes the problem.

I'd rather not read from a file, in case it causes problem with synthesis under ISE, but that is probably just me being paranoid.  It certainly should work.

Paul.

tgingold, 2016-04-02:
On 02/04/16 08:31, gardners wrote:
> Yes, you are right that I should have stripped out the generics, since I
> am initialising the memories. I'll strip that out. It might be that that
> just fixes the problem.

not sure it will fix the problem, but it will be easier to tackle.

> I'd rather not read from a file, in case it causes problem with
> synthesis under ISE, but that is probably just me being paranoid. It
> certainly should work.

Good point.

Tristan.

gardners, 2016-04-02:
Confirming that removing the generics did not solve the problem.

diogratia, 2016-04-14:
This looks like the memory allocation bug fixed not too long ago.

> On 15/03/16 00:40, Adrien Prost-Boucle wrote:
> Hi,
> 
> Waow.
> I know this isn't the best mux code ever, but I didn't imagine it could
> be that painful...
> 
> Adrien,
> 
> this issue is now fixed by my latest patches.
> There was a threshold effect that prevented memory reuse and therefore
> slowed down simulation.
> 
> The output of your HLS tool is interesting as it could really be
> optimized by simulation tools.
> 
> Thank you for your testcase,
> Tristan.
> 

(This didn't have a ticket number)

These would be New Zealand date stamps and would correspond to:

> Commit: c14f26236d1d3cbdf6d0503f19d0a6a69907d35a [c14f262]
> Parents: 309014dcb9
> Author: Tristan Gingold <tgingold@free.fr>
> Date: 15 March 2016 7:43:53 pm NZDT
> 
> Improve behaviour of secondary stack.

> Commit: f4a61202e89d2e29416ae9a4b59bea665336c325 [f4a6120]
> Parents: c14f26236d
> Author: Tristan Gingold <tgingold@free.fr>
> Date: 15 March 2016 8:12:36 pm NZDT
> 
> stack2: use a minimal size for chunks.

and 

> Commit: 37192248646ce7b4688f105877449c640e5039ce [3719224]
> Parents: f4a61202e8
> Author: Tristan Gingold <tgingold@free.fr>
> Date: 16 March 2016 8:28:56 am NZDT
> 
> Add bug040 testcase.
>  
>  
>  
> Fix speed issue of bug040.

While The github model isn't currently in a state to run the test:
>....  
> ghdl  -m cpu_test
> ram0.vhdl:29:50:warning: universal integer bound must be numeric literal or attribute
> ram1.vhdl:29:50:warning: universal integer bound must be numeric literal or attribute
> ram2.vhdl:29:50:warning: universal integer bound must be numeric literal or attribute
> ram3.vhdl:29:50:warning: universal integer bound must be numeric literal or attribute
> gs4502b_instruction_prefetch.vhdl:402:14: no declaration for "get_addressing_modes"
> gs4502b_instruction_prefetch.vhdl:407:14: no declaration for "get_addressing_modes"
> make: *** [cpu_test] Error 1

Looking at analyzing one of the RAMs with initial values of all `"000000000"`:
david_koontz@Macbook: time ghdl -a ram0.vhdl

ram0.vhdl:29:50:warning: universal integer bound must be numeric literal or attribute

ghdl1--llvm (ghdl-0.33 llvm) is eating one core (macbook 5,1, 2.4 GHz, Core 2 Duo, 8 GB, late 2008 Aluminum Macbook, 99.2 % CPU) .

I commented out all the index value expression association except the last and the RAM analyzed in seconds (or less). While the timed process is still going strong after around 45 minutes.

Sampling the process shows it's futzing with malloc and realloc.

> namespace)::JoinVals::computeAssignment(unsigned int, (anonymous namespace)::JoinVals&)  (in ghdl1-llvm) + 2118  [0x10c6a6e56]
>                           +   14 llvm::SmallVectorBase::grow_pod(void*, unsigned long, unsigned long)  (in ghdl1-llvm) + 61  [0x10cb68f4d]
>                           +     13 realloc  (in libsystem_malloc.dylib) + 235  [0x7fff8f85f179]
>                           +     : 13 malloc_zone_realloc  (in libsystem_malloc.dylib) + 78  [0x7fff8f85c5cf]
>                           +     :   4 szone_realloc  (in libsystem_malloc.dylib) + 917  [0x7fff8f85c995]
>                           +     :   | 3 szone_malloc_should_clear  (in libsystem_malloc.dylib) + 1426  [0x7fff8f858c39]
>                           +     :   | + 3 large_malloc  (in libsystem_malloc.dylib) + 1042  [0x7fff8f85f877]
>                           +     :   | +   3 allocate_pages  (in libsystem_malloc.dylib) + 328  [0x7fff8f857e0d]
>                           +     :   | +     3 mach_vm_map  (in libsystem_kernel.dylib) + 85  [0x7fff97a52278]
>                           +     :   | +       3 _kernelrpc_mach_vm_map_trap  (in libsystem_kernel.dylib) + 10  [0x7fff97a4e42a]
>                           +     :   | 1 szone_malloc_should_clear  (in libsystem_malloc.dylib) + 1449  [0x7fff8f858c50]
>                           +     :   |   1 small_malloc_from_free_list  (in libsystem_malloc.dylib) + 44  [0x7fff8f85bec0]
>                           +     :   3 szone_realloc  (in libsystem_malloc.dylib) + 994  [0x7fff8f85c9e2]
>                           +     :   | 3 _platform_memmove$VARIANT$Merom  (in libsystem_platform.dylib) + 228,50  [0x7fff8f262324,0x7fff8f262272]
>                           +     :   2 szone_realloc  (in libsystem_malloc.dylib) + 976  [0x7fff8f85c9d0]
>                           +     :   | 2 vm_copy  (in libsystem_kernel.dylib) + 97  [0x7fff97a4c21c]
>                           +     :   |   2 mach_msg  (in libsystem_kernel.dylib) + 55  [0x7fff97a4d64f]
>                           +     :   |     2 mach_msg_trap  (in libsystem_kernel.dylib) + 10  [0x7fff97a4e4de]
>                           +     :   2 szone_realloc  (in libsystem_malloc.dylib) + 1005  [0x7fff8f85c9ed]
>                           +     :   | 2 madvise  (in libsystem_kernel.dylib) + 10  [0x7fff97a5430a]
>                           +     :   1 szone_realloc  (in libsystem_malloc.dylib) + 55  [0x7fff8f85c637]
>                           +     :   | 1 szone_size  (in libsystem_malloc.dylib) + 611  [0x7fff8f85a430]
>                           +     :   1 szone_realloc  (in libsystem_malloc.dylib) + 1785  [0x7fff8f85ccf9]
>                           +     :     1 vm_allocate  (in

If you're not encountering this issue with a ghdl-0.34dev mcode version it's likley built from a commit after March 15th (NZ time).

The timed ghdl -a is still going strong after 46 minutes.

gardners, 2016-04-14:
Either the git commit ID I supplied, or the current head (commit c5b5ff422069e3b5fdaa781653d9f26c32327545) should be viable options for testing.

However, if I understand correctly, the current head of GHDL should have this problem solved. Have I understood correctly?

diogratia, 2016-04-14:
Yes the current head should have the problem solved (it's worth testing).

> time ghdl -a ram0.vhdl
> ram0.vhdl:29:50:warning: universal integer bound must be numeric literal or attribute
> 
> 3702.165u 77.126s 1:05:56.39 95.5%	0+0k 0+4io 2pf+0w
So it took over an an hour of real time and little actual ghdl run time. Looks like it spent most of it's time allocating and reallocating memory for the 65537 expressions in the default value (that happened to be all zeros in this case).

I haven't built an mcode or llvm version since before my March 15th. I may have to build an llvm for the 0.34 build to overcome a non-Apple llvm bug that shows up with Xcode 7. (Apple doesn't distribute a full llvm tool set).

gardners, 2016-04-14:
Ok. I'm not yet sure when I will be able to sit down and rebuild GHDL here to test it, but I look forward to doing so.

Paul.

gardners, 2017-08-23:
Finally got around to compiling GHDL from the current head (commit 0d72d5def31ba701bfce8750820bcd9bb9f4ad5c)
Testing with https://github.com/MEGA65/mega65-core/commit/0ea1ec67a6dac122061b3ba408faae55e611792c, it still presents the same problem of taking minutes to analyse shadowram.vhdl.

use:

make simulate

diogratia, 2017-08-23:
There appears to be a disconnect here between what you write and  the problem you are experiencing. With the shadowram.vhdl file from MEGA65....

time ghdl -a shadowram.vhdl
0.115u 0.017s 0:00.14 85.7%	0+0k 0+5io 0pf+0w

time ghdl_mcode -a shadowram.vhdl
0.009u 0.004s 0:00.01 0.0%	0+0k 0+7io 0pf+0w

There doesn't appear to be an analysis problem with shadowram.vhdl. 

ghdl --version
GHDL 0.34 (tarball) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
  llvm code generator
(commit 06a78d2275dc837c7724c5dc5ede2e14ef67080b - 06a78d2 Aug 15. 2017, the ghdl-0.34 release https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-0.34-llvm-darwin13.pkg, built today. )

ghdl_mcode --version
GHDL 0.34 (tarball) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 mcode code generator  
(built for this comment, same commit).

Can you narrow this down a bit more than `make simulate`, something that demonstrates a problem with ghdl specifically?

gardners, 2017-08-23:
I'm now running:
time ghdl -a vhdl/shadowram.vhdl

real	2m12.687s
user	2m12.512s
sys	0m0.116s

This is on Ubuntu 16.04 on a recent i7, ghdl LLVM built from 0d72d5def31ba701bfce8750820bcd9bb9f4ad5c:

paul@F96NG92-L:~/Projects/mega65/ghdl$ ghdl --version
GHDL 0.35-dev (2017-03-01-207-g0d72d5d) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
llvm code generator
Written by Tristan Gingold.
 
Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I'll try it in a moment with ghdl-0.34 release.

gardners, 2017-08-23:
Same behaviour:

paul@F96NG92-L:~/Projects/mega65/mega65-core/src$ time ghdl -a vhdl/shadowram.vhdl

real	2m13.484s
user	2m13.360s
sys	0m0.088s
paul@F96NG92-L:~/Projects/mega65/mega65-core/src$ ghdl --version
GHDL 0.34 (2017-03-01-203-g06a78d2) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

md5sum vhdl/shadowram.vhdl 
168695dbb90c4329d56f646352ba55c0  vhdl/shadowram.vhdl

Paebbels, 2017-08-23:
@gardners 
How many files have been analyzed before?
Did you run the time command in a fresh directory?

I just want to eliminate side effects from previous runs for other files.

diogratia, 2017-08-23:
```
> Macbook: md5 vhdl/shadowram.vhdl
> MD5 (vhdl/shadowram.vhdl) = e98ab49e2c458a40e8594ae9e0db5352
```
The md5sum doesn't match what you provided.

Your commit appears 14 hours old about now. shadowram.vhdl was last modified 5 months ago. It's history shows it was moved from vhdl to src/vhdl and nothing else.
[mega65-core/shadowram.vhdl at 195ffb78e8dd9a44c0a93f0dcf2e5697d7fc0bc6 · MEGA65/mega65-core](https://github.com/MEGA65/mega65-core/blob/195ffb78e8dd9a44c0a93f0dcf2e5697d7fc0bc6/src/vhdl/shadowram.vhdl)

```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use ieee.numeric_std.all;

--
entity shadowram is
  port (Clk : in std_logic;
        address : in integer range 0 to 131071;
        we : in std_logic;
        data_i : in unsigned(7 downto 0);
        data_o : out unsigned(7 downto 0);
        writes : out unsigned(7 downto 0);
        no_writes : out unsigned(7 downto 0)
        );
end shadowram;

architecture Behavioral of shadowram is

  signal write_count : unsigned(7 downto 0) := x"00";
  signal no_write_count : unsigned(7 downto 0) := x"00";
  
--  type ram_t is array (0 to 262143) of std_logic_vector(7 downto 0);
  type ram_t is array (0 to 131071) of unsigned(7 downto 0);
  signal ram : ram_t := (
    4096 => x"03",
    4097 => x"08",
    4098 => x"01",
    4099 => x"04",
    4100 => x"0f",
    4101 => x"17",
    4102 => x"18",
    8192 => x"a9",
    8193 => x"01",
    8194 => x"a9",
    8195 => x"02",
    8196 => x"a9",
    8197 => x"03",
    8198 => x"a9",
    8199 => x"04",
    others => x"53"); 
begin

--process for read and write operation.
  PROCESS(Clk,ram,address)
  BEGIN
    data_o <= ram(address);
    writes <= write_count;
    no_writes <= no_write_count;
    if(rising_edge(Clk)) then 
      if we /= '0' then
        write_count <= write_count + 1;        
        ram(address) <= data_i;
        report "wrote to shadow ram" severity note;
      else
        no_write_count <= no_write_count + 1;        
      end if;
    end if;
  END PROCESS;

end Behavioral;
```
Please show the shadoram.vhdl you are using. If it's substantially similar I'd expect you have a platform issue. If not similar ...

gardners, 2017-08-23:
shadowram.vhdl is now auto-generated to contain pre-initialised memory.
I have attached the current version here.  I'll try a clean run shortly.

[shadowram.vhdl](https://github.com/tgingold/ghdl/files/1247253/shadowram.vhdl.txt)

gardners, 2017-08-23:
Confirmed problem still occurs in a fresh directory:

paul@F96NG92-L:~$ mkdir tmp
paul@F96NG92-L:~$ cd tmp
paul@F96NG92-L:tmp$ ls
paul@F96NG92-L:tmp$ cp ../Projects/mega65/mega65-core/src/vhdl/shadowram.vhdl .
paul@F96NG92-L:tmp$ time ghdl -a shadowram.vhdl 

real	2m27.397s
user	2m27.276s
sys	0m0.100s

diogratia, 2017-08-24:
Your shadowram with more elements in the initial value aggregate:

> ```
> Macbook: time ghdl -a shadowram.vhdl
> 297.587u 9.655s 5:15.89 97.2%	0+0k 8+13io 2318pf+0w
> Macbook: time ghdl -a shadowram.vhdl
> 287.253u 10.047s 5:06.09 97.1%	0+0k 0+11io 253pf+0w
> Macbook: time ghdl -a shadowram.vhdl
> 289.919u 9.952s 5:15.00 95.1%	0+0k 0+11io 208pf+0w
> Macbook: time ghdl -a shadowram.vhdl
> 285.059u 9.576s 5:01.76 97.6%	0+0k 0+12io 163pf+0w
> ```

The differences can be accounted by my surfing while waiting. You definitely have a gruntier machine.

The difference in your supplied shadowram.vhdl and the one in the github project is the number of elements in the initial value aggregate.

Technically these are supposed to be applied during elaboration (IEEE Std 1076-2008 14.4.2.5 Object declarations). It's not uncommon to rely on a bit found in 9.4 Static expressions, 9.4.1 General, para 2:

>There are two categories of static expression. Certain forms of expression can be evaluated during the analysis of the design unit in which they appear; such an expression is said to be _locally static_. Certain forms of expression can be evaluated as soon as the design hierarchy in which they appear is elaborated; such an expression is said to be _globally static_.

to give permission to evaluate the aggregate and produce a value of the type ram_t here during analysis. Incurring these delays during elaboration would tend to be a bit more of a squeeky wheel.

So, yes you're code exhibits a problem, no it isn't quite the same as Adrien originally encountered, there's no runaway memory leak, analysis completes. (I'd contend the page faults in my time output are caused by my surfing while analyzing shadowram.vhdl).

It would seem there's something inefficient in evaluating the aggregate. You'd expect the aggregate to be determined to be of subtype type_t (not further constrained). Each element of the aggregate would require evaluating the index value and the unsigned expression with a subtype constraint of (7 downto 0).

There are 20734 unique array element values, each an unsigned with 8 scalar elements (std_ulogic). While it may be possible to optimize the evaluation in ghdl the short term fix might be to use TEXTIO to initialize the ram, specifying an initial value from a function during elaboration.  It wouldn't entail allocating all the expression memory in traversing the aggregate elements.

 I'd suspect from the prevalence of simulator tools initializing RAM directly (using the equivalent of VHPI access) that there's a lower limit to how much effort an analyzer or elaboration can require here although ghdl likely hasn't reached that lower limit. 
 
Looking back through the comments here shows Tristan commenting on it only taking a few seconds with an mcode version, and indeed that's born out:

> ```
> Macbook: time ./ghdl_mcode -a shadowram.vhdl
> 0.243u 0.010s 0:00.25 100.0%	0+0k 0+0io 0pf+0w
> ```

So there's a definite difference in how that aggregate is evaluated during analysis between the llvm back end and an mcode version of ghdl.

gardners, 2017-08-24:
You're welcome to use the file as a test case :)

Meanwhile, I wonder if the issue is because of the others => taking all the zero values causing it to try to be over clever: That's it! Change it to initialise every value, instead of falling to others => x"00", and analysis is now <2 seconds:

paul@F96NG92-L:~/tmp$ time ghdl -a shadowram.vhdl 

real	0m1.180s
user	0m1.160s
sys	0m0.016s

[shadowram.vhdl.txt](https://github.com/tgingold/ghdl/files/1247763/shadowram.vhdl.txt)

So now provided Xilinx ISE still swallows it, we have a reasonable work around, and you have a negative control for your test case :)

gardners, 2017-08-25:
Putting the following patch back, seems to have fixed the reboot problem:

git checkout target/linux/generic/patches-3.18/092-01-spi-Check-to-see-if-the-device-is-processing-a-messa.patch

I'll check the performance soon.