ghdl/ghdl #50: GHDL makes me swap

Bamban, 2016-04-01:
Hi !
After the previous bug [here](https://github.com/tgingold/ghdl/issues/48), I tried to recreate the issue. Unfortunately, something worst happened ! My system swaps whenever I start a simulation now.
I am on commit 19793a92b4289 with mcode backend.

As I am unable to join some code archive to this issue, I'll send them to you by email ... There are 2: idct and vector, both output of AUGH.

The ghdl command I tried is:
```
$ /home/path/to/ghdl/ghdl-mcode/bin/ghdl -c -v *.vhd -r tb
```
And this leads to a huge system swap. Next, I tried it with gdb. idct stopped immediately with this : 
```
$ gdb --args /home/path/to/ghdl/... and all my vhds etc.
Linking in memory

Program received signal SIGSEGV, Segmentation fault.
0x000000000053bc93 in binary_file.gen_8 (b=0) at /home/bourgeal/local/opt/ghdl-build/src/ortho/mcode/binary_file.adb:368
368	      Cur_Sect.Data (Cur_Sect.Pc) := B;
(gdb) bt
#0  0x000000000053bc93 in binary_file.gen_8 (b=0) at /home/bourgeal/local/opt/ghdl-build/src/ortho/mcode/binary_file.adb:368
#1  0x0000000000559f8f in ortho_code.dwarf.finish () at /home/bourgeal/local/opt/ghdl-build/src/ortho/mcode/ortho_code-dwarf.adb:409
#2  0x000000000056315c in ortho_code.x86.emits.finish () at /home/bourgeal/local/opt/ghdl-build/src/ortho/mcode/ortho_code-x86-emits.adb:3253
#3  0x0000000000565db7 in ortho_code.x86.abi.finish () at /home/bourgeal/local/opt/ghdl-build/src/ortho/mcode/ortho_code-x86-abi.adb:682
#4  0x00000000004bb043 in ortho_mcode.finish () at /home/bourgeal/local/opt/ghdl-build/src/ortho/mcode/ortho_mcode.adb:719
#5  0x000000000055beed in ortho_jit.link (status=false) at /home/bourgeal/local/opt/ghdl-build/src/ortho/mcode/ortho_jit.adb:62
#6  0x00000000005f8be1 in ghdlrun.run () at /home/bourgeal/local/opt/ghdl-build/src/ghdldrv/ghdlrun.adb:575
#7  0x00000000005fd708 in ghdlcomp.perform_action (cmd=..., args=...) at /home/bourgeal/local/opt/ghdl-build/src/ghdldrv/ghdlcomp.adb:246
#8  0x00000000005fbe83 in ghdlmain.main () at /home/bourgeal/local/opt/ghdl-build/src/ghdldrv/ghdlmain.adb:342
#9  0x00000000004bb38b in ghdl_jit () at /home/bourgeal/local/opt/ghdl-build/src/ghdldrv/ghdl_jit.adb:34
```
The second (vector) swapped and was killed by my kernel without gdb complaining.

```
$ /home/path/to/ghdl/ghdl-mcode/bin/ghdl --version
GHDL 0.34dev (20151126) [Dunoon edition]
 Compiled with GNAT Version: 4.6
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
$ uname -a
Linux sequoia 3.2.0-4-amd64 #1 SMP Debian 3.2.73-2+deb7u3 x86_64 GNU/Linux
```

tgingold, 2016-04-01:
On 01/04/16 09:59, Albbrg wrote:
> Hi !
> After the previous bug here
> <https://github.com/tgingold/ghdl/issues/48>, I tried to recreate the
> issue. Unfortunately, something worst happened ! My system swaps
> whenever I start a simulation now.

The problem is in assert_uut.vhd
This unit creates a huge number of signals.  That should be a constant 
instead.

> I am on commit 19793a9
> <https://github.com/tgingold/ghdl/commit/19793a92b42898c81db8f069f777e95635130e36>
> with mcode backend.
>
> As I am unable to join some code archive to this issue, I'll send them
> to you by email ... There are 2: idct and vector, both output of AUGH.
>
> The ghdl command I tried is:
>
> |$ /home/path/to/ghdl/ghdl-mcode/bin/ghdl -c -v *.vhd -r tb |
>
> And this leads to a huge system swap. Next, I tried it with gdb. idct
> stopped immediately with this :

I wasn't able to reproduce the crash, but I see a potential issue.
I will fix it and we will see.

Tristan.

tgingold, 2016-04-01:
For vector, I can make it run using this assert_uut.vhd:

--test bench written by Alban Bourge @ TIMA
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;
library work;
use work.pkg_tb.all;

entity assert_uut is
	port(
				clock        : in std_logic;
				reset        : in std_logic;
				context_uut  : in context_t;
				en_feed      : in std_logic;
				stdin_rdy    : in std_logic;
				stdin_ack    : out std_logic;
				stdin_data   : out stdin_vector;
				en_check     : in std_logic;
				stdout_rdy   : in std_logic;
				stdout_ack   : out std_logic;
				stdout_data  : in stdout_vector;
				vecs_found   : out std_logic;
				vec_read     : out std_logic;
				n_error      : out std_logic
			);
end assert_uut;

architecture rtl of assert_uut is

	type vin_table  is array(0 to 2**VEC_NO_SIZE - 1) of stdin_vector;
	type vout_table is array(0 to 2**VEC_NO_SIZE - 1) of stdout_vector;
	constant input_vectors_1 : vin_table := (
		--##INPUT_VECTORS_1_GO_DOWN_HERE##--
		0 => x"00_00_00_07",
		1 => x"00_00_00_03",
		--##INPUT_VECTORS_1_GO_OVER_HERE##--
		others => (others => '0'));
	constant output_vectors_1 : vout_table := (
		--##OUTPUT_VECTORS_1_GO_DOWN_HERE##--
		0 => x"00_00_00_16",
		--##OUTPUT_VECTORS_1_GO_OVER_HERE##--
		others => (others => '0'));
	constant input_vectors_2 : vin_table := (
		--##INPUT_VECTORS_2_GO_DOWN_HERE##--
		0 => x"00_00_00_07",
		1 => x"00_00_00_03",
		--##INPUT_VECTORS_2_GO_OVER_HERE##--
		others => (others => '0'));
	constant output_vectors_2 : vout_table := (
		--##OUTPUT_VECTORS_2_GO_DOWN_HERE##--
		0 => x"00_00_00_16",
		--##OUTPUT_VECTORS_2_GO_OVER_HERE##--
		others => (others => '0'));

	signal out_vec_counter_1 : unsigned(VEC_NO_SIZE - 1 downto 0);
	signal out_vec_counter_2 : unsigned(VEC_NO_SIZE - 1 downto 0);
	signal stdin_ack_sig  : std_logic;
	signal vector_read    : std_logic;

begin

	feed : process(reset, clock) is
	variable in_vec_counter_1  : unsigned(VEC_NO_SIZE - 1 downto 0);
	variable in_vec_counter_2  : unsigned(VEC_NO_SIZE - 1 downto 0);
	begin
		if (reset = '1') then
			in_vec_counter_1 := (others => '0');
			in_vec_counter_2 := (others => '0');
			stdin_data       <= (others => '0');
			stdin_ack_sig    <= '0';
		elsif rising_edge(clock) then
			case context_uut is
				when "01" =>
					if (en_feed = '1') then
						stdin_data <= input_vectors_1(to_integer(in_vec_counter_1));
						stdin_ack_sig  <= '1';
						if (stdin_rdy = '1' and stdin_ack_sig = '1') then
							in_vec_counter_1 := in_vec_counter_1 + 1;
							stdin_ack_sig  <= '0';
						end if;
					else
						--in_vec_counter_1 <= (others => '0');
						stdin_data     <= (others => '0');
						stdin_ack_sig  <= '0';
					end if;
				when "10" =>
					if (en_feed = '1') then
						stdin_data <= input_vectors_2(to_integer(in_vec_counter_2));
						stdin_ack_sig  <= '1';
						if (stdin_rdy = '1' and stdin_ack_sig = '1') then
							in_vec_counter_2 := in_vec_counter_2 + 1;
							stdin_ack_sig  <= '0';
						end if;
					else
						--in_vec_counter_2 <= (others => '0');
						stdin_data     <= (others => '0');
						stdin_ack_sig  <= '0';
					end if;
				when others =>
			end case;
		end if;
	end process feed;

	check : process(reset, clock) is
	begin
		if (reset = '1') then
			n_error     <= '1';
			vec_read <= '0';
		elsif rising_edge(clock) then
			vec_read <= '0';
			if (en_check = '1') then
				if (stdout_rdy = '1') then
					vec_read <= '1';
					case context_uut is
						when "01" =>
							assert (stdout_data = 
output_vectors_1(to_integer(out_vec_counter_1)))
							report "ERROR ---> Bad output vector found";
							--synthesizable check
							if (stdout_data /= 
output_vectors_1(to_integer(out_vec_counter_1))) then
								n_error <= '0';
							end if;
						when "10" =>
							assert (stdout_data = 
output_vectors_2(to_integer(out_vec_counter_2)))
							report "ERROR ---> Bad output vector found";
							--synthesizable check
							if (stdout_data /= 
output_vectors_2(to_integer(out_vec_counter_2))) then
								n_error <= '0';
							end if;
						when others =>
					end case;
				end if;
			end if;
		end if;
	end process check;

	read_counter : process(reset, clock) is
	begin
		if (reset = '1') then
			out_vec_counter_1 <= (others => '0');
			out_vec_counter_2 <= (others => '0');
		elsif rising_edge(clock) then
			if (en_check = '1') then
				if (stdout_rdy = '1') then
					case context_uut is
						when "01" =>
							out_vec_counter_1 <= out_vec_counter_1 + 1;
						when "10" =>
							out_vec_counter_2 <= out_vec_counter_2 + 1;
						when others =>
					end case;
				end if;
			--else
			--	case context_uut is
			--		when "01" =>
			--			out_vec_counter_1 <= (others => '0');
			--		when "10" =>
			--			out_vec_counter_2 <= (others => '0');
			--		when others =>
			--	end case;
			end if;
		end if;
	end process read_counter;

	--asynchronous declarations
	stdout_ack <= en_check;
	stdin_ack <= stdin_ack_sig;
	vecs_found <= '1' when (out_vec_counter_1 /= 0 or out_vec_counter_2 /= 
0) else '0';

end rtl;

tgingold, 2016-04-01:
In a short, just make input_vector_X and output_vector_Y constant instead of signal.

Bamban, 2016-04-03:
That perfectly did the trick, thanks.
In conclusion that's my mistake: manipulating a huge table like that necessitate constants.
PS: ghdl handles differently array of signals and array of constants ?

closed 2016-04-03 by Bamban

tgingold, 2016-04-03:
On 03/04/16 12:10, Albbrg wrote:
> That perfectly did the trick, thanks.
> In conclusion that's my mistake: manipulating a huge table like that
> necessitate constants.
> PS: ghdl handles differently array of signals and array of constants ?

Oh yes.  There are metadata for signals, like drivers, last value, last 
event...

dtp555-1212, 2018-11-18:
[vhdljunk.zip](https://github.com/ghdl/ghdl/files/2592999/vhdljunk.zip)

I also got the stack overflow crash in V 0.35 ... see the attached file to recreate it.

Paebbels, 2018-11-18:
@dtp555-1212 please create a new issue, because this one is already closed. If you feel it its related, you can mark the new issues as related.