ghdl/ghdl #52: building instructions for gcc libbacktrace for working with llvm

flip111, 2016-04-02:
I downloaded the latest libbacktrace from https://github.com/gcc-mirror/gcc/tree/master/libbacktrace when i run configure i get:

```
configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."
```

i checked shtool is available on my prompt

```
> shtool --version
GNU shtool 2.0.8 (18-Jul-2008)
```

tgingold, 2016-04-03:
You also need to get the top-level files (the ones beside libbacktrace).
IIRC, you also need the config/ (or was it include/ ?) subdirectory.

closed 2016-04-03 by tgingold

flip111, 2016-04-03:
I'm confused by this [you don't need to configure gcc](https://github.com/tgingold/ghdl/blob/master/BUILD.txt#L101). So i don't need to configure gcc, but i still need to download all of gcc?

tgingold, 2016-04-03:
On 03/04/16 12:04, flip111 wrote:
> I'm confused by this you don't need to configure gcc
> <https://github.com/tgingold/ghdl/blob/master/BUILD.txt#L101>. So i
> don't need to configure gcc, but i still need to download all of gcc?

No, not all of gcc.  You can skip many subdirectories (like gcc/, the
libraries but libbacktrace).

flip111, 2016-04-04:
Well i downloaded all of gcc with `git clone --single-branch --branch master --depth=1 https://github.com/gcc-mirror/gcc.git` after that it worked, not sure which directories where needed specifically. However on building ghdl itself i'm getting this:

```
./configure --with-llvm-config --with-backtrace-lib=/home/flip111/Downloads/gcc/libbacktrace/.libs/libbacktrace.a
```
no errors here

```
make -j3
```

now i get halfway through:
```
clang++ -c `llvm-config --cxxflags` -o llvm-cbindings.o src/ortho/llvm/llvm-cbindings.cpp
/bin/sh: 1: clang++: not found
src/ortho/llvm/Makefile:17: recipe for target 'llvm-cbindings.o' failed
make[1]: *** [llvm-cbindings.o] Error 127
```

`clang++` is not installed on my system. So i guess that means that next to llvm (i downloaded this http://llvm.org/releases/3.5.2/llvm-3.5.2.src.tar.xz ) i also need to download and compile clang? If yes, do i also need that same version 3.5.2?

tgingold, 2016-04-04:
On 04/04/16 19:04, flip111 wrote:
> Well i downloaded all of gcc with |git clone --single-branch --branch
> master --depth=1 https://github.com/gcc-mirror/gcc.git| after that it
> worked, not sure which directories where needed [...]

No.  Try 'make -j3 CXX=g++'

Tristan.

flip111, 2016-04-04:
I tried your suggestion, these are some of the last lines of the make output

```
raw_ostream.cpp:(.text._ZN4llvm14raw_fd_ostreamC2EPKcRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_3sys2fs9OpenFlagsE+0x3c6): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long)'
/usr/local/lib/libLLVMSupport.a(Errno.o): In function `llvm::sys::StrError[abi:cxx11](int)':
Errno.cpp:(.text._ZN4llvm3sys8StrErrorB5cxx11Ei+0x64): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace(unsigned long, unsigned long, char const*, unsigned long)'
collect2: error: ld returned 1 exit status
gnatlink: error when calling /usr/gnat/bin/g++
gnatmake: *** link failed.
src/ortho/llvm/Makefile:11: recipe for target 'ghdl1-llvm' failed
make[1]: *** [ghdl1-llvm] Error 4
make[1]: Leaving directory '/home/flip111/Downloads/ghdl'
Makefile:220: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
```

tgingold, 2016-04-05:
On 04/04/16 21:18, flip111 wrote:
> I tried your suggestion, these are some of the last lines of the make output [...]

I see.
If you are on linux, you'd better to install clang, as your llvm 
libraries were built with it.

I am not sure that the clang++ and g++ c++ libraries are compatible.

Tristan.

flip111, 2016-04-05:
I downloaded clang and rebuild llvm (this is the recommended procedure if you want to compile clang). After that compiling seemed to succeed :+1: However now on the `make install` i get this

```
[ -d /usr/local ] || mkdir -p /usr/local
[ -d /usr/local/bin ] || mkdir -p /usr/local/bin
make -f ./src/ortho/llvm/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNAT_FLAGS="-aI./src -aI./src/vhdl -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/grt -aI./src/ortho -aI./src/ortho/llvm -gnaty3befhkmr -gnatwae -aO. -gnatf -gnat05 -g -gnata" LDFLAGS="" \
 LLVM_CONFIG="llvm-config" all
make[1]: Entering directory '/home/flip111/Downloads/ghdl'
gnatmake -o ghdl1-llvm -aI./src/ortho/llvm -aI./src/ortho \
-aI./src -aI./src/vhdl -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/grt -aI./src/ortho -aI./src/ortho/llvm -gnaty3befhkmr -gnatwae -aO. -gnatf -gnat05 -g -gnata ortho_code_main -bargs -E \
-largs llvm-cbindings.o --LINK=clang++ \
`llvm-config --ldflags --libs --system-libs` 
/bin/sh: 1: gnatmake: not found
src/ortho/llvm/Makefile:11: recipe for target 'ghdl1-llvm' failed
make[1]: *** [ghdl1-llvm] Error 127
make[1]: Leaving directory '/home/flip111/Downloads/ghdl'
Makefile:220: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
```
`gnatmake` is available on my command line

tgingold, 2016-04-06:
On 05/04/16 22:13, flip111 wrote:
> I downloaded clang and rebuild llvm (this is the recommended procedure
> if you want to compile clang). After that compiling seemed to succeed
> :+1: However now on the |make install| i get this
>
> |[ -d [...]

Are you sure of that ?  Can you try:
gnatmake -v; make install

Tristan.

flip111, 2016-04-07:
`gnatmake -v` doesn't seem to do much

```
» gnatmake -v

GNATMAKE GPL 2015 (20150428-49)
Copyright (C) 1992-2015, Free Software Foundation, Inc.
try "gnatmake --help" for more information.
```

tgingold, 2016-04-09:
On 08/04/16 00:49, flip111 wrote:
> |gnatmake -v| doesn't seem to do much
>
> |» gnatmake -v GNATMAKE GPL 2015 (20150428-49) Copyright (C) 1992-2015,
> Free Software Foundation, Inc. try "gnatmake --help" for more information. |

Ok.  What is the output of: which gnatmake

Can you also try to do: make; make install

Tristan.

flip111, 2016-04-10:
```
> which gnatmake
/usr/gnat/bin/gnatmake
```

if you do `make` and `make install` it's the same result as if i do `make -j3 CXX=g++` and then `make install`

flip111, 2017-11-14:
I tried it again ... this time without downloading from the adacore website. It was very easy using ubuntu 16.04 package manager

```
sudo apt install gnat-5
git clone https://github.com/tgingold/ghdl.git
cd ghdl
./configure --prefix=~/.local
make -j4
make install
```