ghdl/ghdl #55: Integration of Tool Libraries in Regular Installation Flow & Alternative User Target Directory

preusser, 2016-04-19:
Please, consider:
* to integrate the compilation of the third-party tool libraries into the regular installation flow by providing additional `configure` options like:
 *  `--tool-altera-quartusii=DIR` compile Quartus II simulation libs.
 *  `--tool-xilinx-ise=DIR`       compile ISE simulation libs.
 *  `--tool-xilinx-vivado=DIR`    compile Vivado simulation libs.
 *  `--tool-osvvm=DIR`            compile OSVVM libs.
 *  `--tool-vunit=DIR`            compile VUnit libs.
* to place the system-wide tool libraries together with the installation into `PREFIX/$libdirsuffix/tools`, and
* to apply the [https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html](XDG Base Directory Specification) and allow users to compile their private tool libraries into a path searched by GHDL, e.g., `$XDG_DATA_HOME/ghdl/lib/tools` with a default of `.local/share/ghdl/lib/tools`.

tgingold, 2016-05-02:
Providing additional configure options: not sure this is very useful as not many people build ghdl from sources. I prefer the vendor script way.

Having a system-wide and per user libraries is a good idea.

preusser, 2016-05-03:
This is a reasonable choice.

LarsAsplund, 2016-08-24:
@preusser Once you have VUnit installed you can run

``` console
cd <vunit installation directory>
python vunit\vunit\vhdl\compile_vunit_lib.py
```

to compile `vunit_lib` into the default location (`<current directory>\vunit_out`) or add `-o <other location>` to place it elsewhere.

If you use VUnit to build your entire project then your normal VUnit run script will take care of that