ghdl/ghdl #56: GHDL 0.34dev crashes when used with cocotb

jhotl, 2016-04-19:
I can't get cocotb to work with the latest clone of the git archive. I have cloned the cocotb git archive and the GHDL archive (0.34dev) on a ubuntu 15.10 installation (i tried both i686 and x86_64. On the x86_64 i use the llvm backend of GHDL). 

On the 32 bit version I get the following error report:

```
/usr/local/bin/ghdl -r --workdir=sim_build endian_swapper_vhdl --vpi=/home/jonathan.hofman/proj/cocotb/cocotb_ghdl/build/libs/i686/libvpi.so
loading VPI module '/home/jonathan.hofman/proj/cocotb/cocotb_ghdl/build/libs/i686/libvpi.so'
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:47   in gpi_print_registered_impl       VPI registered

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
In directory: /home/jonathan.hofman/proj/cocotb/cocotb_ghdl/examples/endian_swapper/tests/
Command line:
/usr/local/bin/ghdl -r --workdir=sim_build endian_swapper_vhdl --vpi=/home/jonathan.hofman/proj/cocotb/cocotb_ghdl/build/libs/i686/libvpi.so
Exception STORAGE_ERROR raised
Exception information:
Exception name: STORAGE_ERROR
Message: stack overflow or erroneous memory access
Call stack traceback locations:
0xf762f0d4 0xf778cb7e 0xf779ba72 0xf779bbf6 0xf7796de0 0xf779b213 0xf77a178e 0xf71ae2cc 0xf71ae627 0xf71abfe0 0x804f5d8 0x809b34f 0x806b089 0x808ebe8 0x8228e5a 0x822c3c4 0x822b727 0x80f9cad 0x804ef45 0xf71d873c 0x804deaf
******************************************************************
/home/jonathan.hofman/proj/cocotb/cocotb_ghdl/makefiles/simulators/Makefile.ghdl:38: recipe for target 'results.xml' failed

```

on the 64bit variant I get:

```
/usr/local/bin/ghdl -r --workdir=sim_build endian_swapper_vhdl --vpi=/home/jonathan.hofman/proj/cocotb/cocotb_ghdl/build/libs/x86_64/libvpi.so
loading VPI module '/home/jonathan.hofman/proj/cocotb/cocotb_ghdl/build/libs/x86_64/libvpi.so'
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:47   in gpi_print_registered_impl       VPI registered
/home/jonathan.hofman/proj/cocotb/cocotb_ghdl/makefiles/simulators/Makefile.ghdl:38: recipe for target 'results.xml' failed

```
Is there a combination with GHDL and cocotb that works? I tried both the mainstream cocotb and the fork in tgingold/cocotb? Better yet is it possible to get cocotb and ghdl to work?

Paebbels, 2016-04-21:
Hello,

I tried to use Cocotb with GHDL by myself and I found several bugs in Cocotb regarding the GHDL specific makefile ...

 1. No VHDL standard version is set, so VHDL defaults to VHDL-93 (`--std=93c`). I don't think the VHPI interface works with such an old standard, because VHPI was defined in 200x. Unfortunately, the VHPI specification is gone since www.eda.org is offline.
 2. The top-level variable `RTL_LIBRARY` is not supported (needed for `--work=<lib>`)
 3. Very useful options like:
      - `--mb-comments` -> allow multi-byte characters in comments (aka. UTF-8 chars in comments)
      - `-fexplicit` -> in case of overloaded operators, give preference to the explicit declaration
      - `-frelaxed-rules` -> relax some LRM rules, so it's possible to compile/embedd vendor library primitives, e.g. from unisim.

    are not set.
 4. `-P` parameters are not supported to load external pre-compiled libraries. e.g. unisim

The `workdir` parameter does not work as intended, so it's saver to rewrite it to an explicit call to `cd` before running GHDL. (See this [StackOverflow](http://stackoverflow.com/questions/36167474/ghdl-workdir-option-not-working-when-running-simulation) question.)

Here are my current lines from Makefile.ghdl

```makefile
# Compilation phase
analyse: $(VHDL_SOURCES) $(SIM_BUILD)
	cd $(SIM_BUILD) && $(GHDL) -a --std=08 --work=$(RTL_LIBRARY) --mb-comments -fexplicit -frelaxed-rules -P../../precompiled/ghdl/xilinx/ $(VHDL_SOURCES) && $(GHDL) -e --std=08 --work=$(RTL_LIBRARY) -P../../precompiled/ghdl/xilinx/ $(TOPLEVEL)

results.xml: analyse  $(COCOTB_LIBS) $(COCOTB_VPI_LIB)
	cd $(SIM_BUILD) && PYTHONPATH=$(LIB_DIR):$(SIM_ROOT):$(PWD):$(PYTHONPATH) LD_LIBRARY_PATH=$(LIB_DIR):$(LD_LIBRARY_PATH) MODULE=$(MODULE) \
        TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
	$(GHDL) -r --std=08 --work=$(RTL_LIBRARY) $(TOPLEVEL) --vpi=$(LIB_DIR)/libvpi.$(LIB_EXT)
```

This started GHDL and produced the following error:

```
loading VPI module '/home/paebbels/git/PoC/lib/cocotb/build/libs/x86_64/libvpi.so'
/home/paebbels/git/PoC/lib/cocotb/build/libs/x86_64/libgpilog.so: undefined symbol: _Py_TrueStruct
./sort_lru_list:error: cannot load VPI module
```

**So there seems to be a symbol discrepancy between the `libgpilog.so` and Python2.7....**

Notify the cocotb team: @chiggs, @stuarthodgson

jhotl, 2016-04-27:
I did some digging with the debugger. I see two issues:

1) The symbol `vpi_chk_error` is not found. The symbol `vpi_get_time` is found and is called from the "cocotb" code to "GHDL". I can trace this with the debugger. When commenting out the `vpi_chk_error` code to ignore this issue, the next symbol `vpi_handle_by_name` is not found. I didn't find yet why these symbols can't be found, while `vpi_get_time` works fine. I found one difference in de ghdl source code in `grt-vpi.ads`. The pragma was missing the "vpi_check_error" string, but adding this doesn't solve the issue. So I expect it doesn't matter/is not required. 

Original code:
```
   function vpi_chk_error (Info : p_vpi_error_info) return Integer;
   pragma Export (C, vpi_chk_error);

```
Changed code:
```
   function vpi_chk_error (Info : p_vpi_error_info) return Integer;
   pragma Export (C, vpi_chk_error, "vpi_chk_error");

```

2) GHDL segfaults on _dl_signal_error in the clib file dl-error.c. The segfault doesn't seem to be caused by GHDL itself, but something with libc (I'm using ubuntu wily (glibc-2.21)). I didn't apply the patches when looking at the source code in the debugger. So possibly the glibc source is slightly different than the original one as debial always aplies patches.

Anybody any clue on issue 1). Issue 2) seems like an libc issue (possibly with ubuntu) so I don't think we should spend to much time on that.

jhotl, 2016-04-27:
BTW: I have changed the makefiles with most of the changes suggested by @Paebbels. Originally the makefiles didn't even run GHDL, because of some errors in the GHDL specific makefile.

Paebbels, 2016-04-27:
I have no news so far, because we are working on a better QuestaSim integration for PoC. If my ideas work, we should be able to get rid of the makefiles. ...

I hope this idea works for GHDL too!

I found more issues in Cocotb on Debian Testing...

 1. The Cocotb-VHPI adapter should be compiled with simulator's GCC not the system GCC (in my case QuestaSim)
 2. QuestaSim comes with `libstdc++` but has no `libstdc`, so it defaults to the system libstdc (ugly but ok), BUT ...
 3. Using some system libs requires that the `ld`/`collect2` must to be the system linker. E.h. rename the tools linker to `*.old` so it defaults to the system linker.

Can you check if your environment uses the correct linker?

jhotl, 2016-04-28:
I'm using an LXC environment. Where I build both cocotb and ghdl using the system gcc. As I build both ghdl as cocotb from source I would not expect any linker issues or related. Also checking the environment, doesn't show in strange settings. So I would expect somewhat else

jhotl, 2016-04-28:
One of the things `vpi_chk_error` and `vpi_handle_by_name` have in common is that they use the Ghdl_C_String. Could that be the cause of the mismatch? I see some ADA examples that use a different string definition are used to exchange strings (or better `char *` between ADA and C).

Paebbels, 2016-04-28:
Thanks for digging into the details. I think Tristan will look into it. He is currently on vacation :)

tgingold, 2016-04-29:
On 21/04/16 02:43, Patrick Lehmann wrote:
> Hello,
>
> I tried to use Cocotb with GHDL by myself and I found several bugs in
> Cocotb regarding the GHDL specific makefile ...
>
>  1. No VHDL standard version is set, so VHDL defaults to VHDL-93 [...]

VHPI should work with any version.

Tristan.

tgingold, 2016-04-29:
On 27/04/16 23:06, Jonathan Hofman wrote:
> I did some digging with the debugger. I see two issues:
>
> 1) The symbol |vpi_chk_error| is not found. The symbol |vpi_get_time| is
> found and is called from the "cocotb" code to "GHDL". I can trace [...]

The symbols are missing from grt.ver  I will add them.

Thanks!

Tristan.

jhotl, 2016-05-01:
Adding those did indeed solve the issue with the undefined symbols (and resulting segfaults). Thanks for that. However the simulation does not run in combination with cocotb yet.

Running the endian swapper results in the output attached:
[output_cocotb_endian_swapper.txt](https://github.com/tgingold/ghdl/files/244163/output_cocotb_endian_swapper.txt)

The missing properties related to property type 4 (vpi_size). These are not implemented for GHDL yet. I couldn't figure out yet if this is easy to implement. There my knowledge of the GHDL code base is to small.

Dolu1990, 2016-06-06:
Any crispy news of working GHDL cocotb combo ?

tgingold, 2016-06-07:
On 06/06/16 20:07, Dolu1990 wrote:
> Any crispy news of working GHDL cocotb combo ?

Crispy ? No.
I have started to investigate. Cocotb has modified the interface with
the simulator and is getting new properties using the vpi API.
Some of these properties have to be implemented.

Tristan.

commit 7d3b9baa (2016-06-15): vpi: add missing properties for cocotb.

Fixes #56
closed 2016-06-15 by tgingold