ghdl/ghdl #57: Remove unreferenced entities

viric, 2016-04-20:
The makefile builds treating "warnings-as-errors". For me, this makes the ghdl build fail (for mcode). For example:

```
gcc -c -I./ -I./src -I./src/vhdl -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/grt -I./src/ortho -I./src/ortho/mcode -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -gnatw.A -I- /tmp/nix-build-ghdl-mcode-git-20160402.drv-0/ghdl-6f53685c28aefce0f31b85904a8b1cf6dcaf1d9e-src/src/grt/grt-fst.adb
grt-fst.adb:27:06: warning: no entities of "System" are referenced
gnatmake: "/tmp/nix-build-ghdl-mcode-git-20160402.drv-0/ghdl-6f53685c28aefce0f31b85904a8b1cf6dcaf1d9e-src/src/grt/grt-fst.adb" compilation error
Makefile:101: recipe for target 'ghdl_mcode' failed
```

Does it fail only for me? I patch the makefile so it doesn't treat warnings as errors.

Paebbels, 2016-04-20:
Is this mcode on Windows or Linux? Which GHDL version (current git HEAD?) are you using?

The Linux version is build and checked via Travis-CI. There was no abort due to a warning handled as error.

For example here is the log of [Build #8.2](https://travis-ci.org/Paebbels/ghdl/jobs/124075218) for mcode on Linux.

viric, 2016-04-20:
It's GNU/Linux, current HEAD. The same happened with 0.33. If you read the grt-fst.adb file, you will see that System is loaded but not referenced, so the warning-as-error is right. There are more "warnings" like these in other files.

viric, 2016-04-20:
For the record, I'm using this to disable "warnings-as-errors" and have a succesful build:
```
sed -i s/-gnatwae/-gnatwa/ Makefile.in ghdl.gpr.in
```

Paebbels, 2016-04-20:
Hello,

I have searched the build logs from Travis and yes there is an active `gnatwae` flag, but not such warning is flagged as an error. The Travis build process uses Ubuntu 12.04 + gcc 4.6 + gnat 4.6.

Can you give the versions numbers of you build environment for comparisons?

I'll try to trigger a mcode build on Travis with a newer gcc version in my GHDL fork at Paebbels/ghdl, to reproduce your issue.

tgingold, 2016-04-21:
On 20/04/16 11:42, Patrick Lehmann wrote:
> Hello,
>
> I have searched the build logs from Travis and yes there is an active
> |gnatwae| flag, but not such warning is flagged as an error. The Travis
> build process uses Ubuntu 12.04 + gcc 4.6 + gnat [...]

Depending on the exact version, the compiler may emit or not the warning.

I will remove the unused referenced package.

Thanks!

tgingold, 2016-04-29:
On 20/04/16 11:42, Patrick Lehmann wrote:
> Hello,
>
> I have searched the build logs from Travis and yes there is an active
> |gnatwae| flag, but not such warning is flagged as an error. The Travis
> build process uses Ubuntu 12.04 + gcc 4.6 + gnat [...]

It looks like warning accuracy depends on gcc version.
Anyway, the issue in grt-fst could be easily fixed.

Tristan.

tgingold, 2016-05-30:
Fixed.

closed 2016-05-30 by tgingold