ghdl/ghdl #58: Backtrace doesn't work in mcode-x86_64

viric, 2016-04-20:
In mcode-i686, backtrace works fine (current HEAD).

Examples:
```
# x86_64-linux
../../src/ieee/numeric_std-body.v93:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
ghdl:error: assertion failed
  from: ieee.numeric_std.to_integer at numeric_std-body.v93:2101

# same in i686-linux
../../src/ieee/numeric_std-body.v93:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
ghdl:error: assertion failed
  from: ieee.numeric_std.to_integer at numeric_std-body.v93:2098
  from: process work.xxx(behavioral).input_be_zero_i.first_half.B1.P0 at xxx.vhd:166
```

tgingold, 2016-05-03:
I think I do understand the issue: mcode doesn't create .eh_frame, so backtrace() fails.

gardners, 2020-06-03:
Is there any plan to rectify this?  It would make life much easier to be able to get backtraces on x86_64. I have tried both mcode and llvm with libbacktrace, and neither either prints me a backtrace, nor has a usable backtrace when run in gdb.

gardners, 2020-06-03:
Or as a work around, is there an easy way to build ghdl 32bit on a 64bit linux machine?

tgingold, 2020-06-03:
Backtraces can work with llvm and you should get results in gdb too unless something went very wrong.

Did you try with a very simple example ?

tgingold, 2020-06-03:
Just install a gcc compiler (with gnat) for 32bit.  But I am not sure this is the right way.
You'd better to investigate why backtraces don't work in 64bit.

gardners, 2020-06-03:
Let me try a very simple example. I'll start with llvm 64bit, as that is easiest to test initially.

gardners, 2020-06-03:
Speaking of which... what is the correct sequence for building an llvm-backend executable with debugging info for backtracing?  In particular, where should I put the -g ?

I'm doing the following, but it gives an error, even though it produces the executable.

```
ghdl -i src/vhdl/ghdl_bug.vhdl src/vhdl/cputypes.vhdl src/vhdl/debugtools.vhdl 
ghdl -m ghdl_bug
analyze src/vhdl/debugtools.vhdl
analyze src/vhdl/cputypes.vhdl
analyze src/vhdl/ghdl_bug.vhdl
elaborate ghdl_bug
ghdl -c -g src/vhdl/ghdl_bug.vhdl src/vhdl/cputypes.vhdl src/vhdl/debugtools.vhdl -e -g ghdl_bug 
ghdl:error: bad character in identifier
Makefile:309: recipe for target 'ghdl_bug' failed
make: *** [ghdl_bug] Error 1
```

tgingold, 2020-06-03:
Keep only the '-g' after the '-c'.

gardners, 2020-06-03:
Thanks. Will try.

gardners, 2020-06-03:
That fixes the error message, thank you.
Now I just get the refusal to print trace-backs:

```
paul@F96NG92-L:~/Projects/mega65/mega65-core$ ./ghdl_bug --backtrace-severity=error
...
src/vhdl/ghdl_bug.vhdl:177:7:@98823ns:(assertion error): The calculated value is $00
src/vhdl/ghdl_bug.vhdl:137:7:@98847ns:(report note): bare type=$001600, embedded value=$420000
...
```

This is with a ghdl built with llvm and the path to a libtraceback.a on the configure command line.  

Is there more magic I need to do to activate the tracebacks? 
Will it give an error if there is no traceback support, when I ask for it (so that we can be sure the binary really does support it)?

gardners, 2020-06-03:
Looks like it gets included:

```
ghdl -c -g -v src/vhdl/ghdl_bug.vhdl src/vhdl/cputypes.vhdl src/vhdl/debugtools.vhdl -e ghdl_bug/usr/local/bin/ghdl1-llvm -g -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ --anaelab ghdl_bug --ghdl-source=src/vhdl/ghdl_bug.vhdl --ghdl-source=src/vhdl/cputypes.vhdl --ghdl-source=src/vhdl/debugtools.vhdl -c -fpic -o e~ghdl_bug.o e~ghdl_bug
/usr/bin/gcc -o ghdl_bug e~ghdl_bug.o -g /usr/local/lib/ghdl/libgrt.a -ldl -lm /usr/local/lib/ghdl/libbacktrace.a -L./ -lz -ldl -Wl,--version-script=/usr/local/lib/ghdl/grt.ver -Wl,--export-dynamic
```

But I am still only seeing the process name, not a file name and line number:

```
paul@F96NG92-L:~/Projects/mega65/mega65-core$ ./cpu_test --assert-level=warning
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
./cpu_test:error: assertion failed
in process .cpu_test(behavior).P5
./cpu_test:error: simulation failed
paul@F96NG92-L:~/Projects/mega65/mega65-core$ ./ghdl_bug --assert-level=warning
src/vhdl/ghdl_bug.vhdl:57:5:@0ms:(report note): beginning simulation
src/vhdl/ghdl_bug.vhdl:137:7:@15ns:(report note): bare type=$000000, embedded value=$000000
src/vhdl/ghdl_bug.vhdl:141:7:@15ns:(report note): bare type=%000000000000000000000000, embedded value=%000000000000000000000000
src/vhdl/ghdl_bug.vhdl:145:7:@15ns:(report note): The bare bits are '0''0''0''0''0''0''0''0'
src/vhdl/ghdl_bug.vhdl:156:7:@15ns:(report note): The wrapped bits are 'U''U''U''U''U''U''U''U'
src/vhdl/ghdl_bug.vhdl:177:7:@15ns:(assertion error): The calculated value is $00
./ghdl_bug:error: assertion failed
in process .ghdl_bug(behavior).P1
./ghdl_bug:error: simulation failed
```

gardners, 2020-06-03:
Putting a breakpoint at do_report via gdb, it looks like there is no traceback on the stack to be enumerated.  Perhaps that is the problem?

Reading symbols from ./ghdl_bug...fertig.
(gdb) set args --assert-level=warning
(gdb) break do_report
Haltepunkt 1 at 0x3e0aa: file /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb, line 48.
(gdb) run
Starting program: /home/paul/Projects/mega65/mega65-core/ghdl_bug --assert-level=warning

```
Breakpoint 1, grt.lib.do_report (msg=..., str=0x555555835970 <_UI00000080>, default_str=..., severity=0, loc=0x555555835960 <_UI0000007e>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:48
48	   procedure Do_Report (Msg : String;
(gdb) where
#0  grt.lib.do_report (msg=..., str=0x555555835970 <_UI00000080>, default_str=..., severity=0, loc=0x555555835960 <_UI0000007e>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:48
#1  0x00005555555928e2 in <__ghdl_report> (str=0x555555835970 <_UI00000080>, severity=0, loc=0x555555835960 <_UI0000007e>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:151
#2  0x00005555555726eb in work__ghdl_bug__ARCH__behavior__P0( const &&&*) ()
#3  0x0000555555588538 in grt.processes.run_processes (postponed=false)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:807
#4  0x0000555555588739 in grt.processes.initialization_phase () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:869
#5  0x0000555555588d29 in <__ghdl_simulation_init> () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:1122
#6  0x0000555555589079 in grt.processes.simulation () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:1247
#7  0x0000555555593b60 in __ghdl_run_through_longjump (func=0x55555558906c <grt.processes.simulation>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/./src/grt/config/jumps.c:232
#8  0x0000555555593796 in grt.main.run_simul () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-main.adb:169
#9  0x000055555559386f in grt.main.run () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-main.adb:204
#10 0x00005555555aca36 in ghdl_main (argc=2, argv=0x7fffffffdbf8) at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/ghdl_main.adb:60
#11 0x0000555555593ddc in main (argc=2, argv=0x7fffffffdbf8) at /home/paul/Projects/mega65/mega65-core/ghdl/./src/grt/main.adb:31
(gdb) 
```

Is it perhaps LLVM being a bit too clever pants and flattening things that should not be flattened? Is the -g option being passed to the corret entails of LLVM, so that it leaves structure intact enough to leave a useful backtrace?  (Note I am using LLVM 8, but could easily use 6 or 7, if you think they would be better.)

tgingold, 2020-06-04:
__ghdl_report is for the report statement. So I suppose this is the report statement in your code, which is directly in a process.
Can you stop on an assertion from an ieee package ?  You should be able to have a backtrace from gdb (or if you cannot we are dead!).

Also, there is no debug info generated by the llvm backends.  That's a TODO work for ghdl.  But generating the line infos shouldn't be very difficult.

gardners, 2020-06-04:
Like this?

```
(gdb) break __ghdl_report
Haltepunkt 1 at 0x7f2552: file /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb, line 151.
(gdb) run
Starting program: /home/paul/Projects/mega65/mega65-core/cpu_test 
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0

Breakpoint 1, <__ghdl_report> (str=0x7fffffffcb90, severity=0, loc=0x55555650e788 <_UI00000e9d>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:151
151	      Do_Report ("report", Str, "Assertion violation", Severity, Loc);
(gdb) where
#0  <__ghdl_report> (str=0x7fffffffcb90, severity=0, loc=0x55555650e788 <_UI00000e9d>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:151
#1  0x00005555558494b9 in work__gs4510__ARCH__behavioural__P18( const &&&*) ()
#2  0x0000555555d3c208 in grt.processes.run_processes (postponed=false)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:807
#3  0x0000555555d3c409 in grt.processes.initialization_phase () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:869
#4  0x0000555555d3c9f9 in <__ghdl_simulation_init> () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:1122
#5  0x0000555555d3cd49 in grt.processes.simulation () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:1247
#6  0x0000555555d47830 in __ghdl_run_through_longjump (func=0x555555d3cd3c <grt.processes.simulation>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/./src/grt/config/jumps.c:232
#7  0x0000555555d47466 in grt.main.run_simul () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-main.adb:169
#8  0x0000555555d4753f in grt.main.run () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-main.adb:204
#9  0x0000555555d60116 in ghdl_main (argc=1, argv=0x7fffffffdc18) at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/ghdl_main.adb:60
#10 0x0000555555d47aac in main (argc=1, argv=0x7fffffffdc18) at /home/paul/Projects/mega65/mega65-core/ghdl/./src/grt/main.adb:31

```

It doesn't look like we can, unless I have misinterpretted the output. Basically I am seeing nothing above that indicates any file/line info in my source.

I've been tearing my hair out all week on this, because I have some subtle bug in GHDL simulation that I am not sure if it is  a GHDL bug, or a bug in my code, which in turn is stopping me actually debugging some critical VHDL.  So if you can think of any reliable way I can start getting GHDL backtraces on x86_64 Linux, I'm willing to try just about anything at this point.

tgingold, 2020-06-04:
> Like this?
> 
> (gdb) break __ghdl_report

You should break on grt__lib__do_report if you want to also stop on assertions.

> It doesn't look like we can, unless I have misinterpretted the output. Basically I am seeing nothing above that indicates any file/line info in my source.

Yes, that's because there is no debug output with LLVM.  I have to add that.

> I've been tearing my hair out all week on this, because I have some subtle bug in GHDL simulation that I am not sure if it is a GHDL bug, or a bug in my code, which in turn is stopping me actually [...]

Ok, then use the gcc backend, although it is a little bit harder to build.

gardners, 2020-06-04:
Ok. I'll give that a go.

gardners, 2020-06-04:
Ok, built that using gcc 10.1, confirmed it is linking with libbacktrace, but still not seeing any evidence of backtrace generation:

```
$ ghdl -v
GHDL 1.0-dev (v0.37.0-700-g648f428a) [Dunoon edition]
 Compiled with GNAT Version: 8.4.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ make ghdl_bug
ghdl -i src/vhdl/ghdl_bug.vhdl src/vhdl/cputypes.vhdl src/vhdl/debugtools.vhdl 
ghdl -m ghdl_bug
analyze src/vhdl/debugtools.vhdl
analyze src/vhdl/cputypes.vhdl
analyze src/vhdl/ghdl_bug.vhdl
elaborate ghdl_bug
ghdl -c -g -v src/vhdl/ghdl_bug.vhdl src/vhdl/cputypes.vhdl src/vhdl/debugtools.vhdl -e ghdl_bug 
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/10.1.0/ghdl1 -g -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ --anaelab ghdl_bug --ghdl-source=src/vhdl/ghdl_bug.vhdl --ghdl-source=src/vhdl/cputypes.vhdl --ghdl-source=src/vhdl/debugtools.vhdl -quiet -o e~ghdl_bug.s e~ghdl_bug
/usr/bin/as -o e~ghdl_bug.o e~ghdl_bug.s
/usr/local/bin/gcc -o ghdl_bug e~ghdl_bug.o -g /usr/local/lib/ghdl/libgrt.a -ldl -lm /usr/local/lib/ghdl/libbacktrace.a -L./ -lz -ldl -Wl,--version-script=/usr/local/lib/ghdl/grt.ver -Wl,--export-dynamic

$ gdb ./ghdl_bug 
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./ghdl_bug...fertig.
(gdb) break do_report
Haltepunkt 1 at 0x4510b2: file /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb, line 48.
(gdb) run
Starting program: /home/paul/Projects/mega65/mega65-core/ghdl_bug 

Breakpoint 1, grt.lib.do_report (msg=..., str=0x7fffffffd690, default_str=..., severity=0, loc=0x4d9f10 <_UI00000080.218>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:48
48	   procedure Do_Report (Msg : String;
(gdb) where
#0  grt.lib.do_report (msg=..., str=0x7fffffffd690, default_str=..., severity=0, loc=0x4d9f10 <_UI00000080.218>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:48
#1  0x00000000004518ea in <__ghdl_report> (str=0x7fffffffd690, severity=0, loc=0x4d9f10 <_UI00000080.218>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-lib.adb:151
#2  0x000000000042967d in work__ghdl_bug__ARCH__behavior__P1__PROC (INSTANCE=0x7ffff7fad010) at e~ghdl_bug:205
#3  0x0000000000447540 in grt.processes.run_processes (postponed=false)
    at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:807
#4  0x0000000000447741 in grt.processes.initialization_phase () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:869
#5  0x0000000000447d31 in <__ghdl_simulation_init> () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:1122
#6  0x0000000000448081 in grt.processes.simulation () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-processes.adb:1247
#7  0x0000000000452b68 in __ghdl_run_through_longjump (func=0x448074 <grt.processes.simulation>)
    at /home/paul/Projects/mega65/mega65-core/ghdl/./src/grt/config/jumps.c:232
#8  0x000000000045279e in grt.main.run_simul () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-main.adb:169
#9  0x0000000000452877 in grt.main.run () at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/grt-main.adb:204
#10 0x000000000046ba3e in ghdl_main (argc=1, argv=0x7fffffffdc28) at /home/paul/Projects/mega65/mega65-core/ghdl/src/grt/ghdl_main.adb:60
#11 0x0000000000452de4 in main (argc=1, argv=0x7fffffffdc28) at /home/paul/Projects/mega65/mega65-core/ghdl/./src/grt/main.adb:31
```

Any thoughts?

tgingold, 2020-06-05:
But you haven't analyzed with the option `-g`.  That's needed to add debug information.  Ideally you should do:

```
ghdl -i src/vhdl/ghdl_bug.vhdl src/vhdl/cputypes.vhdl src/vhdl/debugtools.vhdl 
ghdl -m -g ghdl_bug
```

The `-m` command will produce the executable.

You will get a backtrace by default on assertion failure.

gardners, 2020-06-05:
ok, I'll try that

gardners, 2020-06-05:
So, with that, if I then do something like:

`$ ./cpu_test --assert-severity=warning 
`

I now get this:

```
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@15ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
./cpu_test:error: assertion failed
in process .cpu_test(behavior).core0@machine(behavioral).iomapper0@iomapper(behavioral).audio0@audio_complex(elizabethan).pwm0@pcm_to_pdm(nordic).P0
```

Still no line-numbers, but it does at least give me some info on where it is being triggered from.

But something like this still fails to generate any backtrace output for any report or assertion:

`./cpu_test --backtrace-severity=note` 

Only tangentially related, I have been seeing a funny possible bug with record use (its why I have been trying to get backtraces to confirm it). Are you aware of any dragons in the use of records and arrays of records? I vaguely remember there being some issue with them a while back, but I might be quite confused in my memory.

Also, thanks as always for your prompt and helpful responses :)

Paul.

tgingold, 2020-06-05:
> ../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@15ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
> ./cpu_test:error: assertion failed
> in process [...]

So this is a backtrace.

> Still no line-numbers, but it does at least give me some info on where it is being triggered from.

Did you clean-up before rebuilding with `-g` ?  `ghdl -m` only recompile the files that have changed, so if you just re-run with the `-g` flag, the existing files haven't been recompiled

> Only tangentially related, I have been seeing a funny possible bug with record use (its why I have been trying to get backtraces to confirm it). Are you aware of any dragons in the use of records [...]

This is fully supported for vhdl-93, there might be crashes for vhdl-08 and its unbounded records/arrays.

gardners, 2020-06-05:
Nope, till no improvement on displaying backtraces after removing all the .o files.

As for the record problem, I'll keep poking about, and see if I can get you an easily buildable example.  Might take me a while, as just pulling out the simple record manipulation on its own into a file doesn't trigger it.