ghdl/ghdl #60: State of multi core / parallel simulation

StefanD986, 2016-04-21:
This is a question/enhancement: I wanted to know what the state of ghdl is with respect to simulations on multi threading/core machines.
I found a [discussion](https://mail.gna.org/public/ghdl-discuss/2008-07/msg00005.html) that about the topic that is about 8 years old, and just wanted to know whether something has changed.

Thank you for all your work Tristan!

tgingold, 2016-04-29:
Hello,

no nothing has changed.  I made some tests a long time ago, and the results weren't impressive.  The conclusion was: before using multicores, first optimize for a single core.

Thanks,
Tristan.

closed 2016-04-29 by tgingold

mbitsnbites, 2019-05-07:
Not sure if this ticket should be re-opened, but seeing that 8-16 core CPU:s are becoming mainstream this year, and 32-64 core CPU:s will likely be common in high end workstations a couple of years from now, there is a huge potential for performance improvements if that CPU power could be utilized (or put another way: it would be a huge waste *not* to use all that power).

Due to the nature of the functionality that GHDL simulates, it *should* be possible to extract a healthy amount of parallelism for most VHDL designs.

umarcor, 2019-05-07:
Hi @mbitsnbites! Although it is not a built-in feature, it is currently possible to co-execute multiple binaries generated with GHDL. Assuming that the design can be split into some submodules which are connected to each other through some standard interface, VHPIDIRECT can be used to transfer data between them.

In the following VUnit script, you can find and example of how to load and execute a GHDL binary in a Python thread: https://github.com/dbhi/vunit/blob/feat-external-arrays/examples/vhdl/external_buffer/run.py#L104-L118. Note that it is possible to achieve the same result with C and pthreads.

mbitsnbites, 2019-05-08:
Thanks for the feedback. I'm already using the LLVM backend, so I'll be sure to check out the reference you pointed to about optimization passes.

Since I don't have any knowledge about the GHDL internals, I am not going to suggest any technical solutions. Anything based on frequent locking is likely going to kill performance, as @Paebbels suggests, so you may need some other construct. If the architecture isn't built with massive parallelism as a key design goal from the start, it can be hard to retrofit an efficient model later.

umarcor, 2019-05-08:
> If the architecture isn't built with massive parallelism as a key design goal from the start, it can be hard to retrofit an efficient model later.

Related to this, multi-thread support was recently added to Verilator. The following video might be inspiring for anyone willing to think about how could this be done: https://www.youtube.com/watch?v=en8JMz7v3LU (min 7:30 and forward).

tgingold, 2019-05-10:
This is a matter of efforts and priority.  I am convinced that before going parallel, a program must already be optimized for sequential execution.

go2sh, 2019-05-10:
It depends. If your problem is a just the calculations, then optimizing for a single core. But especially for big designs with a lot of memory, multi threading might help as loading stuff from memory gets paralelized.

mbitsnbites, 2019-05-10:
I respect that line of reasoning (again, especially since I'm not familiar with the GHDL internals).

However it is also easy to argue that since multi-threaded execution can potentially give a many-fold performance increase (as demonstrated by Verilator), you'd have to have a pretty aggressive plan for matching that performance win with single-threaded execution improvements. In other words: which path will give you the biggest win per invested development hour?

I may be biased, but for most things that I do (tune build systems for CI, implement raytracers, and optimizing SW in general) I tend to spen a lot of time in a CPU load monitor (e.g. htop), and whenever I see unused cores I get an itch to fix it.