ghdl/ghdl #61: Can't configure with llvm version 3.5.1

albertschulz, 2016-04-25:
I installed llvm version 3.5 on my mac machine using `brew install llvm35`.
The configure script won't let me configure:
./configure --with-llvm-config=/usr/local/bin/llvm-config-3.5

I receive the output:
```
➜ ghdl git:(master) ✗ ./configure --with-llvm-config=/usr/local/bin/llvm-config-3.5
Build machine is: x86_64-apple-darwin13.4.0
grep: brackets ([ ]) not balanced
Mismatch llvm version 3.5.1 from /usr/local/bin/llvm-config-3.5
Need llvm version llvm version 3.5 [do not modify this line as this is read by scripts]
```

I seem like the check_version function is a little bit to strict

albertschulz, 2016-04-25:
Bypassing the check function works fine

tgingold, 2016-04-29:
On 25/04/16 12:40, Albert Schulz wrote:
> I installed llvm version 3.5 on my mac machine using |brew install llvm35|.
> The configure script won't let me configure:
> ./configure --with-llvm-config=/usr/local/bin/llvm-config-3.5
>
> I receive the [...]

You got an error from grep.  Looks like a shell script issue.
Can you run (and paste the output):

sh -x ./configure --with-llvm-config=/usr/local/bin/llvm-config-3.5

Tristan.

tmeissner, 2016-04-29:
I think it's a issue with the different egrep version installed on OSX. I've runned in a similar problem some time ago when building GHDL the first time.

albertschulz, 2016-06-20:
I just pulled the latest version and it's working now.

FYI: this is the output of the command:

`+ backend=mcode
+ CC=gcc
+ CFLAGS=-g
+ GNATMAKE=gnatmake
+ LDFLAGS=
+ prefix=/usr/local
+ libdirsuffix=lib/ghdl
+ libdirreverse=../..
+ gcc_src_dir=
+ gcc_version=unknown
+ llvm_config=
+ backtrace_lib=
+ build=
+ build_mode=
+ show_help=no
+ progname=./configure
+ subst_vars='CC GNATMAKE CFLAGS LDFLAGS build srcdir prefix backend libdirsuffix libdirreverse gcc_src_dir llvm_config backtrace_lib build_mode'
++ dirname ./configure
+ srcdir=.
+ test x. = x
++ echo -n
+ test x-n = x-n
+ echon=
+ echoc='\c'
++ grep '^gcc version' ./BUILD.txt
++ sed -e 's/gcc version \([0-9.]*\) \[.*\]\.$/\1/'
+ gcc_version='gcc version 4.9 [do not modify this line as this is read by scripts].'
++ grep '^llvm version' ./BUILD.txt
++ cut -d ' ' -f 3
+ llvm_version=3.5
+ for opt in '"$@"'
++ expr x--with-llvm-config=/usr/local/bin/llvm-config-3.5 : 'x[^=]*=\(.*\)'
+ optarg=/usr/local/bin/llvm-config-3.5
+ case "$opt" in
+ llvm_config=/usr/local/bin/llvm-config-3.5
+ backend=llvm
+ test no '!=' no
+ gnatmake --version
+ gcc -v
+ test x = x
++ gcc -g -dumpmachine
+ build=x86_64-apple-darwin13.4.0
+ echo 'Build machine is: x86_64-apple-darwin13.4.0'
Build machine is: x86_64-apple-darwin13.4.0
+ test llvm = mcode
+ test llvm = gcc
+ test llvm = llvm
++ /usr/local/bin/llvm-config-3.5 --version
+ llvm_ver=3.5.1
+ '[' 0 '!=' 0 ']'
+ check_version 3.5 3.5.1
++ echo 3.5
++ sed 's/\./v/g'
+ exp_ver=3v5
++ echo 3.5.1
++ sed 's/\./v/g'
+ tool_ver=3v5v1
+ echo 3v5v1
+ grep -q '^3v5'
+ return 0
+ case "$build" in
+ LDFLAGS=' -Wl,-no_compact_unwind'
+ rm -f config.status
+ echo '#! /bin/sh'
+ echo '# Generated by configure.'
+ echo '# Run this file to recreate the current configuration.'
+ echo
+ echo '# Generated by:'
+ echo '# ./configure\c'
+ for opt in '"$@"'
+ echo ' "--with-llvm-config=/usr/local/bin/llvm-config-3.5"\c'
+ echo
+ echo
+ echo 'subst_vars="CC' GNATMAKE CFLAGS LDFLAGS build srcdir prefix backend libdirsuffix libdirreverse gcc_src_dir llvm_config backtrace_lib 'build_mode"'
+ for v in '$subst_vars'
+ eval 'vval=$CC'
++ vval=gcc
+ echo 'CC="gcc"'
+ for v in '$subst_vars'
+ eval 'vval=$GNATMAKE'
++ vval=gnatmake
+ echo 'GNATMAKE="gnatmake"'
+ for v in '$subst_vars'
+ eval 'vval=$CFLAGS'
++ vval=-g
+ echo 'CFLAGS="-g"'
+ for v in '$subst_vars'
+ eval 'vval=$LDFLAGS'
++ vval=' -Wl,-no_compact_unwind'
+ echo 'LDFLAGS="' '-Wl,-no_compact_unwind"'
+ for v in '$subst_vars'
+ eval 'vval=$build'
++ vval=x86_64-apple-darwin13.4.0
+ echo 'build="x86_64-apple-darwin13.4.0"'
+ for v in '$subst_vars'
+ eval 'vval=$srcdir'
++ vval=.
+ echo 'srcdir="."'
+ for v in '$subst_vars'
+ eval 'vval=$prefix'
++ vval=/usr/local
+ echo 'prefix="/usr/local"'
+ for v in '$subst_vars'
+ eval 'vval=$backend'
++ vval=llvm
+ echo 'backend="llvm"'
+ for v in '$subst_vars'
+ eval 'vval=$libdirsuffix'
++ vval=lib/ghdl
+ echo 'libdirsuffix="lib/ghdl"'
+ for v in '$subst_vars'
+ eval 'vval=$libdirreverse'
++ vval=../..
+ echo 'libdirreverse="../.."'
+ for v in '$subst_vars'
+ eval 'vval=$gcc_src_dir'
++ vval=
+ echo 'gcc_src_dir=""'
+ for v in '$subst_vars'
+ eval 'vval=$llvm_config'
++ vval=/usr/local/bin/llvm-config-3.5
+ echo 'llvm_config="/usr/local/bin/llvm-config-3.5"'
+ for v in '$subst_vars'
+ eval 'vval=$backtrace_lib'
++ vval=
+ echo 'backtrace_lib=""'
+ for v in '$subst_vars'
+ eval 'vval=$build_mode'
++ vval=
+ echo 'build_mode=""'
++ echo CC GNATMAKE CFLAGS LDFLAGS build srcdir prefix backend libdirsuffix libdirreverse gcc_src_dir llvm_config backtrace_lib build_mode
++ sed -e 's/\([a-zA-Z_]*\)/ -e "s%@\1@%$\1%g"/g'
+ sed_opts=' -e "s%@CC@%$CC%g"  -e "s%@GNATMAKE@%$GNATMAKE%g"  -e "s%@CFLAGS@%$CFLAGS%g"  -e "s%@LDFLAGS@%$LDFLAGS%g"  -e "s%@build@%$build%g"  -e "s%@srcdir@%$srcdir%g"  -e "s%@prefix@%$prefix%g"  -e "s%@backend@%$backend%g"  -e "s%@libdirsuffix@%$libdirsuffix%g"  -e "s%@libdirreverse@%$libdirreverse%g"  -e "s%@gcc_src_dir@%$gcc_src_dir%g"  -e "s%@llvm_config@%$llvm_config%g"  -e "s%@backtrace_lib@%$backtrace_lib%g"  -e "s%@build_mode@%$build_mode%g"'
+ echo 'echo "Creating ghdl.gpr"'
+ echo sed -e '"s%@CC@%$CC%g"' -e '"s%@GNATMAKE@%$GNATMAKE%g"' -e '"s%@CFLAGS@%$CFLAGS%g"' -e '"s%@LDFLAGS@%$LDFLAGS%g"' -e '"s%@build@%$build%g"' -e '"s%@srcdir@%$srcdir%g"' -e '"s%@prefix@%$prefix%g"' -e '"s%@backend@%$backend%g"' -e '"s%@libdirsuffix@%$libdirsuffix%g"' -e '"s%@libdirreverse@%$libdirreverse%g"' -e '"s%@gcc_src_dir@%$gcc_src_dir%g"' -e '"s%@llvm_config@%$llvm_config%g"' -e '"s%@backtrace_lib@%$backtrace_lib%g"' -e '"s%@build_mode@%$build_mode%g"' '< $srcdir/ghdl.gpr.in > ghdl.gpr'
+ echo 'echo "Creating Makefile"'
+ echo sed -e '"s%@CC@%$CC%g"' -e '"s%@GNATMAKE@%$GNATMAKE%g"' -e '"s%@CFLAGS@%$CFLAGS%g"' -e '"s%@LDFLAGS@%$LDFLAGS%g"' -e '"s%@build@%$build%g"' -e '"s%@srcdir@%$srcdir%g"' -e '"s%@prefix@%$prefix%g"' -e '"s%@backend@%$backend%g"' -e '"s%@libdirsuffix@%$libdirsuffix%g"' -e '"s%@libdirreverse@%$libdirreverse%g"' -e '"s%@gcc_src_dir@%$gcc_src_dir%g"' -e '"s%@llvm_config@%$llvm_config%g"' -e '"s%@backtrace_lib@%$backtrace_lib%g"' -e '"s%@build_mode@%$build_mode%g"' '< $srcdir/Makefile.in > Makefile'
+ chmod +x ./config.status
+ sh ./config.status
Creating ghdl.gpr
Creating Makefile
+ test llvm = mcode
+ echo 'Generate default_pathes.ads'
Generate default_pathes.ads
++ pwd
+ curdir=/Users/albertschulz/git/ghdl
+ sed -e s%@COMPILER_GCC@%ghdl1-gcc% -e s%@COMPILER_DEBUG@%ghdl1-debug% -e s%@COMPILER_MCODE@%ghdl1-mcode% -e s%@COMPILER_LLVM@%bin/ghdl1-llvm% -e s%@POST_PROCESSOR@%oread-llvm% -e s%@INSTALL_PREFIX@%/usr/local% -e s%@LIB_PREFIX@%lib/ghdl%
+ exit 0`

closed 2016-06-20 by albertschulz