ghdl/ghdl #63: Set output file name for -a

cclassic, 2016-05-01:
Hi,

I am currently working on adding the Lattice Semi libraries to your library compile script. The problem is that some source files have the same name even though they are from different devices. So if I compile all devices into one directory (as with the Altera lib), things might break even though everything seems to work in this case.

I tried using `-o filename.o`, but that did not have any effect. Is there another way to set the output file name. If not, that might be a useful addition. 

Markus

tgingold, 2016-05-02:
No, -o xx won't work that easily.  Object file names are computed from source file name and that's also used during the final link.

Why aren't they simply in different libraries ?

cclassic, 2016-05-02:
They are in different libraries, but I compile both libs into one folder.

Scenario files:
* chip-a/LUT.vhd [lib chipa]
* chip-b/LUT.vhd [lib chipb]

Now when I compile these into one directory, I will get these files:
* chipa-obj93.cf
* chipb-obj93.cf
* LUT.o

Assuming the content of the two LUT.vhd is different, does LUT.o contain both implementation or will one be lost?

Paebbels, 2016-05-02:
You can compile each library into a seperate directory. They can later be referenced with the `-Ppath` option.

cclassic, 2016-05-03:
@Paebbels Thanks for the note, but I am aware of that possibility. I intentionally used a single directory because that is how the existing Altera libs are compiled.

Paebbels, 2016-05-03:
I'm compiling all Altera libs into one directory, because the components reside in different VHDL libraries. In case of Xilinx, I implemented two scripts for ISE and Vivado, because of the naming clashes.

Do you have a Lattice Diamond installation on Windows, too? I can't install Diamond in my Debian VM, because because of the *.rpm packages and the missing ability to install a *.deb package (transformed by alien) into /opt/, wherein such a big package should be installed...

tmeissner, 2016-05-03:
>  I can't install Diamond in my Debian VM, because because of the *.rpm packages and the missing ability to install a *.deb package (transformed by alien) into /opt/, wherein such a big package should be installed...

Why not installing it first in /usr/local, as the converted rpm does, and then moving it into /opt? This is what we've done at work on our debian based build-server.

Paebbels, 2016-05-03:
Unfortunately, the Debian main installation in my VM is mapped to a virtual disk on my SSD (18 GB) and requires already 15 GB for several build tools, except the EDA tools. It can't be resized as easy as I wish. The /opt/ folder is mapped to a 100 GB virtual disk on a conventional HDD. The /opt/ folder has space for another 50 GB of EDA tools.

I'm not sure how my Debian + XFCE desktop grow to that size... Is there a way to find the biggest and maybe unused installations in Debian?

tmeissner, 2016-05-03:
Maybe I'm wrong, but you can also simply copy the files from the rpm/deb into a folder of your choice. You can't manage the installation with the apt-tool when doing this, but I think, Diamond is self-contained, so no problem with this.

Paebbels, 2016-05-03:
@tmeissner Do you have a Gitter account? If so, we could continue the discussion there. If not, you can login with a GitHub or Twitter account.

[=> Gitter Chat](https://gitter.im/VLSI-EDA/PoC)

tgingold, 2016-05-15:
I am closing this issue as I think the right solution is to use different directories.
If that doesn't work for you, please reopen this issue.

Tristan.

closed 2016-05-15 by tgingold