ghdl/ghdl #64: Building ghdl with gnat-5 on Travis-CI

Paebbels, 2016-05-09:
**Enhancement**

Is it possible to build ghdl on Travis-CI with gnat-5? This could ease some Travis settings for my own CI runs an Travis-CI, which download your "weekly" built ghdl.

Currently I have to use a `sudo` enabled VM (Trusty) to install `gnat-4.8` with root rights, because `gnat-4.8` is not in the list of [trusted packages](https://github.com/travis-ci/apt-package-whitelist/blob/master/ubuntu-precise). `gnat-5` is listed.

I'm compiling ghdl in my own private Debian VM with `gcc-5`/`gnat-5` (5.3.x) + `llvm-3.5` since month and haven't found any issues. Are there any advantages to use `gnat-4.8` over `gnat-5`?

Trusted packages for Ubuntu:
- [gcc-5](https://github.com/travis-ci/apt-package-whitelist/blob/master/ubuntu-precise#L1155)
- [gnat-5](https://github.com/travis-ci/apt-package-whitelist/blob/master/ubuntu-precise#L1525)
- [llvm-3.5](https://github.com/travis-ci/apt-package-whitelist/blob/master/ubuntu-precise#L6334)

Regards
    Patrick

hakusdream, 2016-05-30:
@Paebbels I don't see any.

@tgingold Check this.

Brian-Drummond, 2016-07-21:
With the LLVM backend, it should be much less dependent on a specific gcc version. If the result passes the testsuite it should be good to go.

Getting the gcc backend working with gcc-5 (or 6 or 7) may turn out to be a different story - at the moment I'm using gcc-4.9 because I want code coverage via gcov. Is gcc-4.9 among the trusted options?

Paebbels, 2016-07-21:
The newest list (see the link in my first post) lists:
* gcc-4.4 - 4.9
* gcc-5
* gcc-6

Brian-Drummond, 2016-07-21:
ah, gcc-4.8 is trusted but gnat-4.8 (which is part of gcc-4.8 though supplied by a separate package) is not? The same probably applies to gnat-4.9

tgingold, 2016-12-11:
gnat-5 and llvm-3.8 are trusted, but not available:

Build system information
Build language: c
Build group: stable
Build dist: precise
Build id: 183027157
Job id: 183027160
...
Installing APT Packages (BETA)
$ export DEBIAN_FRONTEND=noninteractive
4.86s$ sudo -E apt-get -yq update &>> ~/apt-get-update.log
2.63s$ sudo -E apt-get -yq --no-install-suggests --no-install-recommends --force-yes install gnat-6 llvm-3.8-dev llvm-3.8-tools libedit-dev zlib1g-dev
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package gnat-6
E: Unable to locate package llvm-3.8-dev
E: Couldn't find any package by regex 'llvm-3.8-dev'
E: Unable to locate package llvm-3.8-tools
E: Couldn't find any package by regex 'llvm-3.8-tools'

I may have missed something...

tgingold, 2016-12-11:
closing as gnat-5 is not available.

closed 2016-12-11 by tgingold