ghdl/ghdl #65: From StackOverflow: GHDL --workdir option not working when running simulation

Paebbels, 2016-05-09:
I'm not sure if you know this issue reported on [StackOverflow](http://stackoverflow.com/questions/36167474/ghdl-workdir-option-not-working-when-running-simulation), so I'll add it here for completeness. I can't find such an issue in the closed issue list.

*I think the issue is still present in GHDL, because I encountered a similar phenomena, but I didn't save the causing files/scripts.*

Original post from user [**damage**](http://stackoverflow.com/users/1609276/damage), 23.03.2016

------------
I am currently spending some time to get familiar with GHDL and have run into an issue I can't get around.
The folder structure I created for simulation is the following:

    ghdl/
      +-out/
      +-run/
      +-tmp/

In the `run` folder is a script which calls GHDL to analyze, elaborate and run my design.
For now I'd like to put all files created by GHDL into the folder `/ghdl/out`. To do this I specify the option `--workdir=../out` whenever calling GHDL.  
This is the script I have in a file inside `/ghdl/run`:

    ghdl -i --workdir=../out ../../../rtl/test_module.vhd ../../../test/bench/test_module_tb.vhd
    ghdl -m --workdir=../out -o ../out/test_module_tb test_module_tb
    ghdl -r --workdir=../out test_module_tb

(The elaboration phase works fine, therefor I do not include any details about the RTL and testbench files).  
When running the simulation script I get the following error message:

    ghdl -r --workdir=../out test_module_tb
    ghdl: file 'test_module_tb' does not exists
    ghdl: Please elaborate your design.

I have checked, the executable `test_module_tb` is actually present in the folder `/ghdl/out`.  
Am I using the `--workdir` option incorrectly? What is the right way to use GHDL in case the simulation executable is in a different directory?

PS: I am using GHDL 0.34dev (20151126) from the Fedora 23 RPM repo.

------------------
Source: [StackOverflow question](http://stackoverflow.com/questions/36167474/ghdl-workdir-option-not-working-when-running-simulation)

kblomqvist, 2016-05-11:
I have understood that `--workdir` is about where the libraries/packages are found -- not where to output.

For example,

```bash
ghdl -a --workdir=work --work=work package.vhd
ghdl -a --workdir=work top.vhd
ghdl -e --workdir=work top
```

Paebbels, 2016-05-12:
Pre-compiled packages and entities can be referenced by `-Ppath/to/lib`.

Example:

```Bash
cd lib1
ghdl -a --work=lib1 ..\src\pack.vhdl
ghdl -a --work=lib1 ..\src\file1.vhdl
ghdl -a --work=lib1 ..\src\file2.vhdl
cd ..
ghdl -a --work=test -Plib1 ..\src\top.vhdl
ghdl -a --work=test -Plib1 ..\src\tb.vhdl
...
```

tgingold, 2016-05-15:
@kblomqvist is correct: --workdir is to specify the directory of the working directory, not the directory of an executable.  The mistake in the original message is the use of '-o' in that line:
ghdl -m --workdir=../out -o ../out/test_module_tb test_module_tb

Tristan.

closed 2016-05-15 by tgingold