ghdl/ghdl #66: Component binding rules

Paebbels, 2016-05-10:
I’m not sure if it’s a real GHDL issue or a VHDL language issue...

The below listed code is a short and hopefully typo free minimal example for one of our PoC arithmetic entities. These entities have mostly 2 implementations: a generic and a vendor specific one..

Here is a short explanations for the example code:  
The entity `dummy` has two implementations of algorithm `DataOut <= DataIn;`. The first is a generic implementations, the second is an optimized version for vendor "abc". Depending on the
generic value 'vendor' the implementations are selected. The vendor specific implementation is instantiated using the "old" component syntax, which allows the instance to be unbound if the generate is not 'active'. The component declaration is stored in a package `dum` in library `lib`.

The problem is as follows:
 1. If this example is compiled using GHDL and vendor = "gen", then everything works, except for a unbound instance warning in an inactive generate.
 2. If this example is afterwards compiled using vendor = "abc", then everything works again, because the generate is active and the component is bound. (GHDL was called with `-Pabc` to the vendor's library and path `abc` was found.)
 3. If this example is compiled using GHDL again, but without `-P`, because it's now generic again, and vendor set to "gen", then component `dummy_abc` gets bound, because it's still on disk and not outdated. Moreover, `dummy_abc` requests for the vendor's library `abc`, which can not be found.

There is a quick workaround: Delete all temporary files before every simulation, but I think it's not correct to bind a component, when the generate is not active.

What does QuestaSim do?

On the first run, it reports similar warnings as GHDL, but then it realizes, the component is not used/reached. If QuestaSim is run a second time, no warning is issued... Some kind of memory effect?

```VHDL
package dum is
  component dummy_abc is
		port (
			clock   : std_logic;
			datain  : std_logic;
			dataout : std_logic
		);
	end component;
end package;

library lib;
use     lib.dum.all;

entity dummy is
  generic (
    vendor : string := "abc"
  );
	port (
    clock   : std_logic;
    datain  : std_logic;
    dataout : std_logic
  );
end entity;

architecture rtl of dummy is
begin
  genGeneric: if (vendor /= "abc") generate
		dataout <= datain;
	end generate;
	genXilinx: if (vendor = "abc") generate
	  abc : dummy_abc
		  port map (
        clock    => clock,
        datain   => datain,
        dataout  => dataout
			);
	end generate;
end architecture;

library abc;
use     abc.comp.all;		-- the vendor's component package for FPGA primitives

entity dummy_abc is
	port (
    clock   : std_logic;
    datain  : std_logic;
    dataout : std_logic
  );
end entity;

architecture rtl of dummy_abc is
begin
  prim : abc_dummy
		port map (
      clock    => clock,
      datain   => datain,
      dataout  => dataout
		);
end architecture;
```

tgingold, 2016-05-17:
I am rereading the LRM with this point in my mind.

tgingold, 2016-05-18:
It is not a good idea to change switches (like -P) between two analyses.
For portability, you should always use components and bind to a different entity/architecture.

I don't really know what could be done: dummy_abc was correctly analysed and after that it is known by ghdl.  If you remove the path to abc, ghdl would report issues.
I can document the fact that -P should be kept.

Paebbels, 2016-05-18:
This `-P` switch is the same for all analysis steps.

If you have a bit of time, we could discuss this at [Gitter](https://gitter.im/VLSI-EDA/PoC) ...

tgingold, 2016-05-18:
That's not what you have said in 3.

tgingold, 2016-05-18:
On 18/05/16 20:39, Patrick Lehmann wrote:
> This |-P| switch is the same for all analysis steps.
>
> If you have a bit of time, we could discuss this at Gitter
> <https://gitter.im/VLSI-EDA/PoC> ...

I prefer to stay on GH to track all the messages.

Paebbels, 2016-05-18:
Gitter stores all conversations, which can be referenced from ouside. It's like having a git for chat messages ...

OK, I should be more precise ...

The `-P` switch is the same for one testbench run, while analyzing, elaborating and running.
`./poc.sh ghdl PoC.arith.prefix_and --board=KC705`

If the user decides to run another testbench or the same testbench with another board, the error occurs. (Here a Lattice dev. board, wich uses a generic code implementation)
`./poc.sh ghdl PoC.arith.prefix_and --board=ECP5Versa`

I found this in the LRM (2008):
> **14.5.3 Generate statements**
> ...
> For an if generate statement, elaboration consists of the evaluation, in succession, of the condition specified after if and any conditions specified after elsif (treating a final else as elsif TRUE generate) until one evaluates to TRUE or all conditions are evaluated and yield FALSE. If one condition evaluates to TRUE, then exactly one block statement is generated; otherwise, no block statement is generated.

So according to this, if `genXilinx: if (vendor = "xilinx") generate` evaluates to false, then no `BLOCK` statement is created as an intermediate result, thus there is no instance, which could be bound. But GHDL is binding an instance.

tgingold, 2016-05-18:
On 18/05/16 21:13, Patrick Lehmann wrote:
> Gitter stores all conversations, which can be referenced from ouside.
> It's like having a git for chat messages ...
>
> OK, I should be more precise ...
>
> The |-P| switch is the same for one testbench [...]

But according to the LRM, the binding is done before elaboration.  See 
14.2 Elaboration of a design hierarchy.

Paebbels, 2016-05-18:
Of cause I'm not as fit as you in reading the LRM ... so I have 3 questions or remarks:

1. As far as I can see, 14.x never states that 14.2 occurs before 14.5.x. I just see paragraph linkings in the PDF file, but not ordering in time.

2. How can the elaboration of the design hierarchy start if not all hierarchy "rewritings" (transforming generate statements into block statements) are done?

3. > **LRM 14.2**
    > Elaboration of a design hierarchy defined by a design entity consists of the elaboration of the block statement equivalent to the external block defined by the design entity.

    Doesn't that require that all blocks have been generated by the transformation?

tgingold, 2016-05-19:
1. states that configuration declaration aren't considered during elaboration, as they are equivalent to configuration specification.

2. Rewriting is done during elaboration.  Well, this is just a model.

3. No, this is just about the top entity.

I don't think this issue could be fix easily.  I will take into consideration while redesigning elaboration.

Paebbels, 2016-05-24:
I discussed this issue with @JimLewis and @lievenlemiengre after the last IEEE telephone conference meeting. We came to no 100% agreement :) ...

I have two notes:

 1. I found a similar behavior in Altera Quartus. For example Quartus checks ranges in inactive generate statements, which causes boundary errors. A solution to this is to restrict indices and array ranges with min and max functions, so the boundary is always satisfied.

 2. Regarding my problem with GHDL and component binding:
     Let's ask the question in 'reverse'. Should deleting/purging the GHDL working directory have an effect on component binding? I would argue, that the state of the working directory shouldn't have any effect on GHDL.