ghdl/ghdl #67: GHDL shows elaboration error messages at runtime

Paebbels, 2016-05-12:
I was extending my coloring rules for GHDL's output and found this error trace:

```
C:\Tools\GHDL\0.34deb\bin\ghdl.exe:error: report failed
  from: poc.sim_protected.t_sim_status.registerprocess at sim_protected.v08.vhdl:376
  from: poc.simulation.simregisterprocess at sim_simulation.v08.vhdl:132
  from: elaboration of test.fifo_cc_got(tb) at fifo_cc_got_tb.vhdl:114
  from: Elaboartion of design
C:\Tools\GHDL\0.34deb\bin\ghdl.exe:error: error during elaboration
```

I think this message shows 3 issues:

 1. It reports elaboration messages at run time. Is this a copy&past error?
 2. This first line starts with `report failed`, but the report of severity failure was successfully printed. Shouldn't it be something like "reported a fatal message (severity: failure), here is the trace", or so? The current version looks like GHDL had problems in printing the message :)
 3. GHDL uses GCC formatted messages at analysis and elaboration time  
    errors: `file.vhdl:<line>:<col>: <message>`  
    warnings: `file.vhdl:<line>:<col>:warning: <message>`  
    Why does this message format differ at runtime? (It's no real problem, but I need different coloring rules for `-a`/`-e` and `-r` :)

[image]

tgingold, 2016-05-17:
1: No, this is not unexpected.  Part of elaboration is done at execution (to support -gGEN=VAL).
2: Is it easy to reproduce ?
3: Interesting point of view.  OTOH, compiler will only emit warnings/error messages, while runtime message can also come from the user.  That's a reason why I didn't consider the use of standard format for errors.

Tristan.

Paebbels, 2016-05-18:
I'll need some time to reproduce it ... I know at least which file I changed at that day :).

I have no problem if compile-time and run-time errors have a different format. I just noticed it.

tgingold, 2017-09-04:
I am not sure there is anything to do, so closing this one.

closed 2017-09-04 by tgingold